mattanger/ckb-next #188: Shift Select or Cmd Select not Working on macOS Sierra

smachi, 2017-05-12:
This issue was theoretically fixed in the latest version but is not working anymore after a computer reset.

More info about this here: #29

danybmx, 2017-05-13:
Hello @smachi, I tried to solve this on the PR #190. It works for me but I'm not sure about if this fix
 will work on previous macOS versions.

smachi, 2017-05-15:
Hi @danybmx. I'll try your PR but I'm using the last Sierra version too

ssvenn, 2017-05-17:
Works for me on 10.12.4 if i do a "sudo killall ckb-daemon" in a terminal after logging in. I'm not sure what the problem is, probably something about the daemon starting before the user login session is properly up and running...

I've tried to add this to /etc/sudoers so it doesn't require a password:
username ALL=(root) NOPASSWD: /usr/bin/killall ckb-daemon

and then creating a shell script that kills the daemon and adding that script as a login item in macos. The daemon will restart automatically after killing it like that, I'm not sure which process starts it but my modifier keys work afterwards :)

danybmx, 2017-05-17:
Hello @ssvenn appears that the ckb-daemon is launched before the desktop and before the CoreGraphics, that is required to register the mouse events. This is because it's launched as root and not by the user. The problem is that macOS needs that the daemon be launched by root in order to work.

Your workaround is great but if you want you can try the solution of the #190 that uses the ckb.app PID (that starts with the user session) to register the mouse events from the ckb-daemon.

Regards!!

hayley, 2017-07-17:
To add to this, I noticed that ctrl/cmd/option clicking did not work with the magic trackpad when using my k95 rgb keyboard. However, it would work fine if I plugged in my k65 rgb mouse.

@ssvenn's trick of `sudo killall ckb-daemon` fixed it for me.

This is with:
k95rgb keyboard
magic trackpad 2
mac 10.12.5
ckb-next 0.2.8

applejack42, 2017-07-25:
I've found that reseting the daemon helps fix this issue for
- Shift Select
- Command Select 

But Option Select is still not working. For testing purposes, on any link in safari Option click will download the asset. Or how I found it was: Traversing sentences with the option key held down is supposed to skip to each string token.

brockbackups, 2017-11-12:
Anyone resolved this issue in High Sierra 10.13.1, having the exact same issue

ssvenn, 2017-11-12:
Sierra / High Sierra users having trouble could try signing up for the experimental Corsair CUE MacOS version here http://forum.corsair.com/v3/showthread.php?t=168738

It has been rock solid for me and my K95 RGB and has roughly the same functionality as ckb but it's only in closed beta/preview at the moment so they might not accept new testers, or maybe they decided the user base was too small which would be a shame because it's a really nice port.

brockbackups, 2017-11-12:
@ssvenn Already tried it, works very well but completely missing all macOS features ie... modifier keys such as Command, Option, and none of them in the re-map keys list, but I do agree it is rock solid and have sent in feedback to the dev for adding macOS stuff.

porkserver, 2017-12-27:
Compiling from source works for me on High Sierra 10.13.2  The pre built binary is from May and from what I can surmise from comments does not implement the fix.  I just did the quickinstall as I already had the other prereqs installed or setup already for my own coding.  But even if you're a coding noob you can set this up just follow the instructions!  

https://github.com/mattanger/ckb-next#building-from-source

tatokis, 2018-01-22:
https://github.com/ckb-next/ckb-next/issues/18

closed 2018-01-22 by tatokis