mattanger/ckb-next #192: Back and Forward buttons on mice not working

paraschadha, 2017-05-14:
Back and forth buttons are not working (special buttons 4 and 5) in Mac OS. 

If it isn't possible to get this running on mac (at least right), does anyone have any workaround suggestions? I tried running an applescript and macro, both essentially simulating a lcmd + [ combo... But I haven't come up with anything that works!

*OS*: macOS Sierra
*Device used*: M65 Pro

kchen13, 2017-06-18:
M65 RGB Pro here as well. macOS 10.12.5

This is a pretty brilliant solution. Thanks!

I also used this method for using the sniper button to open mission control, I hope there may be a better solutions but for now this works.

Mission Control
> osascript -e 'do shell script "/Applications/Mission\\ Control.app/Contents/MacOS/Mission\\ Control"'

paraschadha, 2017-06-19:
@kchen13 Using the sniper button for mission control is brilliant! I'm going to throw that on mine as well!

nyaray, 2017-07-20:
Clever hacks, but wouldn't a keyboard macro be simpler than doing stuff through osascript?

kchen13, 2017-07-21:
I tried macros before they didn't work. Oddly enough they don't work on Corsairs' official CUE software (beta) in macOS either.

fleischie, 2017-07-21:
Is it possible, that Mac OS just doesn't interpret the appropriate signal/key events?
Does somebody have a non-corsair mice with back/forward buttons, that can check, whether it is specifically a Corsair Mice issue?

P.S.: I will ask a colleague, if I find one with a non-corsair mouse (or any mouse at all). :smile:

light2yellow, 2017-07-21:
@fleischie 
I have Steelseries Rival 100, and yes, without Steelseries' software for macOS such buttons do not work. However, I didn't check it with this software though, however, I believe it is very similar to what they offer on Windows. I will check it during this day on Hackintosh so this is as close to real Mac as possible.

kchen13, 2017-07-21:
@fleischie 
@light2yellow is correct. I used a Logitech Performance MX mouse before and the back/forward buttons do not work without Logitech's software. As far as I know it looks like software regardless of vendor invokes the cmd+[ or cmd+] keyboard shortcuts.

fleischie, 2017-07-21:
Standard USB mouse with side-buttons doesn't work either. So "normal" Mac behavior is not to map back/forward button -> back/forward OS event.

So what's the plan? Roll with Mac-like or coerce Windows/Linux-like behavior?

mhennessie, 2017-09-20:
@paraschadha Thank you for the AppleScripts, they work perfectly and bring a much desired function back.

tatokis, 2018-01-22:
https://github.com/ckb-next/ckb-next/issues/19

closed 2018-01-22 by tatokis