mattanger/ckb-next #193: Can't use a keyboard button and mouse scroll wheel simultaneously

paraschadha, 2017-05-14:
On macOS, in accessibility settings, I have zoom enabled (you use control and the scroll wheel at the same time to zoom into the screen). This, however, doesn't work when ckb is running!

When I quit ckb, it works just fine.

**OS**: macOS Sierra
**Keyboard**: K65 LUX
**Mouse**: M65 Pro (this bug has been recreated with other non-corsair mice as well)

danybmx, 2017-05-15:
I think that this is related to #106 and #190.
Compile using the testing branch and check if it works after the quickinstall.

Regards!

paraschadha, 2017-05-23:
@danybmx Compiled using the testing branch, unfortunately, this issue is not resolved.

Control + scroll wheel works perfectly without ckb (and ckb daemon) running. When ckb (and/or ckb launchdaemon) is running this no longer works.

auibrian, 2017-06-02:
Karabiner also have this problem. I have been investigating but have no solution currently.

TrolldadSATX, 2017-06-13:
I have found the same problem for zooming in and out in Adobe Illustrator.

hoeegh, 2017-06-30:
Me to. But the issue only appears after a reboot. Everything works fine after installation, and after sleep/wake, but after a reboot this unintended behaviour appears. Uninstalling, rebooting and then reinstalling fixes the issue until next reboot.

closed 2018-01-21 by tatokis