mattanger/ckb-next #198: K70 LUX RGB

siojima, 2017-05-17:
ckb.app crashes 2 seconds are opening, and just opens and closes on the tabs above. I am running OS X Yosemite Vers. 10.10.5. Not really sure as to why the app constantly crashes, but is there a solution? P.s The keyboard works fine on my PC but due to school reasons I can't always use my PC so I kinda need for this to work

SethBarberee, 2017-05-18:
Can you give us more info please? What version of CKB are you running? The testing branch had a commit concerning the K70 LUX RGB 2 days ago. If you haven't done it yet, try the testing branch

siojima, 2017-05-18:
I've been trying use the v0.2.8 and also this https://github.com/mattanger/ckb-next/tree/testing.

SethBarberee, 2017-05-18:
Could you attach the output of ckb when you run it with this command?
`sudo launchctl unload /Library/LaunchDaemons/com.ckb.daemon.plist && sudo /Applications/ckb.app/Contents/Resources/ckb-daemon`

With the output, we might be able to see how and why it's crashing

siojima, 2017-05-18:
Sorry, not too sure what you're talking about. I'm not too good with that kind of stuff.

light2yellow, 2017-05-18:
@siojima 
Open a new Terminal instance and paste the command:
```
sudo launchctl unload /Library/LaunchDaemons/com.ckb.daemon.plist && sudo /Applications/ckb.app/Contents/Resources/ckb-daemon
```
 After that, press Enter to execute.

Then open a second Terminal instance and execute:
```
/Applications/ckb.app/Contents/MacOS/ckb --close && /Applications/ckb.app/Contents/MacOS/ckb
```
Paste outputs from both terminals here.

closed 2017-05-21 by siojima