mattanger/ckb-next #203: Delays for Macros

epicTABmaster, 2017-05-23:
I was curious if there is a way to add a delay between certain inputs when typing in a macro for my Corsair Scimitar RGB Pro. I bought the mouse today and have been testing out the program. Everything seems to work perfectly besides the fact that you cannot set a delay between button inputs. This particularly is a problem for me because in League of Legends, for instance, if I type "ADC MIA," using a macro, written like this in code "+enter,-enter,+rshift,+a,-a,+d,-d,+c,-c,-rshift,+space,-space,+rshift,+m,-m,+i,-i,+a,-a,-rshift,+enter,-enter", the game does not receive enough time between the enter input and the beginning of typing. This causes the game to freak out because all of the inputs are pressed nearly instantly in sequence. My previous mouse was a Razer Naga Hex, I had the same issue until I created a .5 second delay between the enter command and the typing. Is there a way to do this in ckb-next?

frickler24, 2017-05-24:
@epicTABmaster Please try this:
`+enter,-enter=500000,+rshift,+a,-a,+d,-d,+c,-c,-rshift,+space,-space,+rshift,+m,-m,+i,-i,+a,-a,-rshift,+enter,-enter`
The delay is in microseconds (yes, not milliseconds).
You can try out the output in a shell-window with starting cat.
If you do not see a difference, I must look for the correct daemon version which handles delays.
In an open PR #25 (for 0.2.9 or later) there is  a first GUI implementation for treating the delays while recording the macros, but I think there are some issues with displays having scale factor > 1.

frickler24, 2017-05-24:
I've just checked it: testing branch does not have a delay implementation yet.
I'll update the PR #25 for you with the current testing branch.

frickler24, 2017-05-24:
OK, it's done.
You may clone it for a first try in a separate directory:
`git clone --single-branch -b macrotime.0.2 https://github.com/frickler24/ckb-next.git`

epicTABmaster, 2017-05-25:
Hey this is @epicTABmaster's sister, I'm helping him out with this.  I have some years of coding experience, however I have not worked with github before.  I'm looking through the directory you provided in the Terminal.  How do we implement this specific macro delay update to his ckb-next?

frickler24, 2017-05-25:
@epicTABmaster and his sister: Just clone and quickinstall:
```
mkdir <the directory where you  want to install, maybe ~/testckb>
cd <the directory just created>
git clone --single-branch -b macrotime.0.2 https://github.com/frickler24/ckb-next.git
cd ckb-next
./quickinstall
```
And answer all three questions with \<return\>

epicTABmaster, 2017-05-27:
awesome sauce, this is on a Mac so I just had to install Homebrew and qt5 first.  It's working perfectly now :)

tatokis, 2017-05-28:
Closing.
Hopefully soon I'll get some time and check out macrotime in order to merge it, it's been long overdue.

closed 2017-05-28 by tatokis