mattanger/ckb-next #207: Buttons toggling

nazar-pc, 2017-05-26:
On M65 Pro RGB I have a sniper button, but in a game I play it is sometimes necessary to hold this button for an extended period of time.
Since button is very hard to press comparing to other buttons it is very inconvenient to keep it pressed for a long time.

On Windows Corsair's app allows to specify buttons toggling, namely I can remap Sniper button to toggle itself, so that I can press it once in order to enter Sniper mode and second time to go back.

ckb-next doesn't have this at the moment, the only closest thing I've got is switching profiles, which does work to some extent as workaround, but not flexible enough.

ThePlasmaRailgun, 2017-11-13:
I'd definitely like to see this...

tatokis, 2018-01-22:
https://github.com/ckb-next/ckb-next/issues/22

closed 2018-01-22 by tatokis