mattanger/ckb-next #208: "Save to hardware" doesn't work properly on M65 Pro RBG

nazar-pc, 2017-05-26:
After pressing "Save to hardware" and closing ckb-next M65 Pro RGB resets colors to while, DPI indicator doesn't light up and actual DPI seems to be set to maximum. No DPI presets present and mouse just works in single mode.

This should be fixed in general and button should be disabled until this is fixed.

nazar-pc, 2017-05-26:
Moreover, DPI buttons were broken even after saving profiles through Corsair's utility until I've reflashed mouse's firmware.

ZirconiumX, 2017-05-27:
@tatokis told me to test this, so I'm going to.

[ ] After pressing "Save to hardware" and closing ckb-next
[ ] M65 Pro RGB resets colors to white
[ ] Actual DPI seems to be set to maximum
[ ] No DPI presets present
[ ] Mouse just works in single mode.
[ ] DPI buttons were broken even after saving profiles

Will update as I go.

ZirconiumX, 2017-05-27:
@nazar-pc: I'm pretty sure the problem is that you are killing the GUI without leaving it running in the background; the GUI contains the animation and DPI logic, and killing it breaks the driver.

To test, instead of killing the GUI, let it run in the tray - the mouse works fine for me like that.

nazar-pc, 2017-05-27:
I was closing GUI with context menu in tray icon. I expect that closing GUI restores mouse to the state in its internal memory. And it works so as far as I can tell if I do not store anything with "Save to hardware" button.

Also unplugging mouse and plugging back in doesn't fix this issue, internal memory seems to be corrupted anyway.

tatokis, 2017-05-28:
@nazar-pc that is indeed correct, however could you please try it again?
I asked @ZirconiumX to test, however they were only able to reproduce it once and never again.

My thoughts originally were that the daemon omitted a crucial packet when switching back to "hardware" mode, however this no longer seems to be the case.

nazar-pc, 2017-05-28:
I believe this is not an issue with switching back to hardware, since stopping GUI and daemon and re-plugging mouse should definitely restore mouse to "hardware" mode as like it was connected to another PC. And it kind of does, but mouse doesn't have any profiles after "Save to hardware" and works in single mode.

Colors might be kept indeed, since I was setting it to white before.

What kind of additional tests should I make?

Flat, 2017-10-12:
Pull request #308 seems to fix this for me.

closed 2017-10-13 by tatokis

fdagpigj, 2017-12-27:
Question: Is it possible to rebind the buttons on the (Harpoon) mouse (in particular, the backward/forward buttons are extremely useless, I'd rather bind them to pageup/pagedown or something) without having the daemon always running? At least I can't find a way to do it, yet the DPI and lighting settings/states are available perfectly fine without the software running. So I'm guessing it is a hardware limit, although I can't seem to find anything stating the hardware limitations. But thanks for the awesome software either way.

ZirconiumX, 2017-12-27:
No, you must have the daemon and GUI running, because the GUI performs software macro translation. The hardware itself doesn't support it. Additionally, a lot of the extended buttons need a proprietary Corsair protocol to use, so software like usbhid or hid-generic won't understand the button

light2yellow, 2017-12-27:
> Is it possible to rebind the buttons on the (Harpoon) mouse (in particular, the backward/forward buttons are extremely useless, I'd rather bind them to pageup/pagedown or something) without having the daemon always running?

Yes, but this is obviously out of the scope of this project. On macOS you can do this with AppleScript (search the tracker, someone has already posted the script), on Linux you can start at [Mouse buttons (ArchWiki)](https://wiki.archlinux.org/index.php/Mouse_buttons). Google is your friend.