mattanger/ckb-next #209: No Device Detected

Aeirex, 2017-05-27:
I have put in the quirks into my GRUB under Arch Linux, but it shows both my K70R and Corsair Scimitar don't work.

This has worked fine under Ubuntu, but I compiled from source and it didn't work under Manjaro Arch.

tatokis, 2017-05-27:
The AUR package specifically states you need to manually enable the systemd service

Aeirex, 2017-05-28:
@tatokis I didn't see anything about the service in the AUR. But I got it.
`sudo systemctl enable ckb-daemon`
`sudo systemctl start ckb-daemon`

closed 2017-05-28 by tatokis

light2yellow, 2017-05-28:
Not sure how it is possible to miss it...
It is even painted with `makepkg` colours :)

[image]

Aeirex, 2017-05-28:
I didn't realize Arch did that. I have only used Arch once in my life, give me a break. :(