mattanger/ckb-next #214: Mouse is jumping on profile switching to lower DPI

nazar-pc, 2017-05-30:
My M65 Pro RGB mouse is jumping when I switch DPI.
Here is what happens:
* Sniper button is assigned to cycle through the available profiles (2 of them, one with higher DPI and one with lower)
* when I'm moving the mouse and switching from higher DPI (2000) profile to lower DPI (500) profile mouse is quickly moving to the place that it can't reach that fast with either higher or lower of those 2 DPIs

The feeling is like between previous profile is deactivated and next profile is activated there is a fraction of second when mouse works at its max DPI possible.

Here is the recorded demo where all of the DPI settings are configured (in Desktop profile I'm using the middle one) and then trying relatively uniformly move the mouse and switch profiles in the process, as you can see when going from lower to higher DPI it is accelerating fine, while when going from higher to lower it jumps quite far from where it should have been.
https://nextcloud.mokrynskyi.com/s/d8hbUYY2NjAeT9A

tatokis, 2017-06-10:
As I don't have a Corsair mouse, I have no idea what could be causing this, but I do believe it happens.
I'll mark this as a bug for now.

nazar-pc, 2017-06-10:
My mouse seems to be a bit different, it is M65**Pro**RGB. Not sure how much different is it internally.

nazar-pc, 2017-09-22:
Doesn't help in my case unfortunately, also it seems to jump twice as far when moving mouse from right to left <- comparing to moving from left to right ->.

nazar-pc, 2017-09-22:
Interesting, the issue is actually fixed when going from higher DPI to lower, but now it happens when I'm going from lower to higher (worked correctly before). So essentially the issue flipped from happening in one direction to instead happening in another direction.

I suspect it is related to the code being moved from after the loop to before the loop in linked commit.

I see `usbsend(kb, data_pkt, 1)` is called multiple times in different branches and in each loop iteration, possibly with `data_pkt` being in different intermediate states, should it really be called that often?

pjcv, 2017-09-22:
Thanks for testing, that's interesting that it switched... are you using the same settings as you sent in #298? Those are the settings with which I reproduced this bug (or at least part of it).

Previously the code would actually set a 0 DPI for the current mode before switching. The intention of the new code was to never have the DPI set to any sort of intermediate value, I think this part at least is working correctly. The data_pkt variable used in each case is actually different as they go out of scope at the end of each block.

I definitely saw huge jumps when switching modes before, now I don't see anything noticeable. It could be down to a difference in mice. We're still entering into a bad state with respect to the "enabled" flag in that code, in that we can set the DPI to a disabled state before enabling it. I can try changing that part as well, though I suspect that if the enabled flag is a problem it's not one that shows up on the Glaive.

pjcv, 2017-09-22:
Here's that change to set the enabled flags before switching DPI state:
https://github.com/pjcv/ckb-next/commit/a90600a8b6df4c090a9f99c896edcdb9a146ce34

pjcv, 2017-09-22:
Regarding calling usbsend() multiple times, I think you're right in that it's probably good practice to lump the messages together since this is supported by the api. Though practically I'm not sure this will affect much right now as the underlying _usbsend() function locks the device and sleeps for each packet individually.

nazar-pc, 2017-09-22:
Yes, I've used the same settings as in #298.
Latest commit works wonderfully, thanks again!

pjcv, 2017-09-22:
Great, we got lucky then :)
Thanks for your patience in testing these changes, everything's always a bit speculative when using different hardware.

closed 2017-10-14 by tatokis