mattanger/ckb-next #219: How to install just the driver without QT on raspberry pi?

Boscop, 2017-06-02:
Is it possible to install just the driver on a raspberry pi without having to install QT?
A driver shouldn't require QT. I just need the driver so that my software can set the LED colors. I don't need any GUI.

frickler24, 2017-06-03:
Yes, theoretically it is possible, but I did not test it.
To do this, clone ckb-next to the raspi.
call ./quickinstall and cancel it with CTRL-C when compilation started.
The `cd src/ckb-daemon` and call `make`.
Maybe some installs for the USB-handling will be neceessary.
If all went ok, you find the binary in ../../bin/.
Depending on what system manager you have, first try to run it manually `sudo ../../bin/ckb-daemon &`.
if all is fine, copy the binary to /usr/bin. 
You find installation commands and stubs for systemd and other managers when you have a look into quickinstall.
Please let us know if it works.

frickler24, 2017-06-03:
But be aware of the functionality of the GUI: changing the color is job of the client, not the daemon.
The daemon gets a static color information and displays it in the keyboard/mouse until the information changes.
So if you want to have changing colors, you need to update the color information frequently by your own software.

Boscop, 2017-06-03:
Thanks, I will try that. 
Yes, I'm aware that my software has to update the color data frequently, I'm already doing that on windows with the official CUE SDK.
Btw, is there some documentation for how to talk to the ckb daemon to send the color information?
Or is the source code the only documentation?

frickler24, 2017-06-03:
plz have a look at DAEMON.md in the install directory.

light2yellow, 2017-06-07:
1. Edit `ckb.pro` in the source root to look like this:
    ```pro
    lessThan(QT_VERSION, 5.2) {
        error("ckb requires at least Qt 5.2!")
    }

    TEMPLATE = subdirs
    CONFIG   += debug_and_release
    SUBDIRS = \
        src/ckb-daemon
    ```
1. Build:
    ```console
    ./qmake-auto
    make
    ```
1. Copy the daemon from `bin/ckb-daemon` to `/usr/bin/ckb-daemon`, it should be owned by `root:root` with `755` permissions.
1. Copy systemd service from `service/systemd/ckb-daemon.service` to `/usr/lib/systemd/system/ckb-daemon.service`, it should be owned by `root:root` with `644` permissions.

tatokis, 2017-06-21:
A solution has been provided.
Closing due to inactivity.

closed 2017-06-21 by tatokis