mattanger/ckb-next #222: Have to "save to hardware" on every reboot to turn off side-lights

andrej-mitrovic-sociomantic, 2017-06-06:
I'm running Ubunty Xenial, and using a Corsair Strafe RGB. The ckb app starts on system start, and it turns off the backlights for the keys, as configured. However, there are also side-lights on this keyboard, and they're turned on until I open the ckb GUI and select "save to hardware".

Is there any way to fix this?

frickler24, 2017-06-06:
@andrej-mitrovic-sociomantic What do you mean with side-lights? [image]
And could you pls send me a screen shot where you configure it in the GUI?

andrej-mitrovic-sociomantic, 2017-06-06:
[image]

[image]

I just have to open the GUI and click "Save to hardware" in order for those lights to turn off.

light2yellow, 2017-06-07:
I guess it's a duplicate of #107.

andrej-mitrovic-sociomantic, 2017-06-07:
Looks like it, thanks for the find!

closed 2017-06-07 by andrej-mitrovic-sociomantic