mattanger/ckb-next #223: Keyboard layout is not correct(K65 RGB Lux)

Vxider, 2017-06-08:
In key binding interface, "right window" key should be "Fn" key.

LoganCalloway, 2017-06-09:
hey my k65 lux RGB want even pop up it says no device connected in ckb could you plz tell me how you got yours to show up.

SethBarberee, 2017-06-12:
@LoganCalloway, this thread is not a support thread. Please open a issue for that.

@Vxider, could you provide us with more info about your system and possibly a screenshot of the GUI?

Vxider, 2017-06-12:
My system version is macos  10.12.5 (Mac Pro Late 2013), here is my screenshot.

[image]

And here is my keyboard

[image]

light2yellow, 2017-06-18:
Please, try `newdev` branch. The commit that is supposed to fix this is https://github.com/mattanger/ckb-next/commit/cfc175070f26f47b100e7e8cb8f273b4377c2849.

light2yellow, 2017-06-24:
@Vxider Any success?

Vxider, 2017-06-25:
Sorry, my keyboard can not be tested due to water damage. Hope others can help to test.

tatokis, 2018-01-21:
Should be fixed

closed 2018-01-21 by tatokis