mattanger/ckb-next #224: K70 Detected as K70R

Aeirex, 2017-06-08:
This is a minor problem, I may be able to fix it and do a pull request, but my K70 is detected as a K70R. This should not be displayed in the GUI at all.

tatokis, 2017-06-17:
What does this exactly mean?

light2yellow, 2017-06-18:
It means that K70 might be detected as K70 Rapidfire.

Please post the output of `ckb-dev-detect` from the source tree root (this is a shell script), logs from `ckb-daemon` (by `systemctl status ckb-daemon`) and a screenshot of `ckb` GUI.

Aeirex, 2017-06-18:
K70R is detected on both Linux and Mac.

I do not have SystemCtl on MacOS. I recently switched to MacOS from Linux.

rijujohnx, 2017-07-19:
This is what happens to my Corsair K70 vengeance too.  Two issues bug me a lot.  
1. Shift key and multi select don't work after a reboot.  I had to reinstall the ckb driver every time to make it work.
2. The keys don't respond at times.  I have to manually disconnect/reconnect the USB connection.

SethBarberee, 2017-07-19:
Can you post the output of 'ckb-dev-detect'?

rijujohnx, 2017-07-19:
@SethBarberee `OS Version:
ProductName:    Mac OS X
ProductVersion: 10.12.5
BuildVersion:   16F73
USB Devices:
            Corsair K70R Gaming Keyboard:
              Product ID: 0x1b09
              Vendor ID: 0x1b1c  (CORSAIR MEMORY INC.)
              Version: 1.09
              Speed: Up to 12 Mb/sec
              Manufacturer: Corsair
              Location ID: 0x14140000 / 9
              Current Available (mA): 500
              Current Required (mA): 500`

light2yellow, 2017-08-20:
@Aeirex try `cmake` branch or pre-built package from #281 and post the output of `/var/log/ckb-next-daemon.log`.

@rijujohnx removing your comments as they are not on the topic of this issue. Instead of poking random threads consider searching for existing ones and subscribing, but if you won't find one, open a new issue describing your problem according to the template.

rijujohnx, 2017-10-02:
Any updates on this issue?

tatokis, 2017-10-02:
Can someone explain to me what the actual issue is? It's impossible for the k70 to be detected as a rapidfire. Additionally, where do you see this detection? All the k70 devices are grouped, so ultimately it doesn't matter.

light2yellow, 2017-10-02:
@rijujohnx 

If you want this problem solved,
> Please post the output of `ckb-dev-detect` from the source tree root (this is a shell script), logs from `ckb-daemon` (by `systemctl status ckb-daemon`) and a screenshot of `ckb` GUI.

rijujohnx, 2017-10-02:
@tatokis `ckb-dev-detect - Mon Oct  2 13:50:23 EDT 2017

Kernel: Darwin mbp-XXXXXXXX 16.7.0 Darwin Kernel Version 16.7.0: Thu Jun 15 17:36:27 PDT 2017; root:xnu-3789.70.16~2/RELEASE_X86_64 x86_64

OS Version:
ProductName:    Mac OS X
ProductVersion: 10.12.6
BuildVersion:   16G29

USB Devices:
            Corsair K70R Gaming Keyboard:
              Product ID: 0x1b09
              Vendor ID: 0x1b1c  (CORSAIR MEMORY INC.)
              Version: 1.09
              Speed: Up to 12 Mb/sec
              Manufacturer: Corsair
              Location ID: 0x14140000 / 12
              Current Available (mA): 500
              Current Required (mA): 500
              Extra Operating Current (mA): 0
            Lenovo USB Optical Mouse:
`
GUI screenshot
https://imgur.com/4poCnR2

rijujohnx, 2017-10-02:
I used the launchctl commands to get few more logging info.  Below is the extract from /var/log/ckb.log  /var/log/ckb.err.log is blank

    ckb: Corsair RGB driver v0.2.8
[I] Root controller ready at /var/run/ckb0
[I] Connecting Corsair K70R Gaming Keyboard at /var/run/ckb1
[I] Registering EventTap for modifier keys.
[I] Starting input thread for /var/run/ckb1
[I] Setup finished for /var/run/ckb1

tatokis, 2017-10-02:
Yeah, not much can be done about that.
The device calls itself "K70R", and the daemon just copies and displays it.

closed 2017-10-02 by tatokis

AaronBottegal, 2017-12-20:
Working on fixing the K70 Calculator issue on Linux. Vengence K70 shares the same product code. Nothing to do, ever.