mattanger/ckb-next #225: Corsair Glaive

white82934, 2017-06-09:
Mouse doesn't work on latest build of Ubuntu Gnome 
[ckb-dev-detect-report.gz](https://github.com/mattanger/ckb-next/files/1063373/ckb-dev-detect-report.gz)

radean0909, 2017-06-09:
Same for me, I posted in the Unsupported Devices thread as well, as a heads up =)

tatokis, 2017-06-10:
I am giving this a higher priority, as judging by the error in dmesg, I assume the mouse doesn't work on Linux at all without the driver.
The dev detect script reported firmware version 1.03, which requires a work around.

I have added the mouse as an M65 in the glaive branch.
Until further work can be done on it, that should be enough to get it up and running on Linux, even though one RGB zone will not function.

https://github.com/mattanger/ckb-next/tree/glaive

Gnork, 2017-06-12:
I installed the ckb-next from the glaive branch, but the mouse still does not function at all for me on Fedora 25 (kernel 4.11).

* ckb-next GUI not showing it.
* mouse does not react at all (no clicking/moving)
* rgb not lighting (except for the small blue DPI indicators)

Output from `journalctl  -u ckb-daemon`.  It does detect my keyboard, but not the mouse:

```
Jun 12 10:59:39 localhost.localdomain ckb-daemon[1273]: [I] Connecting Corsair STRAFE RGB Gaming Keyboard at /dev/input/ckb1
Jun 12 10:59:39 localhost.localdomain ckb-daemon[1273]: [I] Connecting Corsair STRAFE RGB Gaming Keyboard at /dev/input/ckb1
Jun 12 10:59:39 localhost.localdomain ckb-daemon[1273]: [I] claiming interfaces. name=Corsair STRAFE RGB Gaming Keyboard, serial=0602500EAEBD18415619390EF5001943, firmware=0205; Got >> 3<< as ep_str
Jun 12 10:59:39 localhost.localdomain ckb-daemon[1273]: [I] claiming 3 endpoints
Jun 12 10:59:39 localhost.localdomain ckb-daemon[1273]: [I] Starting input thread for /dev/input/ckb1
Jun 12 10:59:44 localhost.localdomain ckb-daemon[1273]: [E] os_usbrecv (via firmware.c:15): Connection timed out
Jun 12 10:59:44 localhost.localdomain ckb-daemon[1273]: [W] _start_dev (device.c:38): Unable to load firmware version/poll rate
Jun 12 10:59:44 localhost.localdomain ckb-daemon[1273]: [I] Setup finished for /dev/input/ckb1
```

Please let me know, how I can contribute useful information about this.

tatokis, 2017-06-12:
@Gnork Apologies, it appears I forgot to add it as a mouse.
Do a git pull and try again.

Gnork, 2017-06-12:
I tried it, but nothing changed.

tatokis, 2017-06-12:
@Gnork Apologies again, I really dropped the ball on this one.
Try again please.

Gnork, 2017-06-12:
@tatokis Still no change. Even the journalctl output does not show something new.

By the way, you don't need to apologize. I think it's amazing that you work on this and I'm glad to help you sort this out.

tatokis, 2017-06-12:
@Gnork any chance you're not running the right binary, or maybe not the right code?
The daemon should have at least tried to claim the new device.

Perhaps you can try manually running it from the tree? `sudo ./bin/ckb-daemon`

Gnork, 2017-06-12:
1. I confirmed that I'm on the correct branch, by taking a look at the source code and saw your changes there.
2. I run ./quickinstall and the binary ./bin/ckb-daemon is updated (confirmed by the timestamp)
3. I stop the running instance with `systemctl stop ckb-daemon`
4. I run the binary manually with `sudo ./bin/ckb-daemon` from the git directory.

This is the output from the manual run:

```
    ckb: Corsair RGB driver beta-v0.2.8+testig
[I] Root controller ready at /dev/input/ckb0
[I] >>>vendor = 0x1b1c, product = 0x1b20, path = /dev/bus/usb/001/002, syspath = /sys/devices/pci0000:00/0000:00:01.3/0000:03:00.0/usb1/1-2
[I] Connecting Corsair STRAFE RGB Gaming Keyboard at /dev/input/ckb1
[I] Connecting Corsair STRAFE RGB Gaming Keyboard at /dev/input/ckb1
[I] claiming interfaces. name=Corsair STRAFE RGB Gaming Keyboard, serial=0602500EAEBD18415619390EF5001943, firmware=0205; Got >> 3<< as ep_str
[I] claiming 3 endpoints
[I] Starting input thread for /dev/input/ckb1
[E] os_usbrecv (via firmware.c:15): Connection timed out
[W] _start_dev (device.c:38): Unable to load firmware version/poll rate
[I] Setup finished for /dev/input/ckb1
```

tatokis, 2017-06-12:
I wonder if this has something to do with it. Try unplugging your strafe and your glaive, start the daemon and plug the glaive in only.

Either that, or I am completely missing something.

Gnork, 2017-06-12:
I tried it by only using my old logitech keyboard now.
ckb-daemon does not show any reaction when plugging in the mouse after starting it.

[ckb-dev-detect-report.gz](https://github.com/mattanger/ckb-next/files/1068097/ckb-dev-detect-report.gz)

The mouse shows up in the ckb-dev-detect-report.

Gnork, 2017-06-12:
Now that the mouse is being detected, the ckb-next GUI software suggests a firmware update. I guess this firmware update is only for the M65 and installing might brick the GLAIVE. This should be disabled somehow, until the GLAIVE is added properly as a separate device, to avoid serious issues.

tatokis, 2017-06-13:
@Gnork Don't worry, the daemon refuses to install firmware updates for the wrong devices.
In the future I will add the glaive as its own device with the right GUI image, but until I can get to that, this should be sufficient.

edflores28, 2017-06-13:
@tatokis i have a glaive mouse and I am interested in contributing. Based on your previous comments it looks like there is more work to be done. Is there any direction you can give on where to start?
Thanks

BIRDMOMSENPAI, 2017-06-27:
iI there any support for the Corsair glaive on macOS? The fix above seems to only be for Linux unless I am mistaken. I don't have any coding experience, so following this thread was very confusing...

tatokis, 2017-06-27:
@BIRDMOMSENPAI Try the newdev branch and tell us what works and what doesn't.

tatokis, 2017-06-27:
@edflores28 Apologies for being late, did not see your message.
You could try adding it as its own category in the daemon and then modify the GUI accordingly.
There's a recent-ish commit in the newdev branch that does exactly that with the k63 so you could use that as a guide.

BIRDMOMSENPAI, 2017-06-27:
@tatokis I'm really sorry, but I'm not sure what I should be doing lol. I downloaded the newdev branch, but what should I do from here?

mikevelez, 2017-07-17:
My glaive doesn't work, can anyone tell me why?:

```
ckb-dev-detect - Mon Jul 17 09:52:26 CDT 2017

Kernel: Linux MeMReS-PC 4.4.0-83-generic #106-Ubuntu SMP Mon Jun 26 17:54:43 UTC 2017 x86_64 x86_64 x86_64 GNU/Linux

OS Version:
Distributor ID:	Ubuntu
Description:	Ubuntu 16.04.2 LTS
Release:	16.04
Codename:	xenial

USB Devices:
Couldn't open device, some information will be missing

Bus 005 Device 009: ID 1b1c:1b34 Corsair 
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0 (Defined at Interface level)
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  idVendor           0x1b1c Corsair
  idProduct          0x1b34 
  bcdDevice            1.03
  iManufacturer           1 
  iProduct                2 
  iSerial                 3 
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength           84
    bNumInterfaces          3
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0xa0
      (Bus Powered)
      Remote Wakeup
    MaxPower              300mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      1 Boot Interface Subclass
      bInterfaceProtocol      2 Mouse
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      58
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
[... 73 lines cut ...]
Jul 17 09:37:12.940616 MeMReS-PC kernel: usb 5-1.4: Product: Corsair Gaming GLAIVE RGB Mouse
Jul 17 09:37:12.940760 MeMReS-PC kernel: usb 5-1.4: Manufacturer: Corsair
Jul 17 09:37:12.944046 MeMReS-PC kernel: input: Corsair Corsair Gaming GLAIVE RGB Mouse as /devices/pci0000:00/0000:00:1c.3/0000:07:00.0/usb5/5-1/5-1.4/5-1.4:1.0/0003:1B1C:1B34.0008/input/input25
Jul 17 09:37:12.944372 MeMReS-PC kernel: hid-generic 0003:1B1C:1B34.0008: input,hidraw0: USB HID v1.11 Mouse [Corsair Corsair Gaming GLAIVE RGB Mouse] on usb-0000:07:00.0-1.4/input0
Jul 17 09:37:12.948206 MeMReS-PC kernel: hid-generic 0003:1B1C:1B34.0009: item 0 1 0 8 parsing failed
Jul 17 09:37:12.948256 MeMReS-PC kernel: hid-generic: probe of 0003:1B1C:1B34.0009 failed with error -22
Jul 17 09:48:10.396220 MeMReS-PC kernel: usb 5-1.4: New USB device found, idVendor=1b1c, idProduct=1b34
Jul 17 09:48:10.396577 MeMReS-PC kernel: usb 5-1.4: Product: Corsair Gaming GLAIVE RGB Mouse
Jul 17 09:48:10.396809 MeMReS-PC kernel: usb 5-1.4: Manufacturer: Corsair
Jul 17 09:48:10.400024 MeMReS-PC kernel: input: Corsair Corsair Gaming GLAIVE RGB Mouse as /devices/pci0000:00/0000:00:1c.3/0000:07:00.0/usb5/5-1/5-1.4/5-1.4:1.0/0003:1B1C:1B34.000B/input/input27
Jul 17 09:48:10.400331 MeMReS-PC kernel: hid-generic 0003:1B1C:1B34.000B: input,hidraw0: USB HID v1.11 Mouse [Corsair Corsair Gaming GLAIVE RGB Mouse] on usb-0000:07:00.0-1.4/input0
Jul 17 09:48:10.400541 MeMReS-PC kernel: hid-generic 0003:1B1C:1B34.000C: item 0 1 0 8 parsing failed
Jul 17 09:48:10.400571 MeMReS-PC kernel: hid-generic: probe of 0003:1B1C:1B34.000C failed with error -22

Dmesg on last boot:

Daemon on current boot:
-- Logs begin at Mon 2017-07-17 09:28:48 CDT, end at Mon 2017-07-17 09:51:02 CDT. --
Jul 17 09:47:44.876063 MeMReS-PC systemd[1]: Started Corsair Keyboard Daemon.
Jul 17 09:47:44.876688 MeMReS-PC ckb-daemon[30910]:     ckb: Corsair RGB driver v0.2.8
Jul 17 09:47:44.876821 MeMReS-PC ckb-daemon[30910]: [I] Root controller ready at /dev/input/ckb0

Daemon on last boot:
Specifying boot ID has no effect, no persistent journal was found
```

SethBarberee, 2017-07-19:
@mikevelez 
Which branch did you download from? 
`master` doesn't have Glaive support however `newdev` has **experimental** support
Try downloading the `newdev` branch and see if that works.

tatokis, 2017-08-06:
I have added proper support for the Glaive in the newdev branch.
It would be appreciated if someone could test it and report back with what works and what doesn't.

All three RGB zones should now function.

Gnork, 2017-08-07:
Hey,

I pulled the new code, but the third RGB zone is still not working. I can select a color with the color-picker, but neither the small dot in the GUI, nor the actual LED are working.

Below is the output of `journalctl -u ckb-daemon`, maybe you see something that's helping. I also started the ckb GUI tool from the commandline, but it doesn't show any errors.

```
Aug 07 18:53:43 localhost.localdomain systemd[1]: Started Corsair Keyboard Daemon.
Aug 07 18:53:43 localhost.localdomain ckb-daemon[1442]:     ckb: Corsair RGB driver beta-v0.2.8
Aug 07 18:53:43 localhost.localdomain ckb-daemon[1442]: [I] Root controller ready at /dev/input/ckb0
Aug 07 18:53:43 localhost.localdomain ckb-daemon[1442]: [I] >>>vendor = 0x1b1c, product = 0x1b20, path = /dev/bus/usb/001/002, syspath = /sys/devices/pci0000:00/0000:00:01.3/0000:03:00.0/usb1/1-1
Aug 07 18:53:43 localhost.localdomain ckb-daemon[1442]: [I] >>>vendor = 0x1b1c, product = 0x1b34, path = /dev/bus/usb/001/003, syspath = /sys/devices/pci0000:00/0000:00:01.3/0000:03:00.0/usb1/1-7
Aug 07 18:53:43 localhost.localdomain ckb-daemon[1442]: [I] Connecting Corsair STRAFE RGB Gaming Keyboard at /dev/input/ckb1
Aug 07 18:53:43 localhost.localdomain ckb-daemon[1442]: [I] claiming interfaces. name=Corsair STRAFE RGB Gaming Keyboard, firmware=0205; Got >> 3<< as ep_str
Aug 07 18:53:43 localhost.localdomain ckb-daemon[1442]: [I] claiming 3 endpoints
Aug 07 18:53:43 localhost.localdomain ckb-daemon[1442]: [I] Connecting Corsair Gaming GLAIVE RGB Mouse at /dev/input/ckb2
Aug 07 18:53:43 localhost.localdomain ckb-daemon[1442]: [I] claiming interfaces. name=Corsair Gaming GLAIVE RGB Mouse, firmware=0103; Got >> 3<< as ep_str
Aug 07 18:53:43 localhost.localdomain ckb-daemon[1442]: [I] claiming 3 endpoints
Aug 07 18:53:43 localhost.localdomain ckb-daemon[1442]: [I] Starting input thread for /dev/input/ckb2
Aug 07 18:53:43 localhost.localdomain ckb-daemon[1442]: [I] Starting input thread for /dev/input/ckb1
Aug 07 18:53:45 localhost.localdomain ckb-daemon[1442]: [I] Setup finished for /dev/input/ckb2
Aug 07 18:53:49 localhost.localdomain ckb-daemon[1442]: [E] os_usbrecv (via firmware.c:15): Connection timed out
Aug 07 18:53:49 localhost.localdomain ckb-daemon[1442]: [W] _start_dev (device.c:38): Unable to load firmware version/poll rate
Aug 07 18:53:49 localhost.localdomain ckb-daemon[1442]: [I] Setup finished for /dev/input/ckb1
```

tatokis, 2017-08-07:
Which zone is the third non-functioning one? (Logo, front or side?)

tatokis, 2017-08-07:
Also, sorry, didn't see the edit in the email, what do you get if you do `cat /dev/input/ckb2/features`?

Gnork, 2017-08-07:
$ cat /dev/input/ckb2/features
corsair m65 rgb pollrate adjrate bind notify fwversion fwupdate

The non-functional RGB zone is the stripes at the side.

tatokis, 2017-08-07:
Apologies, made a correction in https://github.com/mattanger/ckb-next/commit/42f4f2da7fad0d47a8a44655b31a6ab3b3acb197
Please try again with the latest code.

Gnork, 2017-08-07:
I tried the new code but nothing changed.

tatokis, 2017-08-07:
One more time, please.
https://github.com/mattanger/ckb-next/commit/40b2c7998c7df2adc413644da73d837ba78f64c1

Gnork, 2017-08-07:
Ok, now the new artwork is showing up. But the only RGB zone working is now the one at the front. The left side RGB is still standard Yellow and the Logo is completely off. DPI cycling still not working.

```bash
$ cat /dev/input/ckb2/features
corsair glaive rgb pollrate adjrate bind notify fwversion fwupdate
```

tatokis, 2017-08-07:
I am aware of the DPI button not cycling, it is on the TODO list to be implemented before the next release.
This should hopefully correct the RGB Zones, https://github.com/mattanger/ckb-next/commit/cf382adc20fd02f422053355688e34aec3ccb1fe

Gnork, 2017-08-08:
This fixed the logo but not the side.

Gnork, 2017-08-08:
Also the DPI indicator will be turned off now, when starting the ckb GUI, which has not happended before. Maybe you mixed up the side with the DPI indicator.

tatokis, 2017-08-08:
@Gnork would you mind dropping by our IRC room? It would be easier to investigate this further.
(#ckb-next/irc.freenode.net / http://webchat.freenode.net/?channels=%23ckb-next&uio=d4)

Gnork, 2017-08-08:
Hey, I have lot of stuff todo today, but I could be in the IRC room at about 23:00 CET.

What do you think about creating a new branch to investigate this and put some printf statements into the code, to get more debugging information. Then I could run the code for you later tonight, when I'm at home.

tatokis, 2017-08-08:
It's less debugging and more sending commands to the daemon to see which ones control the physical lights on the device, and then telling the GUI to use those instead.

I was also not aware the stripes on the left were DPI indicators, so there's that.

tatokis, 2017-08-08:
Yeah, I only realised that later on, but it still doesn't explain why the latest changes affected the DPI indicators.

Gnork, 2017-08-08:
I'm in IRC now.

tatokis, 2017-08-10:
@Gnork This will hopefully get it working https://github.com/mattanger/ckb-next/commit/c1d009cff4241dce71aa033fb0ff8a224d168d95

Gnork, 2017-08-11:
@tatokis The side color is working now 👍 Awesome work.

F1nny, 2017-08-14:
Working great in newdev! Though just to confirm you guys are aware of the 5 DPI step lights not functioning and the change DPI button options are only up OR down (so when the default up is pressed gets crazy high but won't loop back around haha), as that should loop back on itself after the 5th DPI step setting right? Just wanted to confirm you guys were aware of those two items as look to be final two things for 100% glaive support!:) The 5 DPI lights (none lit) and the DPI setting looping back -Thanks guys amazing work, thanks for carrying on CKB!!

Gnork, 2017-08-15:
DPI settings looping is confirmed to not be working yet.

The thing about the DPI lighting is strange though. They have been working during the last week on my Glaive without any problems. On the other hand, I have seen them being turned off completely, while I was testing various ckb versions for @tatokis the week before. It just started working afterwards, so I thought this was some random glitch.

@F1nny Do the DPI lights work, if you quit the CKB GUI?

SethBarberee, 2017-08-29:
@Gnork 
Could you try out PR #284 and see if that fixes your issue? Feedback in the PR would be appreciated!

Gnork, 2017-09-10:
Hey, sorry for the late response @SethBarberee 
I tested your pull request for the newdev branch and it works perfectly. It cycles through the DPI settings as intended.
Thanks for fixing this :)

SethBarberee, 2017-09-10:
I'm not gonna take credit for someone else's work. That PR was good work on @pjcv 's part.

Uskdara, 2017-09-19:
I just switched to Ubuntu from Windows cause I'm not a big fan of Windows 10.  I have a Glaive and I installed the newdev branch but the glave connects and works for about 5 seconds, then disconnects from the device list in CKB GUI and quits working for 5 seconds, and rinse and repeat over and over.  Anyone else having this issue or have any idea what might be causing it.  I've tried different USB ports to no avail.

jackynikola, 2017-09-22:
Hey, guys. I have a problem with Galive.  In CKB it doesn't recognise mouse. It is disconnected. I am running OS Sierra and I installed from newdev. Any solution besides one explained in files?

tatokis, 2017-09-22:
@jackynikola not sure, it should work https://github.com/mattanger/ckb-next/blob/newdev/src/ckb-daemon/usb_mac.c#L870

Can you perhaps get some logs from the daemon? (I don't remember how, on mac, maybe someone more familiar can chime in.)

@Uskdara also check logs. If not sure how, run ckb-dev-detect from the repository.

jackynikola, 2017-09-22:
Thanx for help  Tatokis.  The mouse is working as well as led but since I cant adjust dpi and remove lights its useless to me. Maybe I did instal wrong.  SO I went to quick instal. Then I installed CKB_next. Is that correct?

jackynikola, 2017-09-22:
Thanx for help Tatokis. The mouse is working as well as led but since I cant adjust dpi and remove lights its useless to me.
[image]

superherointj, 2017-09-27:
Installed ckb-next newdev branch on Ubuntu 17.04 and confirmed support to Corsair Glaive mouse. But the strong lights/led is quite annoying. Is there a way to disable it in full already? Or can you suggest a path forward to have this strong lights fixed?
The GUI also crashed when I tried to click on something.

Maw1395, 2017-09-28:
This is amazing. Thank you so much!

markdrake, 2017-10-02:
Thank you so much, my new mouse didn't work and this was the solution, I owe you a beer :)

tatokis, 2017-10-14:
Glaive is now supported in master.

closed 2017-10-14 by tatokis