mattanger/ckb-next #231: Bug: uploading k70 RGB firmware broke my keyboard

Ryunaq, 2017-06-15:
Just wanted to point it out since I already fixed it in a windows machine. Probably better not to update the firmware when it already works I guess.

frickler24, 2017-06-15:
Is there a bug in the install procedure, did you get a wrong firmware (not designed for your KB)  or did you downgrade to an older version of firmware? And what do you mean with "broke"?

Ryunaq, 2017-06-16:
I installed ckb normally. When I opened it, it prompted "There's a new firmware update for your k70 RGB" or something along the lines, so it had my model right. I clicked update, it downloaded normally and installed it to my keyboard. After that, all the leds turned off except for the three that indicate Num-lock, caps-lock and scroll-lock, which instead of turning off started blinking. My pc wasn't recognizing any key press neither. Fix was uninstalling the firmware from a windows machine.

tatokis, 2017-06-16:
What's your keyboard?

Ryunaq, 2017-06-16:
K70 RGB, as I said, it recognized the model correctly.

tatokis, 2017-06-17:
Yes but there hasn't been a new firmware for the K70 RGB for years now. Perhaps you have one of the Lux or Rapidfire variants?

Ryunaq, 2017-06-17:
Nope, just k70 RGB. But there may be something to it. All I did to fix it was delete the firmware I installed through ckb. And when I plugged it again it said that my keyboard had the firmware up to date. So maybe what happened is that it recognized correctly the model, but incorrectly the firmware it had (I had the one up to date, it seems), and somehow installing the "new firmware" (which was actually the same) "broke" my keyboard.

tatokis, 2017-06-17:
@Ryunaq That sounds impossible. The daemon explicitly extracts the usb id from the firmware image and compares it with the one of the actual device.

light2yellow, 2017-06-18:
Seems like I reported exactly the same issue without reading this one in advance: #235.

> After that, all the leds turned off except for the three that indicate Num-lock, caps-lock and scroll-lock, which instead of turning off started blinking.

I didn't get such behaviour on Linux, however I got it under Windows, I suppose it means the keyboard is in bootloader mode.

Feel free to provide the logs.

Ryunaq, 2017-06-18:
Sure, but silly question, where do I find the logs?

light2yellow, 2017-06-18:
OS?

Ryunaq, 2017-06-18:
Arch Linux

light2yellow, 2017-06-18:
You can view the logs using `systemctl status ckb-daemon`. But. I reread your messages.

Right now you have few choices: you can either go to the systemd journal and find a boot at which the update happened, grepping for `ckb-daemon` and posting the logs around update or try to force lower firmware to bait an upgrade. The thing is, when you updated the keyboard in Windows (I mean, when you were willing to fix it), it got the newest firmware version available, so you cannot recreate the process fully. That's why the first option is preferred, however I realize, that it can to too tedious.

Then, the second option is to something similar to #235, but I did this based on the codes from the PR and yet to conduct this on `testing` or `master`, so the results can be different for you. @tatokis tested the PR and for him it was okay, the keyboard was working (he has K70 RGB) and in the end he needed to restart the daemon and everything went well. I'm leaving further description up to him in the PR thread. The PR link in #235 has the description how to manually force version and then it contains a comment with results I got, also with commands I executed (if you use bash, don't add a pipe `|` to the end of the command). Be sure to `killall -9 ckb` before changing `fwversion`.

light2yellow, 2017-06-18:
Derpson Kowalski from #systemd IRC channel just told me how to easily go into history for our daemon.

For example,
```console
journalctl -b -0 --unit=ckb-daemon
```
is the same as
```console
systemctl status ckb-daemon
```
so, it shows the status for current boot.

But
```console
journalctl -b -1 --unit=ckb-daemon
```
shows the status for the last boot, and so on, you can increase the integer to go deeper into previous boots to find the logs we are looking for. And the logs we are looking for are some signs of error.

Ryunaq, 2017-06-22:
I'm sorry for the late response, I accidentaly deleted the email and luckily saw it again while looking for something. Here's the log:

[logs.txt](https://github.com/mattanger/ckb-next/files/1093704/logs.txt)

light2yellow, 2017-06-22:
Yeah, this is the same as in #235.

@tatokis suggested to force `fwupdate` if the name of the device contains `Bootloader`. However, the reason why the update leads to the bootloader mode is yet to be discovered.

Ryunaq, 2017-06-23:
Oh welp, that is interesting. Thanks for caring.

tatokis, 2017-09-05:
Closing as duplicate of #235

closed 2017-09-05 by tatokis