mattanger/ckb-next #232: OSX 10.12.5 KG65

bblue-e, 2017-06-15:
I keep getting "-mmacosx-version-min=" and another issue which only says "[debug/device.o]"
any advice?
[image]

frickler24, 2017-06-15:
Interesting. Since two weeks I get the same issue in the travis-ci compilation with macos. 
Please any idea from our macos users?!

bblue-e, 2017-06-16:
update: It really wasn't an issue because the program works fine oops

frickler24, 2017-06-17:
Even if it runs now for @bblue-e, I still have a problem with it.
When I  perform the compilation via travis-ci, all linux related steps work fine.
Since ca. 2 weeks the compilation for macOS stops with error.
The reason seems to be that since then I do not get qt5.7 as the default, but 5.9.0.
I have no idea how to enforce the usage of 5.7 again (adding 57 to qt does not work).

I get a linker error I do not understand:
`clang: error: invalid version number in '-mmacosx-version-min='`

What he wants to  link is the following:
```
/Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/clang++ -stdlib=libc++ -headerpad_max_install_names \
 -arch x86_64 -Wl,-syslibroot,/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.11.sdk -mmacosx-version-min=10.9 -Wl,-rpath,@executable_path/Frameworks \
 -o ../../../ckb-next/ckb.app/Contents/MacOS/ckb release/main.o release/mainwindow.o release/kbwidget.o release/colorbutton.o release/settingswidget.o release/kblightwidget.o release/keymap.o \
 release/media_linux.o release/kblight.o release/kbprofile.o release/kbanimwidget.o release/animscript.o release/kbanim.o release/animadddialog.o release/animsettingdialog.o release/gradientbutton.o release/gradientdialog.o \
 release/gradientdialogwidget.o release/kbmode.o release/kb.o release/rlistwidget.o release/kbprofiledialog.o release/keywidget.o release/kbbindwidget.o release/kbbind.o release/rebindwidget.o release/modeselectdialog.o \
 release/JlCompress.o release/qioapi.o release/quaadler32.o release/quacrc32.o release/quagzipfile.o release/quaziodevice.o release/quazip.o release/quazipdir.o release/quazipfile.o release/quazipfileinfo.o release/quazipnewinfo.o \
 release/unzip.o release/zip.o release/kbfirmware.o release/fwupgradedialog.o release/autorun.o release/ckbsettings.o release/kbperf.o release/ckbsettingswriter.o release/keyaction.o release/mperfwidget.o release/kperfwidget.o \
 release/layoutdialog.o release/extrasettingswidget.o release/kbmanager.o release/colormap.o release/macroreader.o release/media_mac.o release/qrc_image.o release/qrc_text.o release/qrc_binary.o release/moc_mainwindow.o \
 release/moc_kbwidget.o release/moc_colorbutton.o release/moc_settingswidget.o release/moc_kblightwidget.o release/moc_kblight.o release/moc_kbprofile.o release/moc_kbanimwidget.o release/moc_animscript.o release/moc_kbanim.o \
 release/moc_animadddialog.o release/moc_animsettingdialog.o release/moc_gradientbutton.o release/moc_gradientdialog.o release/moc_gradientdialogwidget.o release/moc_kbmode.o release/moc_kb.o release/moc_rlistwidget.o \
 release/moc_kbprofiledialog.o release/moc_keywidget.o release/moc_kbbindwidget.o release/moc_kbbind.o release/moc_rebindwidget.o release/moc_modeselectdialog.o release/moc_quagzipfile.o release/moc_quaziodevice.o release/moc_quazipfile.o \
 release/moc_kbfirmware.o release/moc_fwupgradedialog.o release/moc_kbperf.o release/moc_ckbsettingswriter.o release/moc_keyaction.o release/moc_mperfwidget.o release/moc_kperfwidget.o release/moc_layoutdialog.o \
 release/moc_extrasettingswidget.o release/moc_kbmanager.o release/moc_macroreader.o   \
 -F/usr/local/Cellar/qt/5.9.0/lib -framework Foundation -framework AudioToolbox -lz -framework QtWidgets \
 -framework QtGui -framework QtCore -framework DiskArbitration -framework IOKit -framework QtNetwork 
```

All modules have been compiled wit this option without error, eg:
```
/Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/clang++ -c -pipe -stdlib=libc++ -Wno-unused-parameter -O2 -std=gnu++11  -arch x86_64 -isysroot \
 /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.11.sdk -mmacosx-version-min=10.9 -Wall -W -fPIC \
 -DCKB_VERSION_STR=\"`cat /Users/travis/build/frickler24/ckb-next/src/ckb/../../VERSION | tr -d \n`\" -DQUAZIP_STATIC -DQT_NO_DEBUG -DQT_WIDGETS_LIB -DQT_GUI_LIB -DQT_NETWORK_LIB -DQT_CORE_LIB -I. \
 -I/usr/local/Cellar/qt/5.9.0/lib/QtWidgets.framework/Headers -I/usr/local/Cellar/qt/5.9.0/lib/QtGui.framework/Headers -I/usr/local/Cellar/qt/5.9.0/lib/QtNetwork.framework/Headers -I/usr/local/Cellar/qt/5.9.0/lib/QtCore.framework/Headers \
 -Irelease -I/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.11.sdk/System/Library/Frameworks/OpenGL.framework/Headers \
 -I/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.11.sdk/System/Library/Frameworks/AGL.framework/Headers -I. -I/usr/local/Cellar/qt/5.9.0/mkspecs/macx-clang \
 -F/usr/local/Cellar/qt/5.9.0/lib -o release/moc_macroreader.o release/moc_macroreader.cpp
```
Please can someone tell me how to enforce qt5.7 again or to fix this issue otherwise?

light2yellow, 2017-06-18:
@frickler24 I believe this is a known bug for 5.7 - 5.9, see [this](https://bugreports.qt.io/browse/QTBUG-56965?jql=text%20~%20%22QMAKE_MACOSX_DEPLOYMENT_TARGET%22).

As the bug tells, it has been fixed in git branches for these versions but since we are stuck with binaries, you can try to workaround the issue by creating a `.qmake.conf` in the root, putting there 
```pro
QMAKE_MACOSX_DEPLOYMENT_TARGET 10.9
```

frickler24, 2017-06-18:
Thank you, I'm just running a test with this setting.

light2yellow, 2017-06-21:
@frickler24 Did that help?

frickler24, 2017-06-22:
Sorry, no, same effect. But I had no time for further testing. Hopefully on sunday.

light2yellow, 2017-06-23:
Hmm, well. Try removing all `QMAKE_MACOSX_DEPLOYMENT_TARGET ...` from all `.pro` files. Also, forcing Qt version is really good for testing, I mean, it can be done to ensure backwards compatibility or something like that, but when a new version of Qt comes out, we should support it.

Also, try removing `QMAKE_MACOSX_DEPLOYMENT_TARGET ...` at all. I suppose the environment should pick up/set up everything as it should be, and the minimal deployment version is required only when we distribute packaged versions (`.pkg`).

light2yellow, 2017-07-12:
Just a fact for those who wait - all (active) maintainers failed to find the reason and fix the problem so far. However, considering everything worked somehow, this must be an upstream's regression - either Qt's or AppleClang's.

light2yellow, 2017-07-22:
I have finally found a workaround and pushed it into master. It depends on your Qt installation. I suppose you are using Homebrew and its `qt` package. Otherwise you will have to edit the script to provide your own paths. Enough room for improvement.

closed 2017-08-26 by light2yellow

sterpe, 2017-10-08:
So, I'm running Qt 5.9.2, and this is still broken.

It appears the original bug has been fixed in Qt 5.8 branch https://codereview.qt-project.org/#/c/180793/2/mkspecs/features/mac/default_post.prf

It looks like the problem is we have that little switch in `default_post.prf` around L100 (in 5.9.2):

```
    macos: deployment_target = $$QMAKE_MACOSX_DEPLOYMENT_TARGET
    ios: deployment_target = $$QMAKE_IOS_DEPLOYMENT_TARGET
    tvos: deployment_target = $$QMAKE_TVOS_DEPLOYMENT_TARGET
    watchos: deployment_target = $$QMAKE_WATCHOS_DEPLOYMENT_TARGET
```

So confirmed taht the value `$$QMAKE_MACOSX_DEPLOYMENT_TARGET` makes it's this far, but the 'macos' switch is never `true`.  So `$$deployment_target` never gets a value.  I'm not sure where this value comes from.

Based upon their own change set (linked above) I just make this little fix:

```
    # default
    deployment_target = $$QMAKE_MACOSX_DEPLOYMENT_TARGET
    # else
    ios: deployment_target = $$QMAKE_IOS_DEPLOYMENT_TARGET
    tvos: deployment_target = $$QMAKE_TVOS_DEPLOYMENT_TARGET
    watchos: deployment_target = $$QMAKE_WATCHOS_DEPLOYMENT_TARGET
```

And then setting ``QMAKE_MACOSX_DEPLOYMENT_TARGET` works as expected, either from the commandline or a `.qmake.conf` file, etc.

It's weird because it's not clear why the `macos:` switch is not hit.  But I wonder if this issue is  related to the original Qt ticket or if the project might be missing some setup condition or other environmental variable.

commit 45d18850 (2017-12-09): qmake: better QMAKE_MACOSX_DEPLOYMENT_TARGET fix

Closes: #232, #310

Squashed commit of the following:

commit 042a6556de67f46c57ad1d6456b5328ac29ab933
Author: Oleksii Vilchanskyi <oleksii.vilchanskyi@gmail.com>
Date:   Sat Dec 9 12:56:39 2017 +0100

    Adopt #310

commit 694fc3e835c5afe8d63803ca31a1495956470d94
Merge: d6782ff ab81204
Author: Oleksii Vilchanskyi <oleksii.vilchanskyi@gmail.com>
Date:   Sat Dec 9 12:42:33 2017 +0100

    Merge branch 'master' of git://github.com/sterpe/ckb-next into sterpe-master

commit ab812045fa64e213525890498b5afad69619980a
Author: Steven R Terpe <srterpe@ [... cut]
commit 91044964 (2017-12-09): qmake: better QMAKE_MACOSX_DEPLOYMENT_TARGET fix

Closes: #232, #310

Squashed commit of the following:

commit 042a6556de67f46c57ad1d6456b5328ac29ab933
Author: Oleksii Vilchanskyi <oleksii.vilchanskyi@gmail.com>
Date:   Sat Dec 9 12:56:39 2017 +0100

    Adopt #310

commit 694fc3e835c5afe8d63803ca31a1495956470d94
Merge: d6782ff ab81204
Author: Oleksii Vilchanskyi <oleksii.vilchanskyi@gmail.com>
Date:   Sat Dec 9 12:42:33 2017 +0100

    Merge branch 'master' of git://github.com/sterpe/ckb-next into sterpe-master

commit ab812045fa64e213525890498b5afad69619980a
Author: Steven R Terpe <srterpe@ [... cut]