mattanger/ckb-next #233: No Devices connecter MacOS K65 RGB

FE-Tobin, 2017-06-16:
tried:
disconnecting & connecting
restarting computer
reinstalling program
reinstalling on a new profile

Any other advice? I'm running on Sierra btw

frickler24, 2017-06-17:
@FE-Tobin please try the following:
- deinstall ckb completely (check with `pgrep ckb` if anything is still running)
- run `echo n | quickinstall`, (answers automatic-questions with "n")
- run the client with `bin/ckb &` (yes, really bin without /, so you get the new compiled version in your path). The client will not detect anything, but this is ok.
- in another terminal (to divide the messages from the client and the daemon): `cd <your ckb-directory>; sudo bin/ckb-daemon`

With starting the daemon some messages should appear and (hopefully) your client should show all devices, the daemon could detect.

Please post the messages of the daemon and - if you find something helpful  - the messages from the client.

tatokis, 2017-06-17:
@frickler24 I doubt they compiled from source.
@FE-Tobin Can you please run ckb-dev-detect from the source code and upload the file it creates?

light2yellow, 2017-06-18:
To post the output of the daemon, follow https://github.com/mattanger/ckb-next/issues/170#issuecomment-298458572.

Sandvich, 2017-07-10:
Hi, I'm having a similar issue on Fedora with my K70 Lux. The output from the daemon is below:

`[I] >>>vendor = 0x1b1c, product = 0x1b36, path = /dev/bus/usb/005/038, syspath = /sys/devices/pci0000:00/0000:00:13.0/usb5/5-5
[I] Connecting Corsair Gaming K70 LUX Keyboard at /dev/input/ckb1
[I] claiming interfaces. name=Corsair Gaming K70 LUX Keyboard , firmware=0104; Got >> 4<< as ep_str
[I] claiming 4 endpoints
[I] Starting input thread for /dev/input/ckb1
[E] loadrgb_kb (led_keyboard.c:273): Bad input header
[W] _start_dev (device.c:70): Unable to load hardware profile
[I] Setup finished for /dev/input/ckb1
[I] Stopping input thread for /dev/input/ckb1
[E] os_usbsend (via led_keyboard.c:119):  Timer expired, res = 0xffffffff
[I] Attempting reset...
[E] os_resetusb (via usb.c:470): resetusb failed: No such device
[E] usb_tryreset (usb.c:478): Reset failed. Disconnecting.
[I] Disconnecting /dev/input/ckb1
[I] Removed device path /dev/input/ckb1`

Anything insightful in there?

tatokis, 2017-09-05:
@Sandvich please open a new issue if you are still having trouble. This is not the right thread.

closed 2017-09-05 by tatokis