mattanger/ckb-next #238: Profiles are randomly deleted

101arrowz, 2017-06-20:
MacOS, ckb-next v0.28. Any ideas?

101arrowz, 2017-06-20:
What happens is this: at night, everything works fine on my new profile, and I quit ckb-next. The next morning, all of my custom settings on ALL of my profiles are reset to white keys without animation, and bindings are reset to default. The only profile where this doesn't happen is HwProfile1, the default that came with my keyboard's firmware. I would just use that, but that has issues too (Cmd key broken). This happens sporadically, usually 2 days apart, but always overnight. I am not good with software, so I don't know how to stop it, and I probably can't fix it without step by step instructions. Help!
EDIT: I use K70 LUX RGB. Does that have anything to do with it?

light2yellow, 2017-06-20:
There are other issues like this with profiles damaged/corrupted/deleted/edited on macOS, somebody might link to them later. Nobody was able to reproduce this, although a temporary workaround was posted in one of the threads, if you happen to spot such repro, don't hesitate to inform.

frickler24, 2017-06-22:
@101arrowz What I do not understand: Why and when do you quit ckb-next? If you powerdown your system or otherwise?
What happens if you restart ckb-next after quitting? Do you run some backup/Restore or OS-changing jobs before shutting down or over night or are you using different OSes?
If you restart your system is ckb-next started automatically or is there any chance to have a look on the date of the config file?

feel2good, 2017-08-10:
I am using cbk for a year now also on K70 LUX RGB and this happens to me randomly every 2-3 months, i do not quit or edit cbk it just happens after a restart, profiles are there but all renamed with no configuration.
The only workaround that i found to save my time is to restore "com.ckb.ckb.plist" in Preferences folder from a backup, i am on a mac using Arq, quiting cbk overwriting "com.ckb.ckb.plist" and opening cbk again did the trick , all my profiles binding settings are there.
I hope that helps.

tatokis, 2017-08-10:
@feel2good Can you send us a working version and a broken version of the file?

feel2good, 2017-08-10:
I renamed the not working version as com.ckb.ckb.plistBK

[Archive.zip](https://github.com/mattanger/ckb-next/files/1214734/Archive.zip)

feel2good, 2017-08-10:
please note that i also have a Corsair Sabre connected that the profiles were not affected , i has the issue only with the keyboard

light2yellow, 2017-08-20:
@feel2good `com.ckb.ckb.plist` is (soon will be "was", `org.next.ckb.plist` will be used instead) a file that gets copied into `~/Library/LaunchAgents/` and ensures GUI autostart at log in. So, I don't really understand what you are talking about.

feel2good, 2017-08-21:
check ~/Library/Preferences/ , i never said anything about LaunchAgents

light2yellow, 2017-08-21:
@feel2good I see now.