mattanger/ckb-next #240: Keyboard *sometimes* doesn't work upon boot

light2yellow, 2017-06-23:
Basically, about 30% of the boots end up with this problem. Keyboard is lighted with a saved profile but is "dead". Plugging in and out helps. Here are messages from both current and last boots provided: last boot was successful, current one wasn't, therefore the keyboard was replugged and got picked up since then.

I don't know where to look and what to look for. I see `usbfs` errors, so it might be something kernel is primarily not able to deal with? Race conditions? Whatever else?

`ckb-dev-detect-report` (modified it a bit to fulfill my needs, download: [ckb-dev-detect-report.gz](https://github.com/mattanger/ckb-next/files/1098629/ckb-dev-detect-report.gz)):

```console
ckb-dev-detect - Pá čen 23 19:51:29 CEST 2017

Kernel: Linux manjaro-workstation 4.9.33-1-MANJARO #1 SMP PREEMPT Sat Jun 17 14:07:44 UTC 2017 x86_64 GNU/Linux

OS Version:
LSB Version:    n/a
Distributor ID: ManjaroLinux
Description:    Manjaro Linux
Release:        17.0.1
Codename:       Gellivara

USB Devices:
Couldn't open device, some information will be missing

Bus 003 Device 005: ID 1b1c:1b20 Corsair 
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0 
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  idVendor           0x1b1c Corsair
  idProduct          0x1b20 
  bcdDevice            2.05
  iManufacturer           1 
  iProduct                2 
  iSerial                 3 
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength           84
    bNumInterfaces          3
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0xa0
      (Bus Powered)
      Remote Wakeup
    MaxPower              500mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      1 Boot Interface Subclass
      bInterfaceProtocol      1 Keyboard
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      67
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
[... 121 lines cut ...]
čen 23 19:07:27.866003 ckb-daemon[383]: [I] Connecting Corsair STRAFE RGB Gaming Keyboard at /dev/input/ckb1
čen 23 19:07:27.866003 ckb-daemon[383]: [I] claiming interfaces. name=Corsair STRAFE RGB Gaming Keyboard, firmware=0205; Got >> 3<< as ep_str
čen 23 19:07:27.866003 ckb-daemon[383]: [I] claiming 3 endpoints
čen 23 19:07:28.011616 ckb-daemon[383]: [I] Starting input thread for /dev/input/ckb1
čen 23 19:07:30.122324 ckb-daemon[383]: [I] Setup finished for /dev/input/ckb1

Daemon on last boot:
-- Logs begin at Sun 2017-03-19 00:13:51 CET, end at Fri 2017-06-23 19:51:25 CEST. --
čen 23 18:26:57.193845 systemd[1]: Started Corsair Keyboard Daemon.
čen 23 18:26:57.202961 ckb-daemon[314]:     ckb: Corsair RGB driver beta-v0.2.8
čen 23 18:26:57.203237 ckb-daemon[314]: [I] Root controller ready at /dev/input/ckb0
čen 23 18:27:03.281625 ckb-daemon[314]: [I] >>>vendor = 0x1b1c, product = 0x1b20, path = /dev/bus/usb/003/005, syspath = /sys/devices/pci0000:00/0000:00:14.0/usb3/3-4
čen 23 18:27:03.381753 ckb-daemon[314]: [I] Connecting Corsair STRAFE RGB Gaming Keyboard at /dev/input/ckb1
čen 23 18:27:03.381753 ckb-daemon[314]: [I] claiming interfaces. name=Corsair STRAFE RGB Gaming Keyboard, firmware=0205; Got >> 3<< as ep_str
čen 23 18:27:03.381753 ckb-daemon[314]: [I] claiming 3 endpoints
čen 23 18:27:03.647289 ckb-daemon[314]: [I] Starting input thread for /dev/input/ckb1
čen 23 18:27:05.758825 ckb-daemon[314]: [I] Setup finished for /dev/input/ckb1
čen 23 19:05:52.324937 systemd[1]: Stopping Corsair Keyboard Daemon...
čen 23 19:05:52.326076 ckb-daemon[314]: [I] Caught signal 15
čen 23 19:05:52.725097 ckb-daemon[314]: [I] Disconnecting /dev/input/ckb1
čen 23 19:05:52.913192 ckb-daemon[314]: [I] Removed device path /dev/input/ckb1
čen 23 19:05:52.916046 ckb-daemon[314]: [I] Closing root controller
čen 23 19:05:52.916195 ckb-daemon[314]: [I] Removed device path /dev/input/ckb0
čen 23 19:05:52.916618 systemd[1]: Stopped Corsair Keyboard Daemon.
```

light2yellow, 2017-06-23:
Comment by the creator of `ckb` from a similar report https://github.com/ccMSC/ckb/issues/159#issuecomment-157592942:
> That's relatively normal. The keyboards sometimes reject the profile setup packets. I've never been able to figure out why exactly but the driver just works around it. CUE has the same issue sometimes so I assume it's just a hardware glitch.

I then assume `Attempting reset...` from the logs tries to resend the setup packages, but fails with a warning and an error afterwards, and then says `Reset success`, `Setup finished` which is not really true, because the keyboard remains dead.

light2yellow, 2017-06-26:
News.

In comparison to the last log, during which I replugged the keyboard after being tired to wait, this time I didn't and it got picked up in ~3 secs.

But. The log above contains

```console
rqt 33 rq 9 len 1 ret -110
rqt 161 rq 1 len 64 ret -110
```

while the log below contains

```console
rqt 33 rq 9 len 1 ret -110
```

So, it appears that `rqt 33 rq 9 len 1 ret -110` is recoverable while `rqt 161 rq 1 len 64 ret -110` is not recoverable. Or the former, followed by the latter combine into unrecoverable state.

_However, if so, the daemon must not report the keyboard to be ready._ Also drawing your attention to the fact that they go one right after another (see timings).

Still, I hope someone capable might take a look on this. I'll try to continue gathering interesting information, that's all I can do for now.

```console
Dmesg on current boot:
čen 26 11:02:57.943039 kernel: Command line: BOOT_IMAGE=/boot/vmlinuz-4.9-x86_64 root=UUID=345a9f45-a318-4395-ba09-ed07772d4abc rw usbhid.quirks=0x1B1C:0x1B20:0x20000408 resume=UUID=de638197-f81c-467c-9839-e5b619e5de3f
čen 26 11:02:57.965763 kernel: Kernel command line: BOOT_IMAGE=/boot/vmlinuz-4.9-x86_64 root=UUID=345a9f45-a318-4395-ba09-ed07772d4abc rw usbhid.quirks=0x1B1C:0x1B20:0x20000408 resume=UUID=de638197-f81c-467c-9839-e5b619e5de3f
čen 26 11:02:58.022988 kernel: input: Corsair Corsair STRAFE RGB Gaming Keyboard as /devices/pci0000:00/0000:00:14.0/usb3/3-4/3-4:1.0/0003:1B1C:1B20.0004/input/input3
čen 26 11:02:58.023106 kernel: hid-generic 0003:1B1C:1B20.0004: input,hidraw3: USB HID v1.11 Keyboard [Corsair Corsair STRAFE RGB Gaming Keyboard] on usb-0000:00:14.0-4/input0
čen 26 11:03:33.877452 kernel: input: ckb1: Corsair STRAFE RGB Gaming Keyboard as /devices/virtual/input/input20
čen 26 11:03:33.877525 kernel: input: ckb1: Corsair STRAFE RGB Gaming Keyboard as /devices/virtual/input/input21
čen 26 11:03:34.387446 kernel: usb 3-4: usbfs: USBDEVFS_CONTROL failed cmd ckb-daemon rqt 33 rq 9 len 1 ret -110

Daemon on current boot:
-- Logs begin at Sun 2017-03-19 00:13:51 CET, end at Mon 2017-06-26 11:04:33 CEST. --
čen 26 11:02:58.277936 systemd[1]: Started Corsair Keyboard Daemon.
čen 26 11:02:58.289523 ckb-daemon[329]:     ckb: Corsair RGB driver beta-v0.2.8
čen 26 11:02:58.289813 ckb-daemon[329]: [I] Root controller ready at /dev/input/ckb0
čen 26 11:02:58.294912 ckb-daemon[329]: [I] >>>vendor = 0x1b1c, product = 0x1b20, path = /dev/bus/usb/003/004, syspath = /sys/devices/pci0000:00/0000:00:14.0/usb3/3-4
čen 26 11:03:33.817925 ckb-daemon[329]: [I] Connecting Corsair STRAFE RGB Gaming Keyboard at /dev/input/ckb1
čen 26 11:03:33.817925 ckb-daemon[329]: [I] claiming interfaces. name=Corsair STRAFE RGB Gaming Keyboard, firmware=0205; Got >> 3<< as ep_str
čen 26 11:03:33.817925 ckb-daemon[329]: [I] claiming 3 endpoints
čen 26 11:03:33.874934 ckb-daemon[329]: [I] Starting input thread for /dev/input/ckb1
čen 26 11:03:34.384231 ckb-daemon[329]: [E] os_sendindicators (usb_linux.c:217): Connection timed out
čen 26 11:03:34.384231 ckb-daemon[329]: [I] Attempting reset...
čen 26 11:03:34.728800 ckb-daemon[329]: [I] claiming 3 endpoints
čen 26 11:03:34.767226 ckb-daemon[329]: [I] >>>vendor = 0x1b1c, product = 0x1b20, path = /dev/bus/usb/003/004, syspath = /sys/devices/pci0000:00/0000:00:14.0/usb3/3-4
čen 26 11:03:36.910593 ckb-daemon[329]: [I] Reset success
čen 26 11:03:36.910901 ckb-daemon[329]: [I] Setup finished for /dev/input/ckb1
```

light2yellow, 2017-06-26:
Apologies, actually the timings for the first case are following:

```console

čen 23 19:06:13.444742 ckb-daemon[383]:     ckb: Corsair RGB driver beta-v0.2.8
čen 23 19:06:13.444945 ckb-daemon[383]: [I] Root controller ready at /dev/input/ckb0
čen 23 19:06:13.448637 ckb-daemon[383]: [I] >>>vendor = 0x1b1c, product = 0x1b20, path = /dev/bus/usb/003/004, syspath = /sys/devices/pci0000:00/0000:00:14.0/usb3/3-4
čen 23 19:06:48.811123 ckb-daemon[383]: [I] Connecting Corsair STRAFE RGB Gaming Keyboard at /dev/input/ckb1
čen 23 19:06:48.811123 ckb-daemon[383]: [I] claiming interfaces. name=Corsair STRAFE RGB Gaming Keyboard, firmware=0205; Got >> 3<< as ep_str
čen 23 19:06:48.811123 ckb-daemon[383]: [I] claiming 3 endpoints
čen 23 19:06:48.867998 ckb-daemon[383]: [I] Starting input thread for /dev/input/ckb1
čen 23 19:06:49.374253 ckb-daemon[383]: [E] os_sendindicators (usb_linux.c:217): Connection timed out
čen 23 19:06:49.374253 ckb-daemon[383]: [I] Attempting reset...
!!!
čen 23 19:06:49.377468 kernel: usb 3-4: usbfs: USBDEVFS_CONTROL failed cmd ckb-daemon rqt 33 rq 9 len 1 ret -110
!!!
čen 23 19:06:49.718805 ckb-daemon[383]: [I] claiming 3 endpoints
čen 23 19:06:49.782793 ckb-daemon[383]: [I] >>>vendor = 0x1b1c, product = 0x1b20, path = /dev/bus/usb/003/004, syspath = /sys/devices/pci0000:00/0000:00:14.0/usb3/3-4
čen 23 19:06:55.107476 ckb-daemon[383]: [E] os_usbrecv (via firmware.c:15): Connection timed out
čen 23 19:06:55.107476 ckb-daemon[383]: [W] _start_dev (device.c:38): Unable to load firmware version/poll rate
čen 23 19:06:55.107476 ckb-daemon[383]: [I] Reset success
čen 23 19:06:55.107927 ckb-daemon[383]: [I] Setup finished for /dev/input/ckb1
!!!
čen 23 19:06:55.110752 kernel: usb 3-4: usbfs: USBDEVFS_CONTROL failed cmd ckb-daemon rqt 161 rq 1 len 64 ret -110
!!!
```

The timings for the second case:
```console
čen 26 11:02:58.289523 ckb-daemon[329]:     ckb: Corsair RGB driver beta-v0.2.8
čen 26 11:02:58.289813 ckb-daemon[329]: [I] Root controller ready at /dev/input/ckb0
čen 26 11:02:58.294912 ckb-daemon[329]: [I] >>>vendor = 0x1b1c, product = 0x1b20, path = /dev/bus/usb/003/004, syspath = /sys/devices/pci0000:00/0000:00:14.0/usb3/3-4
čen 26 11:03:33.817925 ckb-daemon[329]: [I] Connecting Corsair STRAFE RGB Gaming Keyboard at /dev/input/ckb1
čen 26 11:03:33.817925 ckb-daemon[329]: [I] claiming interfaces. name=Corsair STRAFE RGB Gaming Keyboard, firmware=0205; Got >> 3<< as ep_str
čen 26 11:03:33.817925 ckb-daemon[329]: [I] claiming 3 endpoints
čen 26 11:03:33.874934 ckb-daemon[329]: [I] Starting input thread for /dev/input/ckb1
čen 26 11:03:34.384231 ckb-daemon[329]: [E] os_sendindicators (usb_linux.c:217): Connection timed out
čen 26 11:03:34.384231 ckb-daemon[329]: [I] Attempting reset...
!!!
čen 26 11:03:34.387446 kernel: usb 3-4: usbfs: USBDEVFS_CONTROL failed cmd ckb-daemon rqt 33 rq 9 len 1 ret -110
!!!
čen 26 11:03:34.728800 ckb-daemon[329]: [I] claiming 3 endpoints
čen 26 11:03:34.767226 ckb-daemon[329]: [I] >>>vendor = 0x1b1c, product = 0x1b20, path = /dev/bus/usb/003/004, syspath = /sys/devices/pci0000:00/0000:00:14.0/usb3/3-4
čen 26 11:03:36.910593 ckb-daemon[329]: [I] Reset success
čen 26 11:03:36.910901 ckb-daemon[329]: [I] Setup finished for /dev/input/ckb1
```

So, the message gets logged after a reset is attempted. Then, `rqt 33 rq 9 len 1 ret -110` is "harmless", more or less.

This way, `rqt 161 rq 1 len 64 ret -110` remains under question. It might be the result of a failed reset attempt, even though logs report success. It might not. Is it possible to implement a reset attempt after such situation has happened?

embluk, 2017-06-27:
I also have this issue when I was on Linux Mint but recently changed to KDE neon and it has not happened yet, but on Linux Mint for me it happened a lot.

frickler24, 2017-06-27:
With the current testing branch  I have similar effects on linux mint 18.x, but not after boot, but after wakeing up from standby-modus.
What helped as a hack is a short script which is called via systemctl mechnism after wakeup:
`/etc/systemd/system/suspend.target.wants/ckb-daemon-restart.service` has the following content:
```
[Unit]
Description=Restart Corsair Keyboard Daemon at resume
After=suspend.target
After=hibernate.target
After=hybrid-sleep.target

[Service]
Type=simple
#ExecStart=/bin/systemctl restart ckb-daemon.service
ExecStart=/bin/sh -ec 'sleep 3; echo "restart Wakeup" > /dev/input/ckb1/cmd'
StandardOutput=syslog
StandardError=inherit

[Install]
WantedBy=suspend.target
WantedBy=hibernate.target
WantedBy=hybrid-sleep.target
```
The script is enabled with `systemctl enable ckb-daemon-restart.service`

Hopefully at weekend I'll have some time for debugging the reasons for this misbehavior.

light2yellow, 2017-06-27:
So, I browsed the sources for a bit. The error that causes problems happens in `getfwversion`, where `usbrcv` is called.

```c
int _usbrecv(usbdevice* kb, const uchar* out_msg, uchar* in_msg, const char* file, int line){
    // Try a maximum of 5 times
    for (int try = 0; try < 5; try++) {
        // Send the output message
        pthread_mutex_lock(mmutex(kb)); ///< Synchonization between macro and color information
        DELAY_SHORT(kb);
        int res = os_usbsend(kb, out_msg, 1, file, line);
        pthread_mutex_unlock(mmutex(kb));
        if (res == 0)
            return 0;
        else if (res == -1) {
            // Retry on temporary failure
            if (reset_stop)
                return 0;
            DELAY_LONG(kb);
            continue;
        }
        // Wait for the response
        DELAY_MEDIUM(kb);
        res = os_usbrecv(kb, in_msg, file, line);
        if(res == 0)
            return 0;
        else if(res != -1)
            return res;
        if(reset_stop || hwload_mode != 2)
            return 0;
        DELAY_LONG(kb);
    }
    // Give up
    ckb_err_fn("Too many send/recv failures. Dropping.\n", file, line);
    return 0;
}
```

So, instead of resetting the keyboard, like it's done in `sendindicators`, the delay happens. I guess we should try resetting the device like in the mentioned function:

```c
void os_sendindicators(usbdevice* kb) {
    static int countForReset = 0;
    struct usbdevfs_ctrltransfer transfer = { 0x21, 0x09, 0x0200, 0x00, 1, 500, &kb->ileds };
    int res = ioctl(kb->handle - 1, USBDEVFS_CONTROL, &transfer);
    if(res <= 0) {
        ckb_err("%s\n", res ? strerror(errno) : "No data written");
        if (usb_tryreset(kb) == 0 && countForReset++ < 3) {
            os_sendindicators(kb);
        }
    }
}
```

tangy92, 2017-07-27:
I'm having this problem as well, with or without the driver. Rebooting or replugging keyboard will solve the problem about 25% of the time. Tried installing Ubuntu and it still won't work.

embluk, 2017-07-27:
@tangy92 I've since moved to KDE Neon and had no problems.

xavierp, 2017-09-14:
Hi there,
any update on this issue ?

I think i may have a related problem.
I have a K65RGB and a glaive mouse and from this morning i can't boot properly. _(I'm using Archlinux - kernel - ARCH 4.12.12-1)_

Right after boot, i'm asked to type my password to decrypt my HD and keyboard is not responding. Keyboard is lighting but no keys are responding.
I still can boot using the fallback init_ram_fs and the keyboard works fine.

To fix it i tried to:
- plug and unplug it: **no change**.
- set a new HW profile using ckb gui after booting on fallback init_ram_fs: **no change**
- add kernel parameter `usbhid.quirks=0x1B1C:0x1B17:0x20000408`: **no change**
- set a new HW profile using windows CUE software: **no change**

Btw I have an error when i plug in the keyboard, but it was there before when things where working thus i don't know if it is related: `usbhid 1-4:1.2: couldn't fin an input interrupt endpoint` 

I think maybe it stopped working after i updated the default HW profil using ckb gui.

Any idea on what's happening ?
What kind of logs may help you ?

Any helps appreciated.
Tanks.

light2yellow, 2017-09-14:
@xavierp 
> I think i may have a related problem.

> from this morning i can't boot properly.

From the first post:

> Basically, about 10% of the boots end up with this problem.

So, no, your problem is different if it is permanent. This is the wrong issue to post about it.

felix-apo, 2017-10-30:
Experiencing the frozen keyboard at  loginscreen 10-20% of the startup trys of Ubuntu 16.10, 17.04 and 17.10.

Keyboard version:
[Mo Okt 30 09:35:56 2017] input: ckb1: Corsair STRAFE Gaming Keyboard as /devices/virtual/input/input19

Is there anymore information one could try gathering?

riverty, 2017-11-13:
This is happening to me too!

I'm running Ubuntu 17.10 and have the Corsair K70 LUX keyboard. I'm guessing about 10% of the time the keyboard will work at the Ubuntu (Gnome 3) login screen, most of the time = no joy. Unplugging and re-plugging the keyboard seems to get the keyboard recognized and working properly. I have no other issues with it during my session. I should point out that the keyboard works fine in the BIOS and at the Grub bootloader screen.

Anything I can offer to try and troubleshoot this is welcome.

tangy92, 2017-11-14:
@riverty I am still having the problem. The furthest I got was some logs that showed the kernel couldn't read the USB configuration descriptors all the time, but sometimes it can and then the keyboard works. Don't know enough to go any further unfortunately :(

riverty, 2017-11-26:
@tangy92, Try something on your end. If you are running Ubuntu (17.10 here), try removing "splash" and "quiet" from your grub2 config (/etc/default/grub), make sure you run 'update-grub' after your edits, and see what you get. It's not as pretty on the bootup, but several days with several boots now and no troubles so far. Now, I don't know what a splash screen and/or a "quiet" boot would do cause hell detecting a keyboard, but...

Let us know how it goes for you and good luck!

rt...

riverty, 2017-12-16:
@tangy92 Well, I wanted to take a minute and post in that while this worked quite well for me for a while, the problem is back. Now I'm trying different USB settings in the BIOS...

closed 2018-01-21 by tatokis

nsneck, 2018-07-05:
For what it's worth, this issue was fixed upstream (in Linux itself) for the RGB version, but the Corsair Strafe (non-RGB) still suffered from the same issue. I have a patch to fix it here: https://github.com/nsneck/linux/commit/b20ffc3ff5548a911472fcec40de68baf9005744

And I submitted it for inclusion in upstream ( https://marc.info/?l=linux-usb&m=153054877713280&w=2 ). It'll hopefully be in 4.18 and maybe make it to stable series if accepted, not sure.

ZirconiumX, 2018-07-05:
> same issue. I have a patch to fix it here: nsneck/linux@b20ffc3

There are two types of Strafe NRGB. 1b15, and 1b44. Thought it was worth
pointing that out.

nsneck, 2018-07-05:
> There are two types of Strafe NRGB. 1b15, and 1b44. Thought it was worth pointing that out.
Oh, cool. I'll try to spin up a v2 of the patch during the next few days so it'll get fixed for both NRGB's then.

ZirconiumX, 2018-07-05:
The 1b44 has some hardware changes (it's now 8-bit colour instead of 3-bit
colour), but knowing Corsair, both are broken.