mattanger/ckb-next #242: Wrong profile without CKB open

M4R10zM0113R, 2017-06-28:
Saving to hardware and then quitting the software seems to make it change to a very low DPI setting with all colors set to a clear tear, 2 mice with same behavior:
Sabre RGB
Scimitar RGB

Some background:
Changing from ckb to ckb-next didn't bring the issue up, since ckb used to not provoke the problem when quitting the program.

light2yellow, 2017-06-28:
Could you explain the problem in more detail? There's a daemon, called `ckb-daemon` and a frontend for it, `ckb`. The binaries were not renamed so far to `ckb-next`. AFAIK you won't have animations on mouse unless you have `ckb` running. The daemon is used to control the device while GUI is used for different dynamic configurations.

Could you write steps to reproduce the issue? This will help a lot to answer the first question.

light2yellow, 2017-06-28:
So, `Save to hardware` basically resets the device? And you cannot configure it in any way because every time you try to save the configuration it gets reset?

Can we also see the output of `ckb-dev-detect` from the root of the source code tree?

light2yellow, 2017-06-28:
Does your problem look similar to the one on #208?

M4R10zM0113R, 2017-06-29:
Let me Explain:
The "Save to Hardware" feature in ckb-next doesn't stick to the mouse but instead sets it to the low-dpi profile mentioned earlier. I tried saving multiple profiles, resetting profiles in ckb, reinstalling all together, to get the same relative results (both mice)

M4R10zM0113R, 2017-06-29:
And to answer the #208 question: It's precisely that, but to the low-DPI spectrum. This however didn't happen with ccMSC's ckb.

tatokis, 2017-06-29:
Can you please do a git bisect on this?
This should be a good start commit https://github.com/mattanger/ckb-next/commit/34cd4c8920fbd3dc5d8dc4148da24edd515864b1

M4R10zM0113R, 2017-06-29:
is there any way to reflash to default on the mice? I wanna get rid to as many factors as possible on the bisection

M4R10zM0113R, 2017-06-30:
Excuse my dirty work/lack of git knowledge beforehand.
I bisected the project from 34cd4c8 to the latest.
Attached files have a file I made and a zip containing ".git/refs/bisect" 's contents.
[bisect-refs.zip](https://github.com/mattanger/ckb-next/files/1113953/bisect-refs.zip)
[bisect.txt](https://github.com/mattanger/ckb-next/files/1113954/bisect.txt)

M4R10zM0113R, 2017-06-30:
sorry @light2yellow I oversaw the ckb-dev-detect part...
here it is:
[ckb-dev-detect-report.gz](https://github.com/mattanger/ckb-next/files/1114021/ckb-dev-detect-report.gz)

tatokis, 2017-06-30:
Thank you for your time and the detailed bisect.
This is clearly a regression caused by one of indicator bugfix commits.
CC @frickler24

AaronBottegal, 2017-08-08:
I have a new scimitar RGB, if anyone has any patches to fix it, I'll test them. I see the same behavior, have to keep the daemon running or it goes low DPI with nothing happening color-wise. I'll test any patches you want to see if it fixes the issue.

tatokis, 2017-10-12:
@AaronBottegal check out #308, see if it helps.