mattanger/ckb-next #243: Strafe RGB not being recongnized most of the time

NdranC, 2017-06-29:
I recently tried to install arch linux on my desktop pc and I noticed that my STRAFE RGB was hiving some errors on the live USB boot. After a single reboot it looked like everything was fine and proceeded with the installation like normal.

After I was done I noticed I was getting the error almost all the time now. This is what it looks like:
https://i.imgur.com/LMO7iR7.jpg

If I reboot enough times I get a slightly different error that seems to at least allow the keyboard to work smilingly normal. I also noticed I can force this to happen reliably if I boot with the LTS kernel instead of the latest:
https://i.imgur.com/IsU6DfC.jpg

Booting with the usbhid.quirks fix doesn't seem to fix anything (I believe I applied this correctly with bootctl):
https://i.imgur.com/YRwKiYn.jpg

When I "cat /sys/modules/parameters/quirks" I can see the modifications taking place but the keyboard wont work any better.

This are my settings and firmware under windows:
https://i.imgur.com/pOirVww.png

I also tried majaro KDE with similar results. Installing the ckb-next or the ckb-next-git AUR package made no difference and it never managed to detect my keyboard when launched.

I have noticed that the keyboard (when I get the second error that makes it kinda work) shows up correctly under lsusb but as soon as I un-plug and re-plug it again it will almost never work, forcing me to have to reboot.

light2yellow, 2017-06-29:
Post the output of `ckb-dev-detect` from the root of the source code tree.

light2yellow, 2017-06-29:
> Unable to enumerate USB device...

Looks like hardware or kernel issue, try googling your errors. Here's what Google told me, found [here](https://bbs.archlinux.org/viewtopic.php?id=134726):

```
# sudo modprobe -vr uhci_hcd
# sudo modprobe -va ohci_hcd
# sudo modprobe -va uhci_hcd
```

```
$ cat /etc/modprobe.d/usb_hci_order.conf 
#
# create a dependency on ohci for uhci, which fixes problems
# with external usb devices not showing up
#
softdep uhci_hcd pre: ohci_hcd
```

Also, check your USB interfaces with `xusb` as shown in #138.

Strafe RGB works well with ckb, so this is either a hardware or a kernel problem.

tatokis, 2017-06-29:
The interrupt endpoint warning is non-fatal and can be safely ignored.
The Unable to enumerate usb device error basically means that the linux kernel couldn't talk to your device at all.
Make sure it uses the latest firmware from corsair, that it's not plugged on a hub, try it on another usb bus/controller (eg if you use it on USB2, try USB3 and vice versa).

NdranC, 2017-06-29:
This is the output of ckb-dev-detect:
https://pastebin.com/LrmLiamG

I'll look in the mean time into what you found on the arch forums.

@tatokis So we are talking about 2 separate problems then? Witht he newer kernels I get the "enumerate" error that should not be happening. On the other hand, the LTS kernel works as it should even with the "interrupt endpoint" but once inside the OS I have a second issue with the ckb-next not detecting the keyboard.

Despite the fact that I'm able to proceded with the LTS kernel, once inside if for any reason I re-plug my keyboard I will lose the ability to use it and when I check dmesg it shows the same "enumerate" errors like with the newer kernels.

I have also tried several front and back USBs and they all have the same issue. It has indeed the last firmware update which is 2.05. I doubt is hardware since it seems to work fine under windows for the last year.

NdranC, 2017-06-29:
Since I'm in manjaro this time, I decided to run systemctl start ckb-daemon and apparantly that worked. I swear that didn't work when I was on arch. It seems to work fine except that If I un-plug the device and reconnect it it had a hard time working again. It took me over 5 unplugs for it to finally work again.

Thsi is the output of dmesg on the last events while I was doing that.
https://pastebin.com/CSmkYXYB

light2yellow, 2017-06-29:
> I decided to run...

If you are using any package from AUR, you __have to__ start and/or enable the service if you want the daemon to start automatically.

NdranC, 2017-06-29:
@light2yellow Yeah I tried that before but I swear it didn't work while I had arch installed. I'm keeping it under manjaro until I figure out how the system should be working under this keyboard.

Still doesn't solve the issue with the latests kernels or when I replug my keyboard while the OS is running.

NdranC, 2017-06-30:
Using the solution on that old arch linux post did nothing as far as I can tell.

This is what xusb and lsusb say when the keyboard is working but with errors: https://pastebin.com/VKrHpT3J

and this is not working after unplug: https://pastebin.com/NA6aFL8v

dmesg says:
`[ 1149.385327] usb 1-13: new full-speed USB device number 49 using xhci_hcd
[ 1154.605451] usb 1-13: unable to read config index 0 descriptor/all
[ 1154.605453] usb 1-13: can't read configurations, error -110
[ 1154.605465] usb usb1-port13: unable to enumerate USB device`

light2yellow, 2017-08-20:
> Booting with the usbhid.quirks fix doesn't seem to fix anything (I believe I applied this correctly with bootctl):
https://i.imgur.com/YRwKiYn.jpg

No, the quirks are wrong. You forgot `2` after `0x`: `0x1B1C:0x1B20:0x20000408`.

But in general, as @tatokis said, the problem happens before ckb-next-daemon even has a chance to talk with the keyboard, so... look for a hardware problem on fresher kernels, I guess?

closed 2017-08-20 by light2yellow