mattanger/ckb-next #244: Kernel error:couldn't find an input interrupt endpoint

Dr-Noob, 2017-06-29:
I've started using ckb in Linux with a Corsair K70 RGB and everything looks great. First, I needed to put "usbhid.quirks=0x1B1C:0x1B13:0x20000408" in the kernel boot args to get ckb working. That did the trick, and now everything's fine, but, when I boot the computer, linux kernel shows the error: 
"usbhid 3-13:1.2: couldn't find an input interrupt endpoint"
which (I think) is caused by adding this line to kernel args. I'm using ArchLinux.
Thank you very much, and btw, nice software, this is great :)

light2yellow, 2017-06-29:
Please, post the output of `ckb-dev-detect` from the root of the source code tree.

Also, try another USB port.
Also, isn't the keyboard in BIOS mode by any chance?

tatokis, 2017-06-29:
`couldn't find an input interrupt endpoint` is not a problem.
It is emitted by the hid driver, but since ckb doesn't use the kernel's hid driver, you can freely ignore that warning.

closed 2017-06-29 by tatokis

light2yellow, 2017-06-29:
You can always use `0x4` quirks. See README.

tatokis, 2017-06-29:
If that is the case, then the quirks are not being applied properly.
The message will show up with and without the quirks, it's just that the right quirk is going to lessen the boot time as the hid driver won't sit around waiting before it times out.

Dr-Noob, 2017-06-30:
I tried in another USB, but nothing different happened. The keyboard is not in BIOS mode, it's in 1ms polling rate. If you're interested in the output of `ckb-dev-detect` yet, I'll post it.
So, tatokis, you're right. If I remove the quirks, the message is still there.
I read all the documentation about ckb, I knew I could use `0x4`, just thought it was not a good idea, because I don't know how to launch `ckb-daemon` at boot properly. For example, `sudo ckb-daemon &` does not work for me. I need to launch daemon in foreground to get it working...If you could help me out with this it would be really helpful too.
Thanks for the help, but just for sure, can't I do anything more? Is there a way to apply the quirks properly? Thank you very much for your time and attention!

light2yellow, 2017-06-30:
> because I don't know how to launch ckb-daemon at boot properly

If you are issuing `quickinstall`, it enables the daemon for you.
If you are using any of the AUR packages, it explicitly tells you what to do in a colored bold font by showing [this](https://aur.archlinux.org/cgit/aur.git/tree/ckb-next.install?h=ckb-next) message.
If you are unfamiliar with systemd and continue using systemd-based Linux distro, don't hesitate to thoroughly read [this](https://wiki.archlinux.org/index.php/Systemd) wiki page.