mattanger/ckb-next #245: CKB window doesn't show up

vipertrig, 2017-07-02:
Hi, I'm a new linux user. I am using the 2017.1 version of kali linux. After all the I done all the instruction, whenever I tried to run CKB, it just doesn't show up. 

Someone help me pls!

The report is followed:

ckb-dev-detect - Sun Jul  2 00:59:24 EDT 2017

Kernel: Linux LEAF 4.9.0-kali3-amd64 #1 SMP Debian 4.9.18-1kali1 (2017-04-04) x86_64 GNU/Linux

OS Version:
Distributor ID:	Kali
Description:	Kali GNU/Linux Rolling
Release:	kali-rolling
Codename:	kali-rolling

USB Devices:

Bus 001 Device 005: ID 1b1c:0c09 Corsair 
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               1.10
  bDeviceClass            0 (Defined at Interface level)
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  idVendor           0x1b1c Corsair
  idProduct          0x0c09 
  bcdDevice            1.00
  iManufacturer           1 Corsair Components, Inc.
  iProduct                2 H100i v2
  iSerial                 3 7289_2.0
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength           32
    bNumInterfaces          1
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0x80
      (Bus Powered)
    MaxPower               50mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           2
      bInterfaceClass       255 Vendor Specific Class
      bInterfaceSubClass      0 
      bInterfaceProtocol      0 
      iInterface              0 
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x02  EP 2 OUT
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               0
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x82  EP 2 IN
        bmAttributes            2
          Transfer Type            Bulk
[... 225 lines cut ...]
Jul 02 00:49:44.045112 ckb-daemon[3560]: [I] Removed device path /dev/input/ckb1
Jul 02 00:49:44.616183 ckb-daemon[3560]: [I] >>>vendor = 0x1b1c, product = 0x1b13, path = /dev/bus/usb/001/008, syspath = /sys/devices/pci0000:00/0000:00:14.0/usb1/1-1
Jul 02 00:49:44.712523 ckb-daemon[3560]: [I] Connecting Corsair K70 RGB Gaming Keyboard at /dev/input/ckb1
Jul 02 00:49:44.712523 ckb-daemon[3560]: [I] claiming interfaces. name=Corsair K70 RGB Gaming Keyboard , firmware=0205; Got >> 3<< as ep_str
Jul 02 00:49:44.712523 ckb-daemon[3560]: [I] claiming 3 endpoints
Jul 02 00:49:44.879908 ckb-daemon[3560]: [I] Starting input thread for /dev/input/ckb1
Jul 02 00:49:46.989174 ckb-daemon[3560]: [I] Setup finished for /dev/input/ckb1
Jul 02 00:54:09.016176 ckb-daemon[3560]: [I] Caught signal 15
Jul 02 00:54:09.016208 systemd[1]: Stopping Corsair Keyboard Daemon...
Jul 02 00:54:09.219703 ckb-daemon[3560]: [I] Disconnecting /dev/input/ckb1
Jul 02 00:54:09.579445 ckb-daemon[3560]: [I] Removed device path /dev/input/ckb1
Jul 02 00:54:09.583381 ckb-daemon[3560]: [I] Closing root controller
Jul 02 00:54:09.583381 ckb-daemon[3560]: [I] Removed device path /dev/input/ckb0
Jul 02 00:54:09.583225 systemd[1]: Stopped Corsair Keyboard Daemon.
Jul 02 00:54:09.669662 systemd[1]: Started Corsair Keyboard Daemon.
Jul 02 00:54:09.670574 ckb-daemon[5674]:     ckb: Corsair RGB driver v0.2.8
Jul 02 00:54:09.670697 ckb-daemon[5674]: [I] Root controller ready at /dev/input/ckb0
Jul 02 00:54:09.672515 ckb-daemon[5674]: [I] >>>vendor = 0x1b1c, product = 0x1b13, path = /dev/bus/usb/001/008, syspath = /sys/devices/pci0000:00/0000:00:14.0/usb1/1-1
Jul 02 00:54:09.772850 ckb-daemon[5674]: [I] Connecting Corsair K70 RGB Gaming Keyboard at /dev/input/ckb1
Jul 02 00:54:09.772850 ckb-daemon[5674]: [I] claiming interfaces. name=Corsair K70 RGB Gaming Keyboard , firmware=0205; Got >> 3<< as ep_str
Jul 02 00:54:09.772850 ckb-daemon[5674]: [I] claiming 3 endpoints
Jul 02 00:54:09.918409 ckb-daemon[5674]: [I] Starting input thread for /dev/input/ckb1
Jul 02 00:54:12.028313 ckb-daemon[5674]: [I] Setup finished for /dev/input/ckb1

Daemon on last boot:

light2yellow, 2017-07-02:
Firstly, use triple backticks when pasting logs.
Secondly, try switching the keyboard from BIOS mode to normal mode. The usage in BIOS mode is very limited.

light2yellow, 2017-07-08:
So, how's it going without BIOS mode?

closed 2017-07-25 by light2yellow