mattanger/ckb-next #246: Scimitar RGB firmware out of date (now 2.0.4)

OscarBarrett, 2017-07-02:
https://www3.corsair.com/software/HID/Scimitar.zip now contains `SCIMITAR_APP_V204.bin`, as such the hash in FIRMWARE no longer matches.

light2yellow, 2017-07-02:
Can you try to upgrade your gear manually through ckb-next GUI? The only reason of not adding it is the lack of testing. So, if you happen to own Scimitar RGB, you'll help a lot by upgrading it through ckb-next and reporting what got broken.

OscarBarrett, 2017-07-03:
I have upgraded manually and haven't noticed any issues so far. Although the upgrade was from 1.06 😅 

I'll comment again once I've spent some more time using the new firmware.

OscarBarrett, 2017-07-08:
I haven't noticed any new problems since the upgrade. General use, lighting and keybinds are all working fine.

light2yellow, 2017-07-21:
Unfortunately I lost my access to ckb-next key pair due to unreachable private key and revocation file for it which was signed to provide such access, so I am not able to update the file unless I re-request the access. I will need some time to study this topic and figure out a better path to persistent and reliable setup with portable offline private key which is always reachable. Unassigning.

commit 4a69697b (2017-07-21): Update Scimitar RGB firmware to 2.04.
Closes #246
closed 2017-07-21 by tatokis