mattanger/ckb-next #247: Keyboard not working on boot

MiGu3X, 2017-07-03:
So I installed ckb-next and when attempting to make my device work (Corsair Strafe Red) it doesn't.

It appears to work after I unplug and plug, but I don't want to do this every time I boot my linux...

Dev report file:
https://mega.nz/#!JFcTVJoA!HYIjqeVzS6W4nSp_AdjxouLgncHVnxvz4UkUtGEJ5KE

light2yellow, 2017-07-03:
Are you using something else than systemd?

MiGu3X, 2017-07-04:
I installed ckb-next as how it says in the instructions.

I guess no.

light2yellow, 2017-07-10:
Please post the output of `systemctl status ckb-daemon`. Also `systemctl --version`.

gkatsiapis, 2017-07-10:
Same for me. Uppon boot (and login screen) neither my K70 LUX RGB is recognised, nor my M65 Pro RGB. Same with adding cmdline parameters.

If I unplug my keyboard and re-plug it after 30+ seconds it gets detected but this is not the case with my mouse (though it randomly works sometimes)

[ckb-dev-detect-report.gz](https://github.com/mattanger/ckb-next/files/1136456/ckb-dev-detect-report.gz)

light2yellow, 2017-07-10:
@MiGu3X @gkatsiapis 
Please update your `master` branch, I suppose there was an error in `ckb-dev-detect`.

MiGu3X, 2017-07-10:
Ok. Ill update as soon as I can. If the problem persists I will post my logs

gkatsiapis, 2017-07-10:
[ckb-dev-detect-report.gz](https://github.com/mattanger/ckb-next/files/1137201/ckb-dev-detect-report.gz)
 Updated.

MiGu3X, 2017-07-10:
My keyboard worked, daemon status logs:
```
● ckb-daemon.service - Corsair Keyboard Daemon
   Loaded: loaded (/usr/lib/systemd/system/ckb-daemon.service; enabled; vendor p
   Active: inactive (dead) since lun 2017-07-10 16:35:57 ART; 2h 52min ago
  Process: 1234 ExecStart=/usr/bin/ckb-daemon (code=exited, status=0/SUCCESS)
 Main PID: 1234 (code=exited, status=0/SUCCESS)

jul 10 16:35:57 miguel-pc systemd[1]: ckb-daemon.service: Service hold-off time 
jul 10 16:35:57 miguel-pc systemd[1]: Stopped Corsair Keyboard Daemon.
jul 10 16:35:57 miguel-pc systemd[1]: ckb-daemon.service: Start request repeated
jul 10 16:35:57 miguel-pc systemd[1]: Failed to start Corsair Keyboard Daemon.
```

```
systemd 229
+PAM +AUDIT +SELINUX +IMA +APPARMOR +SMACK +SYSVINIT +UTMP +LIBCRYPTSETUP +GCRYPT +GNUTLS +ACL +XZ -LZ4 +SECCOMP +BLKID +ELFUTILS +KMOD -IDN
```

MiGu3X, 2017-07-10:
Updated
[ckb-dev-detect-report.gz](https://github.com/mattanger/ckb-next/files/1137337/ckb-dev-detect-report.gz)

tatokis, 2017-07-11:
`Ιούλ 11 00:54:00.330106 darkdog-desktop ckb-daemon[886]: [E] os_setupusb (usb_linux.c:573): Unable to read endpoint count from udev, assuming 0 and reading >><< or device is in BIOS mode`
That doesn't sound good, it should be something like
`Ιούλ 11 08:04:53.428185 tatokis-PC ckb-daemon[11292]: [I] claiming interfaces. name=Corsair K70 RGB Gaming Keyboard , firmware=0205; Got >> 3<< as ep_str`.

I think we may have come across a regression. I'll see if I can find anything out.
@gkatsiapis Can you check if the mouse by itself is recognised (without the keyboard plugged in).

gkatsiapis, 2017-07-11:
@tatokis uppon boot, mouse is recognized on login screen BUT since I log it does not. It shows on lsusb command but it's not working.

My keyboard is set on Poll Rate 1 and both mouse and keyboard are working under win 10 :(

light2yellow, 2017-07-11:
@MiGu3X 
Please, run this uninstaller: https://github.com/mattanger/ckb-next/blob/cmake/linux/uninstall.sh
You can just press "Raw" in the top right corner, copy it, save somewhere, `chmod +x` it and launch.

@gkatsiapis 
Your issue is different from the one @MiGu3X has, I'm moving the discussion to #254.

gkatsiapis, 2017-07-11:
@light2yellow ok if I can help with any kind of troubleshoot let me know.

MiGu3X, 2017-07-11:
Why would I run the uninstaller? Yesterday my keyboard was detected and
worked correctly without any issues.

light2yellow, 2017-07-11:
> Please, run this uninstaller: https://github.com/mattanger/ckb-next/blob/cmake/linux/uninstall.sh
You can just press "Raw" in the top right corner, copy it, save somewhere, `chmod +x` it and launch. Then install ckb-next once again. I suppose that you were using ckb before ckb-next and the daemons are interfering.

I wrote why. Please, use GitHub's web interface notifications so that you read the final version of the message.

MiGu3X, 2017-07-11:
Oops, thanks

MiGu3X, 2017-07-11:
"I suppose that you were using ckb before ckb-next and the daemons are interfering." 

Yes, I had old ckb and installed ckb-next overwriting it, that's probably the issue. I already executed uninstall, and I'm currently re-installing. 

Thanks! Will update if everything works right.

MiGu3X, 2017-07-12:
After doing what @light2yellow recommended it still does't work. 
I made a script that made it work which I have to run everytime I turn on the pc (with on-screen keyboard) which is the following:

```bash
#!/bin/bash
sudo killall ckb-daemon
sudo ckb-daemon
```

after running that and reconnecting the device, it works

light2yellow, 2017-07-12:
Post the logs once again (ckb-dev-detect).

MiGu3X, 2017-07-12:
Updated
[ckb-dev-detect-report.gz](https://github.com/mattanger/ckb-next/files/1143595/ckb-dev-detect-report.gz)

light2yellow, 2017-07-13:
First of all, on this issue.
Seems like now the daemons at least don't interfere and `ckb-daemon` is normally launched. However, are you still using the script you posted above? If yes, you shouldn't, instead you should use `systemctl` to control systemd units.

Stop using the script (permanently disable it, this is important) and execute:

```console
sudo systemctl stop ckb-daemon
sudo systemctl disable ckb-daemon
sudo systemctl daemon-reload
sudo systemctl enable ckb-daemon
```

Then reboot (also important) and send the logs once again.

Now, other news. Seems like you have an issue similar to #254. I am positive that it doesn't have anything to do with systemd going wild, so if you have something to say on this particular problem, leave your response in #254.

```console
Jul 12 19:15:11.499235 miguel-pc ckb-daemon[855]: [I] Connecting  at /dev/input/ckb1
Jul 12 19:15:11.499308 miguel-pc ckb-daemon[855]: [I] claiming interfaces. name=(null), firmware=0205; Got >><< as ep_str
Jul 12 19:15:11.499361 miguel-pc ckb-daemon[855]: [E] os_setupusb (usb_linux.c:573): Unable to read endpoint count from udev, assuming 0 and reading >><< or device is in BIOS mode
Jul 12 19:15:11.499413 miguel-pc ckb-daemon[855]: [I] Attempting reset...
Jul 12 19:15:11.859609 miguel-pc ckb-daemon[855]: [I] claiming 0 endpoints
```

MiGu3X, 2017-07-13:
Updated
[ckb-dev-detect-report.gz](https://github.com/mattanger/ckb-next/files/1146948/ckb-dev-detect-report.gz)

light2yellow, 2017-07-14:
@MiGu3X

Alright, we're done with daemons. As was suggested, the problem indeed occured due to multiple installations.

I'm closing this issue and merging it with #254 regarding the other problem. Try also installing an older version (the 0.2.8 release, for example) as was proposed there. Yeah, and don't forget to remove current one first.

closed 2017-07-14 by light2yellow

MiGu3X, 2017-07-14:
I did evertyhing, reinstalled and all and still I have to disconnect-reconnect my keyboard for it to work (using USB 3.0 and reconnecting the keyboard cable)

light2yellow, 2017-07-14:
@MiGu3X 

I assume you have seen [Corsair RGB Keyboard USB Connection Requirements](http://forum.corsair.com/v3/showthread.php?t=132322).