mattanger/ckb-next #248: K95 Platinum Shift and Ctrl key bindings locked together

kac487, 2017-07-04:
I have been trying to use my K95P with CKB but was unable to get the keyboard to respond after starting the ckb-daemon on the master branch. After reading some of the other issues, i switched over to the newdev branch. This allows the keyboard to work with CKB running, but my "Shift" and "Ctrl" keys are bound together somehow. Even when I re-assign the "Shift" key, the "Ctrl" key gets re-assigned to the same assignment (and vice-versa). I am running on Ubuntu 16.04

Here's the dev report and an image of the problem.
[ckb-dev-detect-report.gz](https://github.com/mattanger/ckb-next/files/1122738/ckb-dev-detect-report.gz)

[image]

Any help is greatly appreciated.

tatokis, 2017-07-09:
Can you control the lighting on each key of the Ctrl keys, or is lighting on those keys bound together too?

coljac, 2017-07-11:
I'll put this here in case it's a duplicate. With a regular K95 on Mac OS, using v0.2.8, I can't bind either the left shift or caps lock key without binding the other. As soon as I select one, the other becomes also bound, and unassigning does the same.

kac487, 2017-07-12:
I managed to resolve this issue by clearing the ckb.conf file under my home directory. I have been trying to reproduce the problem but haven't had any luck.

tatokis, 2017-07-12:
Could you please upload the old one?
Which version of Qt are you using?

closed 2017-09-05 by tatokis