mattanger/ckb-next #249: Scimitar Pro [New Support Request]

sum01, 2017-07-05:
# New device support request

- [x] Yes
- [ ] No

[ckb-dev-detect-report.gz](https://github.com/mattanger/ckb-next/files/1125788/ckb-dev-detect-report.gz)
---
## General information

ckb gui launches fine, just doesn't find the Scimitar Pro.

### Source
Used `ckb-next-git` from the AUR -- version `1:0.2.8.r99.g1331253-1` as of this issue post.

I did try using this fork https://github.com/mattanger/ckb-next/pull/220 as I saw it might support the Scimitar Pro, but it didn't detect it either.

### Branch
`ckb-next-git` builds from the latest commit on master branch.

light2yellow, 2017-07-05:
Scimitar PRO preliminary support arrived in #68.

Judging by logs - check that the daemon is actually running: https://aur.archlinux.org/cgit/aur.git/tree/ckb-next-git.install?h=ckb-next-git

SethBarberee, 2017-07-05:
I have a Scimitar Pro and it works just fine. Maybe there's a configuration error?

tatokis, 2017-07-05:
```
Daemon on current boot:
-- No entries --

Daemon on last boot:
-- No entries --
```
It's clear the daemon was never launched, and wouldn't be surprised if the GUI said "Driver inactive"

sum01, 2017-07-05:
Yeah that fixed it, although it it prompts for a driver update and fails. Shows `1.09` current, upgrade to `2.03`.

SethBarberee, 2017-07-06:
> Shows 1.09 current, upgrade to 2.03 

That's what #220 solves. Scimitar Pro latest firmware is 1.10 according to CUE. Currenntly, Scimitar Pro is treated as a normal Scimitar, firmware wise but with the PR, each device will have a separate firmware which should solve that issue.

sum01, 2017-07-06:
Oh, alright. I'll close this then, thanks 👍

closed 2017-07-06 by sum01