mattanger/ckb-next #250: QuickInstall script throws an error

kunalkapoor, 2017-07-07:
Quickinstall script fails for my on Mac Sierra 10.12.3. Not sure where the failure is so I pasted a verbose log below.

# Devices information

ckb-dev-report info:

```
$> cat ckb-dev-detect-report

ckb-dev-detect - Fri Jul  7 13:21:07 CDT 2017

Kernel: Darwin kunalk-m01 16.4.0 Darwin Kernel Version 16.4.0: Thu Dec 22 22:53:21 PST 2016; root:xnu-3789.41.3~3/RELEASE_X86_64 x86_64

OS Version:
ProductName:	Mac OS X
ProductVersion:	10.12.3
BuildVersion:	16D32

USB Devices:
            Corsair Gaming K70 LUX RGB Keyboard :

              Product ID: 0x1b33
              Vendor ID: 0x1b1c  (CORSAIR MEMORY INC.)
              Version: 2.05
              Serial Number: 1403D025AEBD15015627CF43F5001943
              Speed: Up to 12 Mb/sec
              Manufacturer: Corsair
              Location ID: 0x14420000 / 4
              Current Available (mA): 500
              Current Required (mA): 500
              Extra Operating Current (mA): 0
```

---

# Steps

Git cloned ckb repo "master" branch. Ran the quickinstall script.

```
clangclang: error: clang: error: : invalid version number in '-mmacosx-version-min='clangclanginvalid version number in '-mmacosx-version-min='error: 
: : error: error
invalid version number in '-mmacosx-version-min='invalid version number in '-mmacosx-version-min='

: invalid version number in '-mmacosx-version-min='
make[2]: *** [debug/main.o] Error 1
make[2]: *** [debug/main.o] Error 1make[2]: *** [debug/main.o] Error 1make[2]: *** [debug/main.o] Error 1

make[2]: *** [debug/main.o] Error 1

make[1]: *** [debug] Error 2
make[1]: *** [debug] Error 2
make[1]: *** [debug] Error 2
make[1]: *** [debug] Error 2
make: *** [sub-src-ckb-pinwheel-make_first] Error 2
make: *** Waiting for unfinished jobs....
make: *** [sub-src-ckb-random-make_first] Error 2
make: *** [sub-src-ckb-wave-make_first] Error 2
make[1]: *** [debug] Error 2
make: *** [sub-src-ckb-ripple-make_first] Error 2
make: *** [sub-src-ckb-gradient-make_first] Error 2
clang: error: invalid version number in '-mmacosx-version-min='
make[2]: *** [debug/main.o] Error 1
make[1]: *** [debug] Error 2
make: *** [sub-src-ckb-rain-make_first] Error 2
clang: error: invalid version number in '-mmacosx-version-min='
make[2]: *** [debug/device.o] Error 1
make[2]: *** Waiting for unfinished jobs....
clang: error: invalid version number in '-mmacosx-version-min='clang: error: invalid version number in '-mmacosx-version-min='
clang: error: 
[... 518 lines cut ...]
In file included from ./kbbind.h:10:
./keyaction.h:185:44: warning: unused parameter 'rhs' [-Wunused-parameter]
    inline void operator=(const KeyAction& rhs) {}
                                           ^
./keyaction.h:186:39: warning: unused parameter 'rhs' [-Wunused-parameter]
    inline KeyAction(const KeyAction& rhs) : QObject() {}
                                      ^
2 warnings generated.
2 warnings generated.
In file included from release/moc_kbmanager.cpp:9:
In file included from release/../kbmanager.h:8:
In file included from ./kb.h:7:
In file included from ./kbprofile.h:9:
In file included from ./kbmode.h:6:
In file included from ./kbbind.h:10:
./keyaction.h:185:44: warning: unused parameter 'rhs' [-Wunused-parameter]
    inline void operator=(const KeyAction& rhs) {}
                                           ^
./keyaction.h:186:39: warning: unused parameter 'rhs' [-Wunused-parameter]
    inline KeyAction(const KeyAction& rhs) : QObject() {}
                                      ^
2 warnings generated.
2 warnings generated.

Press enter to exit.```

ashtonium, 2017-07-08:
I also have a similar problem, but only two errors

light2yellow, 2017-07-08:
Duplicate of #232.

closed 2017-07-08 by light2yellow