mattanger/ckb-next #254: The daemon claims 0 endpoints

light2yellow, 2017-07-11:
Discussion moved from #247.

## @gkatsiapis:

Uppon boot (and login screen) neither my K70 LUX RGB is recognised, nor my M65 Pro RGB. Same with adding cmdline parameters.

If I unplug my keyboard and re-plug it after 30+ seconds it gets detected but this is not the case with my mouse (though it randomly works sometimes).

[ckb-dev-detect-report (6).gz](https://github.com/mattanger/ckb-next/files/1138341/ckb-dev-detect-report.6.gz)

## @tatokis:

`Ιούλ 11 00:54:00.330106 darkdog-desktop ckb-daemon[886]: [E] os_setupusb (usb_linux.c:573): Unable to read endpoint count from udev, assuming 0 and reading >><< or device is in BIOS mode`
That doesn't sound good, it should be something like
`Ιούλ 11 08:04:53.428185 tatokis-PC ckb-daemon[11292]: [I] claiming interfaces. name=Corsair K70 RGB Gaming Keyboard , firmware=0205; Got >> 3<< as ep_str`.

I think we may have come across a regression. I'll see if I can find anything out.
@gkatsiapis Can you check if the mouse by itself is recognised (without the keyboard plugged in).
Also, does your keyboard happen to be in BIOS mode? If so, switch it to one of the numerical poll rate options.

## @gkatsiapis:

@tatokis uppon boot, mouse is recognized on login screen BUT since I log it does not. It shows on lsusb command but it's not working.

My keyboard is set on Poll Rate 1 and both mouse and keyboard are working under win 10 :(

gkatsiapis, 2017-07-11:
Did some trial and error, connected both K70 RGB Lux USB cables on usb 3.0 ports and the M65 Pro RGB on a USB 2.0 port. After ~15 secs, M65 was recognized and in ~30 or more secs K70 was recognised too (these after boot and login and after I unplugged and re-plugged them).

Still no sign of them working uppon boot but nontheless here is the log:

[ckb-dev-detect-report.gz](https://github.com/mattanger/ckb-next/files/1140184/ckb-dev-detect-report.gz)

frickler24, 2017-07-11:
@gkatsiapis I don't know why, but you have two important error lines in your log, both with the text "usb_submit_urb(ctrl) failed: -1". In the second part of the log, it tries 5 times  to reset the usb driver and then closes an reopens the devices successfully.
I'll need some time having a look into the sources.

gkatsiapis, 2017-07-12:
@frickler24

Thanks for your effort, if there is something I can do to help I'd be glad to.

light2yellow, 2017-07-13:
@MiGu3X seem to have the same or similar problem:

```console
Jul 12 19:15:11.499235 miguel-pc ckb-daemon[855]: [I] Connecting  at /dev/input/ckb1
Jul 12 19:15:11.499308 miguel-pc ckb-daemon[855]: [I] claiming interfaces. name=(null), firmware=0205; Got >><< as ep_str
Jul 12 19:15:11.499361 miguel-pc ckb-daemon[855]: [E] os_setupusb (usb_linux.c:573): Unable to read endpoint count from udev, assuming 0 and reading >><< or device is in BIOS mode
Jul 12 19:15:11.499413 miguel-pc ckb-daemon[855]: [I] Attempting reset...
Jul 12 19:15:11.859609 miguel-pc ckb-daemon[855]: [I] claiming 0 endpoints
```

Full report at: https://github.com/mattanger/ckb-next/issues/247#issuecomment-314925716

gkatsiapis, 2017-07-13:
@light2yellow regarding your last comment on #247 , did what you asked, here is the log:"

[ckb-dev-detect-report.gz](https://github.com/mattanger/ckb-next/files/1146148/ckb-dev-detect-report.gz)

PS: No input on neither the mouse, nor the keyboard, also the keyboard has no lights on.

light2yellow, 2017-07-13:
@CharmsyPone from #94 reports (almost) correct behaviour of ckb-next with M65 Pro RGB. K70 LUX RGB should perform well too.

For me personally your kernel ring buffer looks suspicious:

```
Dmesg on current boot:
Jul 13 20:52:00.505513 darkdog-desktop kernel: usb 3-14: New USB device found, idVendor=1b1c, idProduct=1b2e
*end of the log*
```

Don't know about the mouse, but my keyboard usually produces 5-6 messages. Also, K70 LUX RGB is not even shown among USB devices. If you had it plugged in while creating a report, it might be that the USB port is dead/underpowered. For both devices.

I'm sure @tatokis could tell more.

Also, as this is the second case in a few days, could you try another, older version to see if it produces the same issues? For example, the latest release https://github.com/mattanger/ckb-next/releases/tag/v0.2.8 .
Before you install it, please run the uninstaller I posted on #247. Then proceed with the installation as usually. After that the outputs of `sudo systemctl status ckb-daemon` and `journalctl -o short-precise -k -b -0 | grep -i "ckb|corsair|1b1c"` would be nice. IIRC at that time the report didn't show daemon's logs, so you will have to execute the commands manually.

> regarding your last comment on #247 , did what you asked

Actually, I was replying to @MiGu3X on his systemd-related issue, but nevermind :)

gkatsiapis, 2017-07-13:
Don't know if this should be the proper output of uninstall.sh (plus my keyboard still works while I just uninstalled everything):

```
darkdog@darkdog-desktop:~/ckb-next-master$ sudo ./uninstall.sh
Stopping GUI
The ckb GUI should not be run as root.
./uninstall.sh: line 5: ckb-next: command not found
Stopping the daemon and removing service file
systemd detected
Failed to stop ckb-next-daemon.service: Unit ckb-next-daemon.service not loaded.
Failed to execute operation: No such file or directory
Removing GUI and daemon
Removing support files
Success!
```

Output of sudo systemctl status ckb-daemon:

```root@darkdog-desktop:~/ckb-next-0.2.8# sudo systemctl status ckb-daemon
● ckb-daemon.service - Corsair Keyboard Daemon
   Loaded: loaded (/usr/lib/systemd/system/ckb-daemon.service; enabled; vendor p
   Active: active (running) since Παρ 2017-07-14 01:24:40 EEST; 39s ago
 Main PID: 4081 (ckb-daemon)
   CGroup: /system.slice/ckb-daemon.service
           └─4081 /usr/bin/ckb-daemon

Ιούλ 14 01:24:40 darkdog-desktop ckb-daemon[4081]:     ckb: Corsair RGB driver v
Ιούλ 14 01:24:40 darkdog-desktop ckb-daemon[4081]: [I] Root controller ready at 
Ιούλ 14 01:24:40 darkdog-desktop ckb-daemon[4081]: [I] Connecting Corsair Gaming
Ιούλ 14 01:24:40 darkdog-desktop ckb-daemon[4081]: [I] Connecting  at /dev/input
Ιούλ 14 01:24:40 darkdog-desktop ckb-daemon[4081]: [W] os_setupusb (usb_linux.c:
Ιούλ 14 01:24:40 darkdog-desktop ckb-daemon[4081]: [E] os_setupusb (usb_linux.c:
Ιούλ 14 01:24:40 darkdog-desktop ckb-daemon[4081]: [I] Disconnecting /dev/input/
Ιούλ 14 01:24:40 darkdog-desktop ckb-daemon[4081]: [I] Removed device path /dev/
Ιούλ 14 01:24:40 darkdog-desktop ckb-daemon[4081]: [I] Starting input thread for
Ιούλ 14 01:24:42 darkdog-desktop ckb-daemon[4081]: [I] Setup finished for /dev/i
lines 1-17/17 (END)...skipping...
● ckb-daemon.service - Corsair Keyboard Daemon
   Loaded: loaded (/usr/lib/systemd/system/ckb-daemon.service; enabled; vendor p
   Active: active (running) since Παρ 2017-07-14 01:24:40 EEST; 39s ago
 Main PID: 4081 (ckb-daemon)
   CGroup: /system.slice/ckb-daemon.service
           └─4081 /usr/bin/ckb-daemon

Ιούλ 14 01:24:40 darkdog-desktop ckb-daemon[4081]:     ckb: Corsair RGB driver v
Ιούλ 14 01:24:40 darkdog-desktop ckb-daemon[4081]: [I] Root controller ready at 
Ιούλ 14 01:24:40 darkdog-desktop ckb-daemon[4081]: [I] Connecting Corsair Gaming
Ιούλ 14 01:24:40 darkdog-desktop ckb-daemon[4081]: [I] Connecting  at /dev/input
Ιούλ 14 01:24:40 darkdog-desktop ckb-daemon[4081]: [W] os_setupusb (usb_linux.c:
Ιούλ 14 01:24:40 darkdog-desktop ckb-daemon[4081]: [E] os_setupusb (usb_linux.c:
Ιούλ 14 01:24:40 darkdog-desktop ckb-daemon[4081]: [I] Disconnecting /dev/input/
Ιούλ 14 01:24:40 darkdog-desktop ckb-daemon[4081]: [I] Removed device path /dev/
Ιούλ 14 01:24:40 darkdog-desktop ckb-daemon[4081]: [I] Starting input thread for
Ιούλ 14 01:24:42 darkdog-desktop ckb-daemon[4081]: [I] Setup finished for /dev/i
~
~
~
~
~
~
~
lines 1-17/17 (END)...skipping...
● ckb-daemon.service - Corsair Keyboard Daemon
   Loaded: loaded (/usr/lib/systemd/system/ckb-daemon.service; enabled; vendor preset: enabled)
   Active: active (running) since Παρ 2017-07-14 01:24:40 EEST; 39s ago
 Main PID: 4081 (ckb-daemon)
   CGroup: /system.slice/ckb-daemon.service
           └─4081 /usr/bin/ckb-daemon

Ιούλ 14 01:24:40 darkdog-desktop ckb-daemon[4081]:     ckb: Corsair RGB driver v0.2.8
Ιούλ 14 01:24:40 darkdog-desktop ckb-daemon[4081]: [I] Root controller ready at /dev/input/ckb0
Ιούλ 14 01:24:40 darkdog-desktop ckb-daemon[4081]: [I] Connecting Corsair Gaming K70 LUX RGB Keyboard at /dev/input/ckb1
Ιούλ 14 01:24:40 darkdog-desktop ckb-daemon[4081]: [I] Connecting  at /dev/input/ckb2
Ιούλ 14 01:24:40 darkdog-desktop ckb-daemon[4081]: [W] os_setupusb (usb_linux.c:299): Unable to read endpoint count from udev, assuming 4...
Ιούλ 14 01:24:40 darkdog-desktop ckb-daemon[4081]: [E] os_setupusb (usb_linux.c:302): Failed to claim interfaces: No such file or directory
Ιούλ 14 01:24:40 darkdog-desktop ckb-daemon[4081]: [I] Disconnecting /dev/input/ckb2
Ιούλ 14 01:24:40 darkdog-desktop ckb-daemon[4081]: [I] Removed device path /dev/input/ckb2
Ιούλ 14 01:24:40 darkdog-desktop ckb-daemon[4081]: [I] Starting input thread for /dev/input/ckb1
Ιούλ 14 01:24:42 darkdog-desktop ckb-daemon[4081]: [I] Setup finished for /dev/input/ckb1
~
~
~
~
~
~
~
~
~
~
~
~
~
~
~
~
~
~
~
~
~
~
~
~
~
~
~
~
~
~
~
~
~
~
~
~
~
~
~
~
~
~
lines 1-17/17 (END)
```
I had no output for journalctl -o short-precise -k -b -0 | grep -i "ckb|corsair|1b1c" 

PS1: Strange behavior since on each boot either the keyboard OR mouse is working. Not both of them at the same time.

PS2: regarding tha underpowered or dead USB ports, I have tried all of them, plus the are working properly on windows.

light2yellow, 2017-07-14:
1. Don't run the uninstaller with sudo. Type in the password when it asks. Otherwise it will fail to close the GUI. Also, yes, it is intended that it'll burst a bunch of errors about missing files, because it tries to remove ckb from every corner (which it really shouldn't because of packages, but that's another story).

2. Make the log readable. Do `systemctl ... > log` or whatever.

3. Can't really test the other command myself, try `dmesg | grep -i "ckb|corsair|1b1c"` or `dmesg | grep -i "ckb\|corsair\|1b1c"`. And yes, forward the output to some file as well.

gkatsiapis, 2017-07-14:
@light2yellow excuse me for those errors, I have followed your instructions (hopefuly correct this time) and here is the logs after the installation:
[logs_1.tar.gz](https://github.com/mattanger/ckb-next/files/1148220/logs_1.tar.gz)

The keyboard was still not detected, however I unplugged it and re-plugged it again and took logs:

[logs_2.tar.gz](https://github.com/mattanger/ckb-next/files/1148222/logs_2.tar.gz)

light2yellow, 2017-07-14:
@gkatsiapis

Yep, now the logs look well, thank you for the patience. So, release version (0.2.8) works, but current one doesn't_.

This problem:
```
[44600.540102] hid-generic 0003:1B1C:1B2E.0008: usb_submit_urb(ctrl) failed: -1
```
is cured by adding quirks, if I recall correctly. They will improve your devices' boot time responsiveness as well. They are always in the form `VID:PID:0x20000408`, so, as we can see from the logs, for M65 Pro RGB it's `1b1c:1b2e:0x20000408`, for K70 LUX RGB - `1b1c:1b33:0x20000408`. See https://github.com/mattanger/ckb-next#linux for details.

If somebody has hardware (I lack it right now) and time to debug this, remember about the existence of `PTHREAD_MUTEX_ERRORCHECK` attribute so that deadlocks can be easily detected. Also, `helgrind` and `TSan`. It might be the newly introduced macro patch & races/deadlocks. Or it might be something else, trivial.

light2yellow, 2017-07-14:
Forgot to say: if @gkatsiapis or @MiGu3X or someone else experiencing this issue would [bisect](https://stackoverflow.com/questions/4713088/how-to-use-git-bisect) the source, it would be great. JFYI: to bisect once but for sure, `killall -9 ckb`, stop the daemon, daemon-reload systemd, (read [ArchWiki](https://wiki.archlinux.org/index.php/Systemd) if you don't know systemd commands) and on every subsequent recompilation launch the daemon right from the directory where it is compiled, e.g. `./bin/ckb-daemon` or whatever. To terminate the daemon, use Ctrl+C.

gkatsiapis, 2017-07-14:
@light2yellow actually 0.2.8 did work for once randomly.

After several reboots again both keyboard & mouse do not work (even with the quirks, on 0x20000408 or 0x4).

Logs:
[logs.tar.gz](https://github.com/mattanger/ckb-next/files/1148704/logs.tar.gz)

This is frustrating a bit because there is no stable state (for example keyboard and mouse are not working if I unplug/re-plug them) and I'm not sure if the problem has something to do with my setup, a BIOS setting or something else.

light2yellow, 2017-07-14:
@gkatsiapis 
Can you test the devices on Windows, with CUE? Ideally on the same PC.

gkatsiapis, 2017-07-14:
@light2yellow yes they are working properly on win10

gkatsiapis, 2017-07-14:
Ok some more trial and error:

In the common situation of booting and have no input (none of the keyboard, nor the mouse are working)..
If I unplug them and replug them both at the same time, they are both not recognized. However If I plug only the mouse and wait for a while, it gets recognized after ~20-30 secs. Once the mouse gets recognized, if I plug the keyboard, it also gets recognized after ~30 secs.

My point is that they get recognized one at each time so maybe they need some kind of delay in getting connected with ckb? Or am I talking crap? :/

light2yellow, 2017-07-29:
Today I also spotted such problems (newdev branch).

```
$ pacman -Qi ckb-next-latest-git
Name            : ckb-next-latest-git
Version         : 1:0.2.7.r184.gf0e93aa-1
```

```
Jul 29 14:25:35.549444 manjaro-workstation systemd[1]: Started Corsair Keyboard Daemon.
Jul 29 14:25:35.567092 manjaro-workstation ckb-daemon[329]:     ckb: Corsair RGB driver beta-v0.2.8
Jul 29 14:25:35.567092 manjaro-workstation ckb-daemon[329]: [I] Root controller ready at /dev/input/ckb0
Jul 29 14:25:51.301996 manjaro-workstation ckb-daemon[329]: [I] >>>vendor = 0x1b1c, product = 0x1b20, path = /dev/bus/usb/003/004, syspath = /sys/devices/pci0000:00/0000:00:14.0/usb3/3-4
Jul 29 14:25:51.402144 manjaro-workstation ckb-daemon[329]: [I] Connecting  at /dev/input/ckb1
Jul 29 14:25:51.402144 manjaro-workstation ckb-daemon[329]: [I] claiming interfaces. name=(null), firmware=0205; Got >><< as ep_str
Jul 29 14:25:51.402144 manjaro-workstation ckb-daemon[329]: [E] os_setupusb (usb_linux.c:569): Unable to read endpoint count from udev, assuming 0 and reading >><< or device is in BIOS mode
Jul 29 14:25:51.402144 manjaro-workstation ckb-daemon[329]: [I] Attempting reset...
Jul 29 14:25:51.745472 manjaro-workstation ckb-daemon[329]: [I] claiming 0 endpoints
Jul 29 14:25:51.855669 manjaro-workstation ckb-daemon[329]: [E] os_usbsend (via firmware.c:15): No route to host
Jul 29 14:25:51.855669 manjaro-workstation ckb-daemon[329]: [W] _start_dev (device.c:38): Unable to load firmware version/poll rate
Jul 29 14:25:51.855669 manjaro-workstation ckb-daemon[329]: [I] Reset success
Jul 29 14:25:51.855669 manjaro-workstation ckb-daemon[329]: [I] Connecting  at /dev/input/ckb1
Jul 29 14:25:51.855669 manjaro-workstation ckb-daemon[329]: [I] claiming interfaces. name=(null), firmware=0205; Got >><< as ep_str
```

zepo87, 2017-12-17:
Got exactly the same problem as mentioned @gkatsiapis. 
Same setup: K70 LUX RGB + M65 Pro RGB.
```lsusb``` detects both devices everytime. 
Sometimes on boot either work, if they don't, after unplug and plug the usb ( I have to do this this random times when i get the error ) they do.
I have tried all the combinations with usb quircks, none fixed the problem.
Working perfect on Windows 10.
Thanks for your help.
[ckb-dev-detect-report.gz](https://github.com/mattanger/ckb-next/files/1566053/ckb-dev-detect-report.gz)

tatokis, 2018-01-22:
https://github.com/ckb-next/ckb-next/issues/26

closed 2018-01-22 by tatokis