mattanger/ckb-next #256: CKB Says No Devices Connected

hypendex, 2017-07-14:
I am on a Mac using OS Sierra. I've tried all the suggested solutions for this problem, however CKB keeps saying "No Devices Connected". I am using the K70 RGB LUX. The keyboard works on my Mac and the key bindings, like the command key, all work. I just cannot get to the screen to change all the color settings.

hypendex, 2017-07-15:
I tried building from source using quickinstall from ckb-next-0.2.8 after downloading Xcode and QT. But when i try to quickinstall it keeps coming up with an error. 
2 warnings generated.
In file included from release/moc_kbmanager.cpp:9:
In file included from release/../kbmanager.h:8:
In file included from ./kb.h:7:
In file included from ./kbprofile.h:9:
In file included from ./kbmode.h:6:
In file included from ./kbbind.h:10:
./keyaction.h:165:44: warning: unused parameter 'rhs' [-Wunused-parameter]
    inline void operator=(const KeyAction& rhs) {}
                                           ^
./keyaction.h:166:39: warning: unused parameter 'rhs' [-Wunused-parameter]
    inline KeyAction(const KeyAction& rhs) : QObject() {}
                                      ^
2 warnings generated.

light2yellow, 2017-07-15:
Instead of `./quickinstall`, execute:

```
rm -rf .qmake.stash
./qmake-auto
make
```

You should see an error somewhere around `--mmacos--version-min`. You've posted just warnings, no errors.

danmac, 2017-07-31:
I am also having this issue on the High Sierra Beta. Sometimes it works by unplugging and replugging the keyboard. Most of the time I have to restart the computer.

light2yellow, 2017-07-31:
The workaround was provided to be able to build from source. Re-read the section of README that talks about manual compilation on macOS.

light2yellow, 2017-08-20:
You can try a new macOS package from #281. If you will still have issues, open a new thread and post the output of `/var/log/ckb-next-daemon.log`.

closed 2017-08-20 by light2yellow