mattanger/ckb-next #257: CKB Not Detecting Scimitar Pro

mgw1214, 2017-07-20:
I'll start off by saying I'm not good with computers at all. That being said, I followed the instructions for installing CKB, but it doesn't seem to be detecting my Scimitar Pro. I'm not sure if that mouse isn't supported, or if it's similar enough to the original Scimitar where it should be working, but I did something wrong or am missing something.

Any help would be appreciated! Thanks!

SethBarberee, 2017-07-20:
@mgw1214 
Can you give more info about your system? OS? Installation method? (There is an Issue Template for a reason ;)

When you installed CKB-next, there should be a script called `ckb-dev-detect`. Can you run that and post its output?

mgw1214, 2017-07-20:
I apologize for not using the template.

I am running macOS Sierra. I installed CKB-next through the quickinstall script, which generated two warnings but no errors it seems. Here's the output from the script:

OS Version:
ProductName:	Mac OS X
ProductVersion:	10.12.2
BuildVersion:	16C67

USB Devices:
        Corsair Gaming SCIMITAR PRO RGB Mouse:

          Product ID: 0x1b3e
          Vendor ID: 0x1b1c  (CORSAIR MEMORY INC.)
          Version: 1.09
          Serial Number:
          Speed: Up to 12 Mb/sec
          Manufacturer: Corsair
          Location ID: 0x14200000 / 8
          Current Available (mA): 500
          Current Required (mA): 300
          Extra Operating Current (mA): 0

SethBarberee, 2017-07-20:
> two warnings
What warnings? They might be important

mgw1214, 2017-07-20:
2 warnings generated.
In file included from settingswidget.cpp:5:
In file included from ./kb.h:7:
In file included from ./kbprofile.h:9:
In file included from ./kbmode.h:6:
In file included from ./kbbind.h:10:
./keyaction.h:185:44: warning: unused parameter 'rhs' [-Wunused-parameter]
    inline void operator=(const KeyAction& rhs) {}
                                           ^
./keyaction.h:186:39: warning: unused parameter 'rhs' [-Wunused-parameter]
    inline KeyAction(const KeyAction& rhs) : QObject() {}
                                      ^
settingswidget.cpp:143:47: warning: unused parameter 'index' [-Wunused-parameter]
void SettingsWidget::on_capsBox_activated(int index){
                                              ^
settingswidget.cpp:147:48: warning: unused parameter 'index' [-Wunused-parameter]
void SettingsWidget::on_shiftBox_activated(int index){
                                               ^
settingswidget.cpp:151:47: warning: unused parameter 'index' [-Wunused-parameter]
void SettingsWidget::on_ctrlBox_activated(int index){
                                              ^
settingswidget.cpp:155:46: warning: unused parameter 'index' [-Wunused-parameter]
void SettingsWidget::on_altBox_activated(int index){
                                             ^
settingswidget.cpp:159:46: warning: unused parameter 'index' [-Wunused-parameter]
void SettingsWidget::on_winBox_activated(int index){
                                             ^
3 warnings generated.
3 warnings generated.
7 warnings generated.
In file included from kblight.cpp:5:
In file included from ./kbmode.h:6:
In file included from ./kbbind.h:10:
./keyaction.h:185:44: warning: unused parameter 'rhs' [-Wunused-parameter]
    inline void operator=(const KeyAction& rhs) {}
                                           ^
./keyaction.h:186:39: warning: unused parameter 'rhs' [-Wunused-parameter]
    inline KeyAction(const KeyAction& rhs) : QObject() {}
                                      ^
In file included from kbprofile.cpp:1:
In file included from ./kbprofile.h:9:
In file included from ./kbmode.h:6:
In file included from ./kbbind.h:10:
./keyaction.h:185:44: warning: unused parameter 'rhs' [-Wunused-parameter]
    inline void operator=(const KeyAction& rhs) {}
                                           ^
./keyaction.h:186:39: warning: unused parameter 'rhs' [-Wunused-parameter]
    inline KeyAction(const KeyAction& rhs) : QObject() {}
                                      ^
2 warnings generated.
kbanimwidget.cpp:151:94: warning: unused parameter 'previous' [-Wunused-parameter]
void KbAnimWidget::on_animList_currentItemChanged(QListWidgetItem *current, QListWidgetItem *previous){
                                                                                             ^
2 warnings generated.
kbanim.cpp:333:32: warning: unused parameter 'bg' [-Wunused-parameter]
static float blendNormal(float bg, float fg){
                               ^
1 warning generated.
animadddialog.cpp:28:46: warning: unused parameter 'index' [-Wunused-parameter]
void AnimAddDialog::on_animBox_activated(int index){
                                             ^
1 warning generated.
1 warning generated.
animsettingdialog.cpp:487:57: warning: unused parameter 'arg1' [-Wunused-parameter]
void AnimSettingDialog::on_delayBox_valueChanged(double arg1){
                                                        ^
animsettingdialog.cpp:491:58: warning: unused parameter 'arg1' [-Wunused-parameter]
void AnimSettingDialog::on_repeatBox_valueChanged(double arg1){
                                                         ^
animsettingdialog.cpp:495:59: warning: unused parameter 'arg1' [-Wunused-parameter]
[... 392 lines cut ...]
In file included from ./kbmode.h:6:
In file included from ./kbbind.h:10:
./keyaction.h:185:44: warning: unused parameter 'rhs' [-Wunused-parameter]
    inline void operator=(const KeyAction& rhs) {}
                                           ^
./keyaction.h:186:39: warning: unused parameter 'rhs' [-Wunused-parameter]
    inline KeyAction(const KeyAction& rhs) : QObject() {}
                                      ^
2 warnings generated.
2 warnings generated.
In file included from release/moc_kbmanager.cpp:9:
In file included from release/../kbmanager.h:8:
In file included from ./kb.h:7:
In file included from ./kbprofile.h:9:
In file included from ./kbmode.h:6:
In file included from ./kbbind.h:10:
./keyaction.h:185:44: warning: unused parameter 'rhs' [-Wunused-parameter]
    inline void operator=(const KeyAction& rhs) {}
                                           ^
./keyaction.h:186:39: warning: unused parameter 'rhs' [-Wunused-parameter]
    inline KeyAction(const KeyAction& rhs) : QObject() {}
                                      ^
2 warnings generated.

After looking through the script again when running it, it seems to be more than 2 warnings. Sorry again for not being competent with computers, and thanks for your help so far!

SethBarberee, 2017-07-20:
If `quickinstall` failed, try [this](https://github.com/ccMSC/ckb/blob/master/BUILD.md#osx). If that fails too, then we know that it's a problem in the code which I would hope that it isn't.

mgw1214, 2017-07-20:
When trying to build through Qt Creator, I get the error found in this thread: https://github.com/mattanger/ckb-next/issues/232

I don't think there's an active solution to that problem yet. This is the version of Qt Creator I'm running:
Qt Creator 4.3.1
Based on Qt 5.9.1 (Clang 7.0 (Apple), 64 bit)

Would it help if I directly installed the offline installer of Qt 5.9.1?

light2yellow, 2017-07-20:
No, it wouldn't. There's no known way to compile the software under macOS anyhow so far. Hopefully CMake will be introduced soon.

_Sent from my LeEco Le X820 using [FastHub](https://play.google.com/store/apps/details?id=com.fastaccess.github)_

mgw1214, 2017-07-20:
Alright thank you all for the help! Unfortunately it looks like I'll be returning the Scimitar until there's a way to run it on a Mac.

tatokis, 2017-07-20:
Have you tried https://github.com/mattanger/ckb-next/releases/download/v0.2.8/ckb-v0.2.8.pkg ?
It should have support for the scimitar there

closed 2017-07-24 by light2yellow