mattanger/ckb-next #258: Scimitar RGB Left click stops working for a few seconds randomly.

Alkanna, 2017-07-21:
I am using the last pkg release on Sierra.
For some reason it looks like the left click gets stuck on hold or something for a few seconds, randomly, every 20 seconds or so. Changing the active window seems to trigger this bug more frequently.

Anyone else experiencing this ?

OscarBarrett, 2017-07-21:
When it occurs, have you tried restarting the daemon to fix it?

It sounds like the mouse is sticking a bit when you click and hold left click? If so, you may be experiencing https://github.com/mattanger/ckb-next/issues/93

Alkanna, 2017-07-21:
Restarting the daemon does not help because the issue appears and stays for a few seconds only, then goes back to normal.

Alkanna, 2017-07-23:
Any updates on this ?

OscarBarrett, 2017-07-24:
I'm using a Scimitar RGB on Sierra and have never had this issue. If you can find a way to reproduce it I'd be happy to test.

Alkanna, 2017-07-24:
Are you using the last pkg release or did you build it yourself ?
It just happens, mostly when playing World of Warcraft.

Alkanna, 2017-07-24:
The mouse is working fine when ckb isn't running or for example now that I restarted for some reason it says no devices connected, the mouse works fine. However I cannot configure any of the additional buttons on it.

Alkanna, 2017-07-24:
Alright I managed to narrow it down a bit more : What happens is that when I select a window/tab then want to quickly select another, like select a tab in chrome then another one in the next second, it doesn't click it. If I let the mouse hover on the other tab for a second or more, then click it, it works as intended.

Alkanna, 2017-07-24:
Alright I'm closing this issue, what I said above basically resolved it, but I still have a problem with the mouse not snapping to the window like it should in WoW.

closed 2017-07-24 by Alkanna