mattanger/ckb-next #259: [Ubuntu 17.x] Randomly keys don't work as expected.

varrix, 2017-07-23:
# Devices information
My `ckb-dev-detect` results: https://gist.github.com/varrix/17de04becbc0fd40b1275b4e1578ae97

---

# Bug report

- [X] Yes
- [ ] No

<sub><sup>leave empty if you have no bug reports</sup></sub>

## General information

### Source
(how and where did you get this program, e.g.: _"`ckb-next-git` package in AUR"_ or _"built manually using quickinstall script"_ or _"pkg for macOS"_)
I `git clone`d the repository somewhere in my `~`  and proceeded to `quickinstall`. It didn't recognize my keyboard for a while (even after a couple restarts/re-logs/re-plugging it in) but eventually worked after another `quickinstall` executed on-top of the current installation.
### Branch
(fill in if ckb-next was compiled manually, e.g.: _"master"_ or _"newdev"_)

## Logs & crash reports
(any useful information - an OS crash report, debugger's backtrace, any non-standard way of usage (BIOS mode ...)  etc.)

You should upload:
- Linux: `ckb-dev-detect-report.gz` is enough
- macOS:
    * the output of `ckb-daemon`:
    ```console
    $ sudo launchctl unload /Library/LaunchDaemons/com.ckb.daemon.plist
    $ sudo /Applications/ckb.app/Contents/Resources/ckb-daemon
    ```
    * the output of `ckb` (open another terminal):
    ```console
    $ sudo killall -9 ckb
    $ /Applications/ckb.app/Contents/MacOS/ckb
    ```

## What is the program's current behavior?
Currently, my keyboard will randomly go from working as expected to simply pressing particular keys will trigger OS features. So for example in gnome, for some reason after installing `ckb-next` the key `v` will go from actually displaying the letter `v` (as expected) to dropping down my notifications bar. Another example is the letter `a`. Going from displaying the letter `a` to opening my dashboard- and so on. Not entirely sure what you call those, I've never encountered this before installing `ckb-next`. I have two keyboards plugged in to ensure when `ckb-next` craps out I can still use my system, but it actually ends up applying to both keyboards and I can fix this via re-logging or restarting. Which I can then assume simply restarting the daemon will do the trick but since I can't use my keyboard(s) properly and don't have a click-to-launch script restarting it- I can't actually do so via my terminal. Next best bet is to re-log/restart.

## What is the expected behavior of the program?
Never switch between regular keystrokes and odd DE keystroke triggers.

## Any other notes & comments?
Figured I'd dump a bit (if not too much) of info on my system if it can help in any way whatsoever.
- OS: Ubuntu 17.04 x86_64 
- Model: MS-7821 1.0 
- Kernel: 4.10.0-28-generic 
- Shell: bash 4.4.7 
- DE: GNOME 
- WM: GNOME Shell 
- Terminal: gnome-terminal 
- CPU: Intel i7-4790K (8) @ 4.4GHz 
- GPU: NVIDIA GeForce GTX 970 
- Memory: 15991MiB 
---

varrix, 2017-07-23:
This was quickly written up, so please let me know if I did gloss over anything obvious, or if there's more information/clarification required.

light2yellow, 2017-07-24:
Did you try to use the keyboard without ckb-next?

light2yellow, 2017-07-24:
Maybe my question was not correct. I would like to know whether you are or aren't able to reproduce this behaviour without using any additional software. I.e. make sure the ckb-daemon is not running

```
sudo systemctl stop ckb-daemon
sudo systemctl disable ckb-daemon
```

and use the keyboard like that for some time.

varrix, 2017-07-25:
I can't use the keyboard without ckb-next. It will not function. Lights will turn on (the default red+white) denoting power consumption, but will not be recognized without ckb-next.

varrix, 2017-07-25:
It might be a matter of a right ALT key getting stuck somewhere along the line? Not physically. I never use the right ALT key, but I in firefox whilst trying to highlight some text and it started moving the entire window. Which is done via ALT+LEFT-CLICK+drag. Just trying to get to the bottom of this. Not exactly sure how I stop it either. I've been able to get it back to normal by button mashing a few things, pressing ALT didn't necessarily solve it, but I'm also not entirely sure that it didn't (ie: it gets stuck like that, then instead of relogging etc one of the keys or key combinations had reverted the keyboard back to normal state).

varrix, 2017-07-25:
Also to note @light2yellow when it comes to the keyboard itself, the messed up state would apply to the other keyboard plugged in which is an old `IBM KB-9910` so I'm not too sure about that. Despite it all, I will try your suggestion out and get back to you.

tatokis, 2017-07-31:
Any chance of using `xev` and/or `cat /dev/input/ckb1/notify1` to give more details?
The former shows what keys X receives, and the latter shows what the ckb-daemon processes.

varrix, 2017-07-31:
@tatokis 
I'm not able to locate a `/dev/input/ckb1/notify1`:
```
 ~  ⑂ master +  $  cd /dev/input/
 /dev/input  $  ll
total 0
drwxr-xr-x  5 root    540 Jul 30 18:25 ./
drwxr-xr-x 20 root   4.2K Jul 30 18:25 ../
drwxr-xr-x  2 root    140 Jul 30 18:25 by-id/
drwxr-xr-x  2 root    140 Jul 30 18:25 by-path/
drwxr-xr-x  2 root    100 Jul 30 18:25 ckb0/
crw-rw----  1 root 13, 64 Jul 30 18:25 event0
crw-rw----  1 root 13, 65 Jul 30 18:25 event1
crw-rw----  1 root 13, 74 Jul 30 18:25 event10
crw-rw----  1 root 13, 75 Jul 30 18:25 event11
crw-rw----  1 root 13, 76 Jul 30 18:25 event12
crw-rw----  1 root 13, 77 Jul 30 18:25 event13
crw-rw----  1 root 13, 78 Jul 30 18:25 event14
crw-rw----  1 root 13, 79 Jul 30 18:25 event15
crw-rw----  1 root 13, 80 Jul 30 18:25 event16
crw-rw----  1 root 13, 81 Jul 30 18:25 event17
crw-rw----  1 root 13, 82 Jul 30 18:25 event18
crw-rw----  1 root 13, 66 Jul 30 18:25 event2
crw-rw----  1 root 13, 67 Jul 30 18:25 event3
crw-rw----  1 root 13, 68 Jul 30 18:25 event4
crw-rw----  1 root 13, 69 Jul 30 18:25 event5
crw-rw----  1 root 13, 70 Jul 30 18:25 event6
crw-rw----  1 root 13, 71 Jul 30 18:25 event7
crw-rw----  1 root 13, 72 Jul 30 18:25 event8
crw-rw----  1 root 13, 73 Jul 30 18:25 event9
crw-rw----  1 root 13, 63 Jul 30 18:25 mice
crw-rw----  1 root 13, 32 Jul 30 18:25 mouse0
crw-rw----  1 root 13, 33 Jul 30 18:25 mouse1
```

light2yellow, 2017-07-31:
@varrix 

@tatokis meant `/var/run/ckb1/notify1`.

varrix, 2017-07-31:
Oh, gotcha. Not at said machine, but I'll ping ya when I get a moment to
look it over. Thanks by the way!

tatokis, 2017-07-31:
@light2yellow /var/run? That's only on mac, not linux.

light2yellow, 2017-07-31:
Yep, for some reason I thought OP uses Mac. Sorry. It should be /dev/input.

varrix, 2017-07-31:
@tatokis 
When it is running, here's the output of me pressing every single key (singularly, no combinations- seems fine):
https://gist.github.com/varrix/63620d3b6427f3ea969742ee0eada45f

Now, the problem is that it'll cut out here and there. So I'll leave my `notify1` open (from: `/dev/input/ckb1/notify1`) to see if when I encounter an issue, what combination was the cause (if that is related).

@light2yellow 
Yeah, sorry- on `Ubuntu Gnome 17.x`.

varrix, 2017-08-01:
Pretty sure the following has more useful info:
```
 ~  ⑂ master +  $  sudo service ckb-daemon status
[sudo] password for logan: 
● ckb-daemon.service - Corsair Keyboard Daemon
   Loaded: loaded (/usr/lib/systemd/system/ckb-daemon.service; enabled; vendor preset: enabled)
   Active: active (running) since Tue 2017-08-01 04:25:58 EDT; 11h ago
 Main PID: 1090 (ckb-daemon)
    Tasks: 1 (limit: 4915)
   CGroup: /system.slice/ckb-daemon.service
           └─1090 /usr/bin/ckb-daemon

Aug 01 04:26:16 vulkan ckb-daemon[1090]: [I] claiming 0 endpoints
Aug 01 04:26:16 vulkan ckb-daemon[1090]: [I] Reset success
Aug 01 04:26:16 vulkan ckb-daemon[1090]: [I] Connecting  at /dev/input/ckb1
Aug 01 04:26:16 vulkan ckb-daemon[1090]: [I] claiming interfaces. name=(null), firmware=0205; Got >>
Aug 01 04:26:16 vulkan ckb-daemon[1090]: [E] os_setupusb (usb_linux.c:573): Unable to read endpoint 
Aug 01 04:26:16 vulkan ckb-daemon[1090]: [I] Attempting reset...
Aug 01 04:26:16 vulkan ckb-daemon[1090]: [I] claiming 0 endpoints
```

varrix, 2017-08-02:
A line got cut off:
`Aug 01 04:26:16 vulkan ckb-daemon[1090]: [E] os_setupusb (usb_linux.c:573): Unable to read endpoint count from udev, assuming 0 and reading >><< or device is in BIOS mode`
I'll check to see if I'm in UEFI. Probably am. Odd that it works sometimes though.

tatokis, 2018-01-22:
https://github.com/ckb-next/ckb-next/issues/27

closed 2018-01-22 by tatokis