mattanger/ckb-next #265: Coding style

ZirconiumX, 2017-07-28:
(Irrelevant bug list is irrelevant.)

As discussed in the IRC, it might be useful to have a unified coding style - inspired by https://github.com/dolphin-emu/dolphin/blob/master/Tools/lint.sh we could set a shell script as a Git pre-commit hook.

To do *that*, we need to decide on a coding style.

It seems that ccMSC's coding style:
- uses `if(condition){` (if without spaces), similar for functions.
- functions have their starting brace on the same line as the function declaration.
- `case` is not indented after `switch`.
- no line length limit.

I can try to get a clang-format style for this, or we can standardise on a preset style that already exists and is used.

light2yellow, 2017-08-20:
I think we should work out a `.clang-format` file.

One big deal-breaker for me about clang-format is the missing `PointerAlignment: Right` option. See [this](https://github.com/llvm-mirror/clang/blob/master/lib/Format/WhitespaceManager.cpp#L444). It makes impossible to automatically format structs as in the Linux kernel:

```c
struct napi_struct {
         struct list_head        poll_list;
         unsigned long           state;
         int                     weight;
         unsigned int            gro_count;
         int                     (*poll)(struct napi_struct *, int);
         struct net_device       *dev;
         struct sk_buff          *gro_list;
         struct sk_buff          *skb;
         struct hrtimer          timer;
         struct list_head        dev_list;
         struct hlist_node       napi_hash_node;
         unsigned int            napi_id;
};
```

Other than that, I'm fine with WebKit's code style. I've skimmed through the docs describing about. Later I'll dump `.clang-format` and see what it sets up.

light2yellow, 2017-09-20:
With https://reviews.llvm.org/D27651 merged (if ever, see https://lists.llvm.org/pipermail/cfe-dev/2017-September/055539.html) `PointerAlignment: Right` would become a real thing.

light2yellow, 2017-10-12:
So, today I built `clang-format` (`6.0.0 trunk 315573`) with `D276521` applied and this is the result on the struct above:

```c
struct napi_struct {
    struct list_head poll_list;
    unsigned long    state;
    int              weight;
    unsigned int     gro_count;
    int (*poll)(struct napi_struct *, int);
    struct net_device *dev;
    struct sk_buff    *gro_list;
    struct sk_buff    *skb;
    struct hrtimer     timer;
    struct list_head   dev_list;
    struct hlist_node  napi_hash_node;
    unsigned int       napi_id;
};
```

Not-so-good I'd say, but if we move the function pointer below, it looks much better:

```c
struct napi_struct {
    struct list_head   poll_list;
    unsigned long      state;
    int                weight;
    unsigned int       gro_count;
    struct net_device *dev;
    struct sk_buff    *gro_list;
    struct sk_buff    *skb;
    struct hrtimer     timer;
    struct list_head   dev_list;
    struct hlist_node  napi_hash_node;
    unsigned int       napi_id;
    int (*poll)(struct napi_struct *, int);
};
```

So, the code to handle C-like function pointers is missing but I guess for our codebase the above is already decent enough.

I want to know your opinions on whether:
* we should reformat the code with this patched `clang-format` and the config I'll post below
* we should wait for the patch to be mainlined (note that it's been there since Dec 2016) and then do the reformat
* we should reformat the code with `PointerAlignment: Left` (which I'm personally against of)

Hand-checked config from the dump that represents what I think should be a good compromise:

```yaml
---
Language:        Cpp
# BasedOnStyle:  WebKit
AccessModifierOffset: -4
AlignAfterOpenBracket: Align
AlignConsecutiveAssignments: true
AlignConsecutiveDeclarations: true
AlignEscapedNewlines: Right
AlignOperands: true
AlignTrailingComments: true
AllowAllParametersOfDeclarationOnNextLine: true
AllowShortBlocksOnASingleLine: false
AllowShortCaseLabelsOnASingleLine: false
AllowShortFunctionsOnASingleLine: All
AllowShortIfStatementsOnASingleLine: false
AllowShortLoopsOnASingleLine: false
# AlwaysBreakAfterDefinitionReturnType: None <- deprecated
AlwaysBreakAfterReturnType: None
AlwaysBreakBeforeMultilineStrings: false
AlwaysBreakTemplateDeclarations: false
BinPackArguments: true
BinPackParameters: true
BreakBeforeBraces: Custom
BraceWrapping:
  AfterClass:      false
  AfterControlStatement: false
  AfterEnum:       false
  AfterFunction:   true
  AfterNamespace:  false
  AfterStruct:     false
  AfterUnion:      false
  AfterExternBlock: false
  BeforeCatch:     false
  BeforeElse:      false
  IndentBraces:    false
  SplitEmptyFunction: true
  SplitEmptyRecord: true
  SplitEmptyNamespace: true
BreakBeforeBinaryOperators: NonAssignment
BreakBeforeInheritanceComma: false
BreakBeforeTernaryOperators: true
BreakConstructorInitializers: BeforeComma
BreakStringLiterals: false
ColumnLimit:     80
CommentPragmas:  '^ IWYU pragma:'
CompactNamespaces: false
ConstructorInitializerAllOnOneLineOrOnePerLine: false
ConstructorInitializerIndentWidth: 4
ContinuationIndentWidth: 4
Cpp11BracedListStyle: false
DerivePointerAlignment: false
DisableFormat:   false
FixNamespaceComments: true
ForEachMacros:
  - foreach
  - Q_FOREACH
  - BOOST_FOREACH
IncludeCategories:
  - Regex:           '^"(llvm|llvm-c|clang|clang-c)/'
    Priority:        2
  - Regex:           '^(<|"(gtest|gmock|isl|json)/)'
    Priority:        3
  - Regex:           '.*'
    Priority:        1
IncludeIsMainRegex: '(Test)?$'
IndentCaseLabels: false
IndentPPDirectives: AfterHash
IndentWidth:     4
IndentWrappedFunctionNames: true
KeepEmptyLinesAtTheStartOfBlocks: false
MacroBlockBegin: ''
MacroBlockEnd:   ''
MaxEmptyLinesToKeep: 1
NamespaceIndentation: Inner
PenaltyBreakAssignment: 2
PenaltyBreakBeforeFirstCallParameter: 19
PenaltyBreakComment: 300
PenaltyBreakFirstLessLess: 120
PenaltyBreakString: 1000
PenaltyExcessCharacter: 1000000
PenaltyReturnTypeOnItsOwnLine: 60
PointerAlignment: Right # ???
ReflowComments:  true
SortIncludes:    true
SortUsingDeclarations: true
SpaceAfterCStyleCast: false
SpaceAfterTemplateKeyword: true
SpaceBeforeAssignmentOperators: true
SpaceBeforeParens: ControlStatements
SpaceInEmptyParentheses: false
SpacesBeforeTrailingComments: 1
SpacesInAngles:  false
SpacesInContainerLiterals: true
SpacesInCStyleCastParentheses: false
SpacesInParentheses: false
SpacesInSquareBrackets: false
Standard:        Cpp11
TabWidth:        8
UseTab:          Never
```

tatokis, 2018-01-22:
https://github.com/ckb-next/ckb-next/issues/29

closed 2018-01-22 by tatokis