mattanger/ckb-next #268: Can't install ckb-next on ubuntu 16.04 gnome

NguyenHongLoi, 2017-08-05:
I can't install ckb-next on ubuntu 16.04 gnome,
with `./quickinstall` or `./qmake-auto && make`
this is my error:

`keymap.c:298:13: error: expected expression before ‘__attribute__’
             __attribute__ ((fallthrough));  // Type 2: media key (fall through)
             ^
make[2]: *** [debug/keymap.o] Error 1
make[1]: *** [debug] Error 2
make: *** [sub-src-ckb-daemon-make_first] Error 2
make: *** Waiting for unfinished jobs....
`
[image]
How can i fix it?

NguyenHongLoi, 2017-08-05:
i just delete this code:
`else if(urbinput[0] == 2)
            __attribute__ ((fallthrough));  // Type 2: media key (fall through)` 
then everything run okay, so something wrong here?

light2yellow, 2017-08-05:
If you just want to compile, remove only `__attribute__ ((fallthrough))` from that line, so that it looks like

```c
else if(urbinput[0] == 2)
    ; // Type 2: media key (fall through)
```

If you want to help resolving the issue, post the output of `gcc --version`:

```
 gcc --version                                                                                                                                                                     
gcc (GCC) 7.1.1 20170630
Copyright (C) 2017 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
```

light2yellow, 2017-08-05:
I can repro it with clang 4.0.1.

light2yellow, 2017-08-05:
@NguyenHongLoi 

pull the new code and try building now.

light2yellow, 2017-08-05:
The code builds on Ubuntu 16.04 under gcc 5.4. Tested by @tatokis. I tested gcc 7.1.1 and clang 4.0.1.

closed 2017-08-05 by light2yellow