mattanger/ckb-next #270: Customise the number of lines scrolled per mouse wheel 'tick'

Electr0Fi, 2017-08-05:
# Feature request

## What is the program's current behavior?
The lines scrolled per mouse wheel scroll 'tick' is too large and cannot currently be changed.

## What would you like it to do instead?
Be able to customise the number of lines scrolled per mouse wheel 'tick'.

## Any other notes & comments?
See this line for a workaround / possible solution that could be integrated:
https://www.reddit.com/r/MacOS/comments/5espoo/fix_mouse_wheel_scrolling/

Currently using a Harpoon mouse on OSX Sierra 10.12.6
[ckb-dev-detect-report.gz](https://github.com/mattanger/ckb-next/files/1202677/ckb-dev-detect-report.gz)

245-c, 2017-08-05:
I have the same issue in Chrome and Safari, where one "tick" is about 20 lines.
In Atom, however, it's perfectly fine. One tick is only 2 lines.
Terminal: 9 lines
Numbers: 3 cells/lines
Mail: behaves more like PageUp/Down
Thunderbird: 6 lines
Messages: about 20 (similar to Chrome)

Also on 10.12.6 with M65 Pro RGB

245-c, 2017-08-21:
I just found the setting in the Troubleshooting section of the readme:

> Scroll wheel does not scroll As of #c3474d2 it's now possible to disable scroll acceleration from the GUI. You can access it under "OSX tweaks" in the "More settings" screen. Once disabled, the scroll wheel should behave consistently.

Worked perfectly for me. @Electr0Fi , this issue can be closed, I guess.

Electr0Fi, 2017-08-22:
Huh, yeah seems to have fixed the issue for me as well.
I can't believe I missed that. I looked through all the sittings.
*shrugs*
 
Thanks @245-c !

closed 2017-08-22 by Electr0Fi