mattanger/ckb-next #275: No Devices Connected after sleep or device unplugged/plugged back in

GForceWeb, 2017-08-11:
Debugging Files:
[ckb-dev-detect-report.gz](https://github.com/mattanger/ckb-next/files/1219061/ckb-dev-detect-report.gz)
[USB Bus Probe.txt](https://github.com/mattanger/ckb-next/files/1219088/USB.Bus.Probe.txt)

I'm running a K70 Lux RGB Keyboard on my 2015 Macbook Pro. It's typically connected through a Plugable UD-3900 Hub, however, I've seen the same results when plugged directly into the laptop.

I was on the 0.28 release but I've now moved to a manual build running the latest commits on the master branch, with no perceivable change.

Upon a fresh boot with everything plugged in, ckb-next detects the keyboard correctly and everything works as expected (bar a tiny issue with the command key not working for a few specific keyboard shortcuts but I'm not sure that's a relevant issue for ckb-next). If the Macbook sleeps, or the keyboard is unplugged (say to take the laptop into a meeting) and then plugged back in ckb-next can no longer detect the keyboard. I lose my custom keybindings and RGB controls which makes the keyboard a pain to use as I'm used to my custom layout. 

Unplugging and replugging the keyboard (either into the Hub or the laptop directly) has no effect. I've also tried closing ckb-next, unloading the daemon, loading the daemon and restarting ckb-next which also had no effect.

A restart with the keyboard connected pretty much always restores normal function. But it's a massive pain to have to restart every time the laptop sleeps or is moved from my dock.

Both the detect report and USB Bus Probe were taken while the keyboard was undetected by ckb-next.

From another thread I found the following command:

```sudo launchctl unload /Library/LaunchDaemons/com.ckb.daemon.plist && sudo killall -9 ckb-daemon ; sudo /Applications/ckb.app/Contents/Resources/ckb-daemon```

Running this produced the following output (the process seems to get stuck after the last line):

```No matching processes were found
    ckb: Corsair RGB driver v0.2.8
[I] Root controller ready at /var/run/ckb0
[I] Connecting Corsair Gaming K70 LUX RGB Keyboard  at /var/run/ckb1
[W] iterate_devices_hid (usb_mac.c:871): Failed to open device: e00002c5
[E] iterate_devices_hid (usb_mac.c:854): Failed to create device plugin: e00002be
[W] iterate_devices_hid (usb_mac.c:871): Failed to open device: e00002c5
[E] iterate_devices_hid (usb_mac.c:854): Failed to create device plugin: e00002be
[I] Registering EventTap for modifier keys.
```

Checking the /var/run folder I can only see a ckb0 folder (no ckb1 folder)

If there's any other information I can provide on this issue that would be helpful please let me know.

tatokis, 2017-08-11:
@GForceWeb Thank you for your detailed report, truly appreciated. 
Not sure what causes this, but e00002c5 usually means that the the daemon couldn't get exclusive access of the device.

I'll look into it further soon.

light2yellow, 2017-08-11:
Are you sure you are not using any additional device controllers like Karabiner?

light2yellow, 2017-08-11:
From what Google told me, `Failed to open device: e00002c5` means that some other process is using the keyboard. Please, post the output of `ps axco command | uniq -u`, it might help.

GForceWeb, 2017-08-12:
Aha! Thanks for much for the suggestion @light2yellow. I did have Karabiner Elements installed. 

Closing Karabiner Elements and ckb-next and then running the unload, and restart daemon commands from my first post produced the following output (it still got stuck without appearing to complete)

```
/Library/LaunchDaemons/com.ckb.daemon.plist: Could not find specified service
No matching processes were found
    ckb: Corsair RGB driver v0.2.8
[I] Root controller ready at /var/run/ckb0
[I] Connecting Corsair Gaming K70 LUX RGB Keyboard  at /var/run/ckb1
[E] iterate_devices_hid (usb_mac.c:854): Failed to create device plugin: e00002be
[E] iterate_devices_hid (usb_mac.c:854): Failed to create device plugin: e00002be
[I] Starting input thread for /var/run/ckb1
[I] Setup finished for /var/run/ckb1
[I] Registering EventTap for modifier keys.
```

Most importantly though, the device has reappeared in ckb-next and appears to working as intended once again.

I think this issue can safely be chalked up to a compatibility issue with Karabiner Elements. This isn't a big deal for me right now as I'm not really using anything in Karabiner actively. As they expand and improve their project to the former glory of the original Karabiner it would be nice if it could play nice with ckb-next :)

light2yellow, 2017-08-12:
@GForceWeb 

> I think this issue can safely be chalked up to a compatibility issue with Karabiner Elements.

Karabiner-elements and ckb-next are similar in a fact that they act as controllers for the devices. It is impossible to register multiple controllers for a HID device, so one has to choose what controller to use. In future ckb-next might gain most of the functionality of Karabiner-elements - all kind of bindings, remaps etc. - but for Corsair products. But currently there's that.

`Failed to create device plugin: e00002be` still means there's something that prevents ckb-next from taking full control. Try `sudo launchctl list` (besides the `ps` command I posted) and then look for similar programs, also, don't forget to _unload_ Karabiner-elements because it might instantly restart it otherwise.

closed 2017-08-12 by light2yellow

light2yellow, 2017-08-12:
One thing I should add for the full picture: to _permanently_ disable the job from starting at boot use
```
sudo launchctl -w unload [job]
```
e.g. notice the `-w`.