mattanger/ckb-next #276: Keeps crashing!

AmethystWarrior, 2017-08-12:
Everytime I try to open the app it crashes. I have tried re-installing it multiple times and restarting my macbook however it continues to crash.

AmethystWarrior, 2017-08-12:
I have since re-downloaded it another time and for a second before it crashed it would show a little preview of my devices showing that they are actually recognised which is an improvement. However it still continues to regularly crash. I have a K70 rgb lux keyboard and the sabre rgb mouse and I use the 11" macbook air 2015 model.

light2yellow, 2017-08-12:
I would recommend you to open a new issue, but this time reading the issue template and filling it out. You have literally provided zero information to work with.

closed 2017-08-12 by light2yellow