mattanger/ckb-next #277: Any plans for K55 keyboard and Harpoon Mouse?

DGxInfinitY, 2017-08-16:
I just got both and was wondering if I could maybe help in development and what plans are in the works for these pieces of kit.

tatokis, 2017-08-16:
The harpoon is already fully supported, have to get the code from the newdev branch.
Regarding the K55, please run ckb-dev-detect and attach it.

DGxInfinitY, 2017-08-16:
How can I run ckb-dev-detect, I am on Linux and have compiled but I see no executable in the bin directory...

DGxInfinitY, 2017-08-16:
Also, I am getting no devices detected with Harpoon Mouse...

light2yellow, 2017-08-17:
IIRC @SethBarberee was working on K55. I guess we should turn this into a K55 thread, since we don't have one yet.

SethBarberee, 2017-08-17:
I actually was just thinking out loud on how to implement support. I don't have a K55 to test with any code that I would write.

DGxInfinitY, 2017-08-17:
Alright I got the harpoon to work after compiling the newdev branch like you said. It seems to be glitchy but at least it works. 
Back to the K55
I have no idea on how I can help, what development libraries I need or other so if someone could fill me in I would love to be if assistance. Thanks guys you are so quick to help it's quite refreshing to see a couple of great devs!

DGxInfinitY, 2017-08-17:
BTW I do have the K55 in hand so I can test the code you write.

tatokis, 2017-08-19:
Yeah, not quite sure how this is implemented.
I'll add it as a new device soon, and we can experiment and see if we can get any of the lights to work.

tatokis, 2017-08-20:
@DGxInfinitY Does the K55 even work with the CUE software? If not, there's not much we can do about it.

DGxInfinitY, 2017-08-21:
Just looked it up and it appears not but it does use function keys so if there is a way to send out keystrokes to the keyboard we could control the keyboard through specific key combos. We will have to look into the actual keyboard logic board, I don't know why they don't keep everything simple with these things, it actually complicates manufacturing for them...
[image]

tatokis, 2017-09-11:
Yeah, there is no official way to control this keyboard from software.

It might be possible with some modifications, or undocumented instructions, as it does have the required endpoints, however this is out of scope for this project.

If anyone ever figures it out, or adds support, feel free to open a PR.
Until then there is nothing we can do.

tatokis, 2017-12-01:
Apparently control through CUE is possible, reopening.
Anyone who runs Linux and is interested in spending a good few hours on this, feel free to join our IRC room (at sensible EU times).

IbrahimHamid, 2017-12-24:
I am willing to help. As I myself, have an ArchLinux and a K55 RGB Corsair Keyboard. (all I want to do is to be able to use the RGB lights of the keyboard from time to time. Haha.) CKB is not detecting K55 RGB Corsair keyboard. I am willing to spend as long as it takes. To go to a windows computer and save the output info of K55 RGB code, and back into my ArchLinux.. etc... Pacific Time -8. Let me know. Peace be upon You all. :)

ZirconiumX, 2017-12-24:
I have a branch which detects and connects to the K55. At the moment lighting does not work. We're taking a break for the holidays, so don't expect significant progress for a while.

IbrahimHamid, 2017-12-25:
Of course. Rest well. Well if there is anyway I can contribute to helping the linux community code for K55 RGB, then I will be willing. My keyboard is always detected via Arch but not CKB and I would love to help the RGB light button work for the linux community and myself.

Barrosv1, 2018-01-12:
New here and to programming, with a k55 willing to help if needed. Would like to get my k55 running on my mac with ckb.

tatokis, 2018-01-12:
Sorry, my mistake for not posting here. There is some K55 support now that has only been tested on linux. You can get the code from #362.
@Barrosv1 if you would like, compile the code from that branch/PR, instructions on how to do that on macOS are in the readme. If it doesn't work on macOS, please do post the log/output of ckb-daemon.

tatokis, 2018-01-22:
https://github.com/ckb-next/ckb-next/issues/30

closed 2018-01-22 by tatokis