mattanger/ckb-next #278: ckb-next is randomly crashing (segfault's)

Aaron432, 2017-08-16:
Corsair Gaming M65 Pro RGB Mouse @ Linux Mint 18.1 64-bit
CKB downloaded 2017.08.15 (v0.2.8)

```
 ~ $ qmake -v
QMake version 3.0
Using Qt version 5.5.1 in /usr/lib/x86_64-linux-gnu
```

---

Helo!
I've quickinstall-ed "master" branch (installation worked smoothly) and `ckb` app crashes randomly after couple minutes of working (about 1-10 min) without any specific reason or rule to it (could just run in background/tray or be used/active in foreground).

Output from `dmesg` after couple such occurrences:

```
 ~ $ dmesg | tail
[721062.120188] input: ckb1: Corsair Gaming M65 Pro RGB Mouse as /devices/virtual/input/input18
[721062.120747] input: ckb1: Corsair Gaming M65 Pro RGB Mouse as /devices/virtual/input/input19
[722682.655471] ckb[6715]: segfault at 7f1a56e09608 ip 00007f1a61e9a0c1 sp 00007fffad734240 error 4 in libQt5Core.so.5.5.1[7f1a61bd9000+4c5000]
[722905.257355] ckb[13840]: segfault at 7f91d0a3c608 ip 00007f91dbacd0c1 sp 00007ffc82054500 error 4 in libQt5Core.so.5.5.1[7f91db80c000+4c5000]
[725176.655820] ckb[4125]: segfault at 7feb4530a608 ip 00007feb5039b0c1 sp 00007ffd04adabb0 error 4 in libQt5Core.so.5.5.1[7feb500da000+4c5000]
[725606.951687] ckb[11922]: segfault at 7f20a55a7608 ip 00007f20b06380c1 sp 00007fff74bab6b0 error 4 in libQt5Core.so.5.5.1[7f20b0377000+4c5000]
[725721.459038] ckb[12462]: segfault at 7f83d0a73608 ip 00007f83dbb040c1 sp 00007fff94f643c0 error 4 in libQt5Core.so.5.5.1[7f83db843000+4c5000]
[725775.892154] ckb[22957]: segfault at 7f0729445608 ip 00007f07344d60c1 sp 00007fff4ed514d0 error 4 in libQt5Core.so.5.5.1[7f0734215000+4c5000]
[725894.819699] ckb[24894]: segfault at 7f41e7dea608 ip 00007f41f2e7b0c1 sp 00007ffc85c19980 error 4 in libQt5Core.so.5.5.1[7f41f2bba000+4c5000]
[726363.404739] ckb[4375]: segfault at 7fe2024db608 ip 00007fe20d56c0c1 sp 00007ffe30a910f0 error 4 in libQt5Core.so.5.5.1[7fe20d2ab000+4c5000]
```

And here's what `ckb` is returning while run in terminal:

```
 ~ $ ckb
Path  to settings is "/home/user/.config/ckb/ckb.conf"
Scanning  "/usr/lib/ckb-animations/ckb-gradient"
Scanning  "/usr/lib/ckb-animations/ckb-heat"
Scanning  "/usr/lib/ckb-animations/ckb-pinwheel"
Scanning  "/usr/lib/ckb-animations/ckb-rain"
Scanning  "/usr/lib/ckb-animations/ckb-random"
Scanning  "/usr/lib/ckb-animations/ckb-ripple"
Scanning  "/usr/lib/ckb-animations/ckb-wave"
ckb: Fatal IO error 10 (No child processes) on X server :0.
QProcess: Destroyed while process ("sh") is still running.
Segmentation fault
```

After 8 crashes within an hour I've quickinstall-ed "newdev" branch and it seems to be doing slightly better (best run was about 2h), but it is eventually still crashing in this same manner.

BTW... should `ckb-daemon` handle all configuration after quitting GUI? Because it's not and mouse is switching to some crazy profile with ultra-high DPI plus DPI-changing/sniping buttons are not working at all and DPI LED goes completely off. Even when I use default HwProfile1/HwMode1 it's still this same craziness after manually quitting GUI. Strangely, after it quits for itself by crashing (segfault), mouse retains custom LED colors, but buttons re-mapped into launching custom commands are not doing anything.
`ckb-daemon` works very stable though.

Yeah... let me know if and how can I provide more info.

Thanks for your work and best regards!

light2yellow, 2017-08-17:
Please, consult with any `dgb` tutorial, read [How to create useful crash reports](https://community.kde.org/Guidelines_and_HOWTOs/Debugging/How_to_create_useful_crash_reports), install debug symbols, get backtraces and post your results.

tatokis, 2017-08-17:
Duplicate of https://github.com/mattanger/ckb-next/issues/47

closed 2017-08-17 by tatokis