mattanger/ckb-next #280: Issue with Arch instructions in the readme

C0rn3j, 2017-08-17:
The following in the readme is wrong, -S flag should not be used with pacman as Arch linux does not support partial upgrades. 

Replacing it with -Syu would be appropriate.

```
Arch: sudo pacman -S base-devel qt5-base zlib
```

fleischie, 2017-08-18:
`-S` does not partially upgrade. This flag tells pacman to install the list of packages according to the currently available package list.

`-Syu` fetches the new package list, upgrades the system according to it and then installs the package, resolving new and/or outdated dependencies.

The installation instructions don't imply, that you have an outdated system and are not a guide on how you should keep your system up to date and thus only state the package names you would have to install for compiling the project sources.

But I agree, that - if anything - `-Sy` is incorrect, see the [pacman reference](https://wiki.archlinux.org/index.php/Pacman#Installing_packages).

closed 2017-08-18 by fleischie

C0rn3j, 2017-08-18:
https://wiki.archlinux.org/index.php/System_maintenance#Partial_upgrades_are_unsupported

>The installation instructions don't imply, that you have an outdated system and are not a guide on how you should keep your system up to date and thus only state the package names you would have to install for compiling the project sources.

Fair enough, it just happens that an -S flag was breaking things for me too often during various circumstances, and you're right that unless someone does -Sy or -Syu and ^C out of it, nothing bad will happen with using -S.