mattanger/ckb-next #282: STRAFE RGB Not detected at all.

SG420, 2017-08-19:
I recently installed Manjaro i3 with OpenRC via manjaro architecht, and ckb does not recognise my STRAFE RGB at all, unlike the other guy I don't appear to be getting any errors. I have tried all the versions of ckb-next and even tried downloading the old ckb from source.
[This is what it looks like (imgur)](https://i.imgur.com/rFC8PfY.png)
[ckb-dev-detect-report.gz](https://github.com/mattanger/ckb-next/files/1236729/ckb-dev-detect-report.gz)
ckb-next worked before when I was running the default manjaro kde (even in i3), but in this new install it doesnt appear to be detecting it. I am running kernel `4.13.0-1-MANJARO`. `lsusb` outputs
```
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 007: ID 058f:9360 Alcor Micro Corp. 8-in-1 Media Card Reader
Bus 001 Device 005: ID 0b05:17e8 ASUSTek Computer, Inc. 
Bus 001 Device 003: ID 1b1c:1b20 Corsair 
Bus 001 Device 006: ID 1130:1620 Tenx Technology, Inc. 
Bus 001 Device 004: ID 264a:1006  
Bus 001 Device 002: ID 1a40:0101 Terminus Technology Inc. Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
```

light2yellow, 2017-08-19:
Even though there's an OpenRC service file, any branch except `cmake` won't properly work on OpenRC. See #281.

closed 2017-08-19 by light2yellow

SG420, 2017-08-20:
Even after installing the cmake branch, it still says that no keyboard has been detected.

tatokis, 2017-08-20:
Try stopping ckb-daemon using OpenRC and then manually run `sudo ckb-daemon` on a terminal.

light2yellow, 2017-08-20:
@tatokis OpenRC is not even detected by `quickinstall` and instead OpenRC system is treated as an Upstart one (as I mentioned on IRC few days ago). So, regarding OpenRC the installation is completely broken and indeed one can only run `ckb-daemon` on the terminal every time.