mattanger/ckb-next #286: K65 RGB - Japanese version/layout support

manvari, 2017-08-27:
__Switch on the "Preview" mode for better reading experience.__
Use it to put ticks and then switch back to write in the details where needed.

- [Before you proceed](#before-you-proceed)
- [Devices information](#devices-information)
- [New device support request](#new-device-support-request)
- [Feature request](#feature-request)
    - [What is the program's current behavior?](#what-is-the-programs-current-behavior)
    - [What would you like it to do instead?](#what-would-you-like-it-to-do-instead)
    - [Any other notes & comments?](#any-other-notes--comments)
- [Bug report](#bug-report)
    - [General information](#general-information)
        - [Source](#source)
        - [Branch](#branch)
    - [Logs & crash reports](#logs--crash-reports)
    - [What is the program's current behavior?](#what-is-the-programs-current-behavior-1)
    - [What is the expected behavior of the program?](#what-is-the-expected-behavior-of-the-program)
    - [Any other notes & comments?](#any-other-notes--comments-1)
- [Something else](#something-else)

---

# Before you proceed

- [x] __Try__ the latest code by cloning and building the project manually.
- [x] __Read__ the [`README.md`](https://github.com/mattanger/ckb-next/blob/master/README.md) _in full_.
- [x] __Skim__ through other `.md` documents in the project's root. They contain more specific information.
- [x] __Search__ in open _and_ closed [Issues](https://github.com/mattanger/ckb-next/issues). [Here's how](https://help.github.com/articles/searching-issues/).
- [x] __Avoid__ opening a new issue in case a similar one already exists (don't worry, we get the notifications for closed issues as well, it will be reopened if needed).
- [x] __Keep__ unrelated issues separate (create a new one if you are sure yours is unique).
- [x] __Use__ [Markdown](https://github.com/adam-p/markdown-here/wiki/Markdown-Cheatsheet) to write on GitHub. Use triple backticks if copy-pasting more than _one_ line.

Now, __before you proceed even further__, make sure you are following the recommendations above. It will save a lot of time for you and us in the future.

---

# Devices information

Run [`ckb-dev-detect`](https://github.com/mattanger/ckb-next/blob/master/ckb-dev-detect) from the root of the source code tree:

```console
$ ./ckb-dev-detect
```

Upload `ckb-dev-detect-report.gz` on GitHub with this issue.

[ckb-dev-detect-report.gz](https://github.com/mattanger/ckb-next/files/1254263/ckb-dev-detect-report.gz)

---

# New device support request

- [x] Yes
- [ ] No

<sub><sup>leave empty if you have no new device support requests</sup></sub>

You should upload:
* __Linux__: `ckb-dev-detect-report.gz` is enough
* __macOS__: a full dump following [the instructions](https://github.com/mattanger/ckb-next/issues/31#issuecomment-285380447)

---

# Feature request

- [x] Yes
- [ ] No

<sub><sup>leave empty if you have no feature requests</sup></sub>

## What is the program's current behavior?

Currently, ckb-next can be smoothly used with Japanese K65 RGB keyboards, but it lacks support for specific Japanese keys (considering the keyboard has >105 keys).

The following keys are seemingly unmapped in all layouts offered by ckb-next: kana, the two keys surrounding the space bar, the vertical bar/yen symbol, dash/backslash.

## What would you like it to do instead?

To ensure correct Romaji input, a specific Japanese layout should be offered by ckb-next.

## Any other notes & comments?

When not using ckb, a X keyboard layout that works well is `jp 106` (set via `setxkbmap jp 106`).

Also, would ckb-next support switching to kana mode from Romaji and vice-versa on Linux? Or is there another way to switch to kana?

---

# Bug report

- [ ] Yes
- [x] No

<sub><sup>leave empty if you have no bug reports</sup></sub>

## General information

### Source
(how and where did you get this program, e.g.: _"`ckb-next-git` package in AUR"_ or _"built manually using quickinstall script"_ or _"pkg for macOS"_)

I installed ckb-next-latest-git from AUR.

### Branch
(fill in if ckb-next was compiled manually, e.g.: _"master"_ or _"newdev"_)

newdev.

## Logs & crash reports
(any useful information - an OS crash report, debugger's backtrace, any non-standard way of usage (BIOS mode ...)  etc.)

You should upload:
- Linux: `ckb-dev-detect-report.gz` is enough
- macOS:
    * the output of `ckb-daemon`:
    ```console
    $ sudo launchctl unload /Library/LaunchDaemons/com.ckb.daemon.plist
    $ sudo /Applications/ckb.app/Contents/Resources/ckb-daemon
    ```
    * the output of `ckb` (open another terminal):
    ```console
    $ sudo killall -9 ckb
    $ /Applications/ckb.app/Contents/MacOS/ckb
    ```

## What is the program's current behavior?

## What is the expected behavior of the program?

## Any other notes & comments?

Here is a picture of the keyboard. The product SKU is CH-900222-JP.

[image]

---

# Something else

<sub><sup>if your issue is about something else, try your best to describe it here, otherwise leave empty</sup></sub>

manvari, 2017-08-27:
Some initial tinkering, with `setxkbmap jp 106` set and the following setup:

- scancodes obtained after stopping ckb-daemon, with `showkeys --scancodes` (as root)
- keycodes obtained after stopping ckb-daemon, with `showkeys --keycodes` (as root)
- RGB codes obtained with both ckb-daemon and the ckb GUI running, with `cat /dev/input/ckb1/notify1`

|                            Key                            |           Position on keyboard           | Scancode | Keycode | Corsair RGB code |
|:---------------------------------------------------------:|:----------------------------------------:|:--------:|:-------:|------------------|
| halfwidth/fullwidth/kanji (半角 / 全角 / 漢字)            | top left key (above Tab)                 | 0x29     | 41      | grave            |
| non-conversion (無変換)                                   | 1st key left of space bar                | 0x7b     | 94      | #144             |
| conversion (変換)                                         | 1st key right of space bar               | 0x79     | 92      | #145             |
| katakana/hiragana/rōmaji (ひらがな / カタカナ / ローマ字) | 2nd key right of space bar               | 0x70     | 93      | #66              |
| underscore/backslash                                      | 1st key left of right shift (above Menu) | 0x73     | 89      | #83              |
| vertical bar/yen symbol                                   | 1st key left of backspace                | 0x7d     | 124     | #85              |

1. ckb-daemon not running
2. setxkbmap jp 106
3. scancodes obtained using showkey --scancodes (as root)
4. keycodes obtained using showkey --keycodes (as root)
 
Key --- Position on the keyboard --- Scancode --- Keycode
 
halfwidth/fullwidth/kanji (半角 / 全角 / 漢字) --- top left key (above Tab) --- 0x29 --- 41
non-conversion (無変換)  --- 1st key left of space bar --- 0x7b --- 94
conversion (変換) --- 1st key right of space bar --- 0x79 --- 92
katakana/hiragana/rōmaji (ひらがな / カタカナ / ローマ字) --- 2nd key right of space bar --- 0x70 --- 93
vertical bar and yen symbol --- 1st left of backspace --- 0x7d --- 124
underscore and backslash --- 1st left of right shift (above Menu) --- 0x73 --- 89

manvari, 2017-09-22:
K65 Wireshark traces, to find out RGB codes for Japanese keys:
[k65_wireshark.tar.gz](https://github.com/mattanger/ckb-next/files/1325941/k65_wireshark.tar.gz)

closed 2018-01-21 by tatokis

ZirconiumX, 2018-02-12:
Hi Maxime, would you mind dropping by the IRC channel? We need to test a theory on keyboard layouts in the protocol.

ZirconiumX, 2018-02-24:
Hi again, I was reading the CKB source code and stumbled across [this](https://github.com/ckb-next/ckb-next/blob/master/src/ckb-daemon/structures.h#L18-L21) which doesn't have an entry for Kana; I was just wondering if that light works on your keyboard or not. If not, I'll try to add it to the daemon.

manvari, 2018-02-25:
Actually, the katakana/hiragana/rōmaji (the one to the left of Fn) is highlighted, but Fn is not highlighted. In the CKB gui however, both show up as being highlighted.

Here's my keyboard:

[image]

and the matching CKB config:

[image]

tatokis, 2018-02-25:
That has been fixed. 
https://github.com/ckb-next/ckb-next/blob/master/src/ckb-daemon/keymap_patch.c#L5

manvari, 2018-02-25:
@tatokis Unfortunately, I'm still getting the same issue with ckb-next at the latest commit (https://github.com/ckb-next/ckb-next/commit/5edc7af254fdcce811506f96fc3af98d0249526c).

tatokis, 2018-02-25:
@manvari I'm dumb.
Try again from the latest master.
https://github.com/ckb-next/ckb-next/commit/30131b18de631d709b5df2fdad4116b345597b58

manvari, 2018-02-26:
Now works like a charm. Thank you!

tatokis, 2018-02-26:
Glad to hear :)