mattanger/ckb-next #287: modifier keys become stuck

MatkoUSKO, 2017-08-29:
[ckb-dev-detect-report.gz](https://github.com/mattanger/ckb-next/files/1261654/ckb-dev-detect-report.gz)

I am running latest version of ckb with k70rgb keyboard (alongside with corsair harpoon mouse). OS: macOS Sierra 10.12.6

Sometimes when i press and release modifier keys, they become "stuck" and my computer "thinks" they are still pressed (and not released). This usually happens when I am pressing them with mouse buttons (I have mapped some shortcuts e.g. when i press shift and one of mouse buttons I move one space left or right. This is handled by Better Touch Tool software). 

It usually doesn't matter if I hold shift for longer time or not. I can recreate this issue in a few tries and it seems to happen randomly.

Yndoendo, 2017-09-22:
I also have this just start occurring on a **STRAFE RGB** constantly with the **Super** / **Windows** key on Arch Linux.  When ```ckb-daemon``` and ```ckb``` background are running the key is either stuck in pressed or unpressed mode. Verified with **xev** and **xinput**.

Found it to be a corrupt config file, attached a copy. **Gibby** cause the key to stop working.  Everything looks correct for the binding in the UI. Didn't take time to analyse the config file itself.

This started happening after cleaning out the keyboard by tilting and shaking it and not by directly changing and editing the configuration.
[ckb.conf.txt](https://github.com/mattanger/ckb-next/files/1325009/ckb.conf.txt)

tatokis, 2018-01-22:
https://github.com/ckb-next/ckb-next/issues/32

closed 2018-01-22 by tatokis