mattanger/ckb-next #296: Ubuntu Unity login screen freezes on startup

HeroCC, 2017-09-14:
## General information 

### Source
Cloned from github and ran `./quickinstall' (currently d65d782)

## Logs & crash reports
Log of `dmesg`: https://pastebin.com/GRGA4j5D
Ckb-dev-detect-report: https://mega.nz/#!w9IECbZa!vGtHhoF1HTfgN73AxW5RHQlz8HM23o4_HBxTBFEJ184

## What is the program's current behavior?
Upon starting the computer, the (Ubuntu 14.04) login screen immediately freezes. No wallpaper is loaded, the background is purple with white dots, and I cannot click anything / use the mouse. Attempting to type doesn't do anything. A workaround is to press CTRL+ALT+F2 to open a new terminal, run `sudo service ckb-daemon stop && sleep 30 && sudo service ckb-daemon start` and switch back to the desktop once the keyboard starts responding.

## What is the expected behavior of the program?
The login screen should work as normal, not frozen on boot

## Other Information
I don't believe I updated anything before this started happening (I may be wrong though), and I am using my keyboard and mouse through a USB 3 hub if that makes a difference. 

I cut out a bunch of the template ticket info that didn't apply, but if you need something I missed let me know

light2yellow, 2017-09-15:
Did it work before?

Did you try plugging directly into a USB 3/2 port?

HeroCC, 2017-09-15:
Yes, it worked before (before ckb-next was started). I tried both and it's the same thing. I also found that if I leave it for a while (like a few mins?) it will start working and I can start typing without the command.

tatokis, 2018-01-22:
https://github.com/ckb-next/ckb-next/issues/35

closed 2018-01-22 by tatokis