mattanger/ckb-next #297: Issue with windows freezing when using multiple input devices simultaneously

XtrmJosh, 2017-09-15:
I don't believe this is an issue with ckb as I've seen very similar behaviour using a Razer Naga, as well as a Logitech G602. I'm not sure where the problem lies and in all honesty I suspect it will be within xserver, but I'm fairly confident that somebody here will have seen this issue, or will see it as soon as they go looking. Sadly, I don't think I'll get much of a positive experience trying to file a bug with xserver, so I came here first (after Ask Ubuntu guided me here).

I've made a video of the issue, can be seen here: https://youtu.be/UDOb0bTeW8g

The device which I'm using is a Scimitar RGB, in conjunction with a KBC Poker 3 (although I've tried a few other keyboards with the same result).

What I'm doing is repeatedly pressing 1 button on the keyboard and mouse at the same time. In the video above, you'll see I'm typing 'asd' repeatedly, whilst pressing the '3' button on the mouse, which is bound to F3 (have tried binding to the number 3, with the same result).

What's happening is that when both buttons are pressed at the same time, the window freezes for a few seconds (about 3 seconds, in the video), and input as well as output is frozen. Sometimes the freeze is considerably shorter, other times it's been up to 10 seconds.

fleischie, 2017-12-06:
Hey @XtrmJosh 

Do you still experience this issue? Did you find out more. Sorry for not responding in so long. :blush:

XtrmJosh, 2018-01-04:
Hi @fleischie 

I do still experience it, on Ubuntu 16.04 LTS and 17.10. I've kinda adapted so I never press a mouse and keyboard button simultaneously as a temporary fix, but haven't found any more info on why it happens...

tatokis, 2018-01-22:
https://github.com/ckb-next/ckb-next/issues/36

closed 2018-01-22 by tatokis