mattanger/ckb-next #298: Unable to change DPI when DPI on specific step concides with Sniper DPI

nazar-pc, 2017-09-19:
# Bug report

- [x] Yes
- [ ] No

### Source
Manually build from `master` branch as of right now: https://github.com/mattanger/ckb-next/commit/d65d782f8595fd90316a19df1607f1267e013d6e

## What is the program's current behavior?
When DPI level coincides with Sniper DPI I can't change DPI with Up/Down buttons anymore.

## What is the expected behavior of the program?
DPI Stages should not stuck when coincide with Sniper.

## Any other notes & comments?
For instance, this happens when Sniper DPI is 500 and DPI Stage 1 is also 500. Changing DPI Stage 1 to 501 fixes the issue.

pjcv, 2017-09-19:
Please take a look at https://github.com/pjcv/ckb-next/tree/dpistack (branched from newdev).

nazar-pc, 2017-09-19:
Doesn't help unfortunately.
Here are my settings:
[image]

pjcv, 2017-09-19:
Using those settings I can reproduce your issue on newdev but not on my branch... what are your bindings?

nazar-pc, 2017-09-19:
Attached complete config
[ckb.conf.gz](https://github.com/mattanger/ckb-next/files/1315319/ckb.conf.gz)

pjcv, 2017-09-19:
Thanks. I'm not able to use your settings exactly on my Glaive, but with something very similar (binding the mode-switch to the right mouse button) I'm able to reproduce the issue on newdev but not on my branch.

nazar-pc, 2017-09-20:
Maybe there is someone else with M65 Pro RGB to try?

pjcv, 2017-09-20:
Agreed it would be nice to have some other people try this, we could use some more data here. I was able to reproduce and fix at least one cause of this issue, so it seems there must be some other factor in play.

pjcv, 2017-09-20:
I made one further change which _might_ affect you, @nazar-pc:
https://github.com/pjcv/ckb-next/commit/2d5b628c97ba97ab3d9af7cc9dacf9913fe09306
This should prevent the sniper DPI from being assigned on startup, though in my case, I was still able to change DPI after this.  If you have a chance please give it another try.

nazar-pc, 2017-09-20:
Compiled from mentioned commit and it now works correctly!

pjcv, 2017-09-20:
Great! Thanks for your patience. I'll put together a PR.

closed 2017-10-14 by tatokis