mattanger/ckb-next #299: K95 Platinum support ?

Extarys, 2017-09-20:
---

# Devices information

Run [`ckb-dev-detect`](https://github.com/mattanger/ckb-next/blob/master/ckb-dev-detect) from the root of the source code tree:

```console
$ ./ckb-dev-detect
[ckb-dev-detect-report.gz](https://github.com/mattanger/ckb-next/files/1318841/ckb-dev-detect-report.gz)

```

Upload `ckb-dev-detect-report.gz` on GitHub with this issue.
Done
---

# New device support request

- [x] Yes (and no)
- [ ] No

<sub><sup>leave empty if you have no new device support requests</sup></sub>

You should upload:
* __Linux__: `ckb-dev-detect-report.gz` is enough
* __macOS__: a full dump following [the instructions](https://github.com/mattanger/ckb-next/issues/31#issuecomment-285380447)

---

The current K95 support is for the non Platinum edition, the one with the bunch of macro keys and no top light bar/no corsair rgb logo.

light2yellow, 2017-09-20:
Duplicate of #53.

closed 2017-09-20 by light2yellow

Extarys, 2017-09-20:
I though having more file would help - oe more people whatev. ok