mattanger/ckb-next #300: M65 RGB Pro sniper button makes cursor jump

Misklahr, 2017-09-21:
# Bug report

- [x] Yes
- [ ] No

## General information

### Source
Manually build from `master` branch: `d65d782`

## What is the program's current behavior?
When moving the mouse and clicking the sniper button on my Corsair M65 RGB Pro, the cursor jumps a few centimetres/inches in the direction that it is moving. This is much more noticeable when using higher DPI settings. 
When exiting the program, this behavior disappears.

## What is the expected behavior of the program?
The cursor should not jump when the sniper button is pressed.

## Any other notes & comments?
This Is probably the same issue:
http://forum.corsair.com/v3/showthread.php?t=162708
I don't have this issue on my M65 RGB (Windows) anymore.

pjcv, 2017-09-21:
I suspect this is due to an incorrect DPI being set before the new DPI is selected. Please try
https://github.com/pjcv/ckb-next/commit/86b7ae779eb21c855666473a987fd919c9fff487
and let me know if this helps.

This is likely the same cause as for #214.

Misklahr, 2017-09-22:
Hello and thank you for your fast response!
After a lot of trouble -> The mouse wasn't detected by the old software I had installed for some reason, and neither by any manually installed software (had to install the pkg version and uninstall it again for it to work) -> the software you provided unfortunately didn't work :/

The video in #214 is exactly how the mouse behaves. And since I'm using DPIs over 9000 (lol)  and the sniper button for precision, this is very  frustrating :)

nazar-pc, 2017-09-22:
I think this should be closed than in favor of #214

closed 2017-09-22 by tatokis