mattanger/ckb-next #302: Corsair M65 RGB Pro never detected on startup

Misklahr, 2017-09-26:
# Bug report

- [x] Yes
- [ ] No

## Source
Manually build from master branch: d65d782 / Latest pkg file

## What is the program's current behavior?
When starting the system, the mouse is not detected.

## What is the expected behavior of the program?
The mouse should be detected.

## Any other notes & comments?
The only way to fix this is to install the drivers with the pkg file, then it works to install the drivers manually again. 
I think this might have to do with the fact that I'm plugging the mouse into my monitor, since this is the only place I have the "older" USB ports.
The monitor is plugged in to my MacBook Pro via USB type C and does not show anything at startup until I log in, the mouse doesn't work until I log in as well.

sterpe, 2017-10-07:
I have this a lot as well.  Where I have to plugin the mouse multiple times before CKB can detect it on both Sierra and High Sierra.

tatokis, 2018-01-22:
https://github.com/ckb-next/ckb-next/issues/37

closed 2018-01-22 by tatokis